raspberrypi/rpicam-apps #633: [BUG] Trying to run rpicam-vid as a systemd service coredumps on exit

LeBirlante, 2024-01-22:
**Describe the bug**

A simple script involving rpicam-vid works flawlessly when invoked from shell:

`$ cat video.sh 

#!/bin/bash
rpicam-vid -t 0 -v 1 -n --framerate 56 --width 1920 --height 1080 --inline --listen -o tcp://0.0.0.0:8000 --autofocus-mode=manual --autofocus-range=normal
`

Now we try to create a systemd service in order to run this script:

`[Unit]
Description=RPI-cam video streaming 

[Service]
ExecStart=/home/foobar/scripts/video.sh

[Install]
WantedBy=multi-user.target
`

You can start the service and everything will work as expected. As soon as an incoming connection arrives, video will start to broadcast. 
The problem comes when the client disconnects. rpicam-video should exit cleanly, but when it's been invoked from a systemd service unit, it coredumps:

`Jan 22 21:48:54 chiquitinx systemd[1]: Started RPI-cam video streaming.
Jan 22 21:48:57 chiquitinx video[2412]: [31:21:52.096248026] [2412]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
Jan 22 21:48:58 chiquitinx video[2412]: [31:21:52.733715473] [2418]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
Jan 22 21:48:58 chiquitinx video[2412]: [31:21:52.840775951] [2418]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
Jan 22 21:48:58 chiquitinx video[2412]: [31:21:52.841023087] [2418]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Jan 22 21:48:58 chiquitinx video[2412]: Overriding H.264 level 4.2

Jan 22 21:49:03 chiquitinx websockify[2421]: 127.0.0.1 - - [22/Jan/2024 21:49:03] 127.0.0.1: Plain non-SSL (ws://) WebSocket connection
Jan 22 21:49:03 chiquitinx websockify[2421]: 127.0.0.1 - - [22/Jan/2024 21:49:03] 127.0.0.1: Path: '/ws/'
Jan 22 21:49:03 chiquitinx websockify[2421]: 127.0.0.1 - - [22/Jan/2024 21:49:03] connecting to: 0.0.0.0:8000

Jan 22 21:49:03 chiquitinx video[2412]: Mode selection for 1920:1080:12:P(56)
Jan 22 21:49:03 chiquitinx video[2412]:     SRGGB10_CSI2P,1536x864/120.135 - Score: 2200
Jan 22 21:49:03 chiquitinx video[2412]:     SRGGB10_CSI2P,2304x1296/56.0255 - Score: 1150
Jan 22 21:49:03 chiquitinx video[2412]:     SRGGB10_CSI2P,4608x2592/14.3536 - Score: 85342.8
Jan 22 21:49:03 chiquitinx video[2412]: [31:21:57.753523957] [2412]  INFO Camera camera.cpp:1183 configuring streams: (0) 1920x1080-YUV420 (1) 2304x1296-SBGGR10_CSI2P
Jan 22 21:49:03 chiquitinx video[2412]: [31:21:57.755760521] [2418]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA

Jan 22 21:49:06 chiquitinx video[2412]: terminate called after throwing an instance of 'std::runtime_error'
Jan 22 21:49:06 chiquitinx video[2412]:   what():  failed to send data on socket
Jan 22 21:49:06 chiquitinx systemd[1]: Started Process Core Dump (PID 2427/UID 0).

Jan 22 21:49:35 chiquitinx systemd-coredump[2428]: [LNK] Process 2412 (rpicam-vid) of user 0 dumped core.
                                                   
                                                   Stack trace of thread 2423:
                                                   #0  0x00000000b63ee310 n/a (libc.so.6 + 0x7e310)
                                                   ELF object binary architecture: ARM
Jan 22 21:49:35 chiquitinx systemd[1]: systemd-coredump@1-2427-0.service: Deactivated successfully.
Jan 22 21:49:37 chiquitinx video[2411]: /home/foobar/scripts/video.sh: line 2:  2412 Aborted                 (core dumped) rpicam-vid -t 0 -v 1 -n --framerate 56 --width 1920 --height 1080 --inline --listen -o tcp://0.0.0.0:8000 --autofocus-mode=manual --autofocus-range=normal

`

the crux of the matter seems to be in the  **_terminate called after throwing an instance of 'std::runtime_error'_**

**Bug report**

Interestingly, if you edit video.sh and try wrapping the rpicam-vid command using camera-bug-report:

`$ cat video.sh 

#!/bin/bash
camera-bug-report -o report.txt -c 'rpicam-vid -t 0 -v 1 -n --framerate 56 --width 1920 --height 1080 --inline --listen -o tcp://0.0.0.0:8000 --autofocus-mode=manual --autofocus-range=normal'
`

it works flawlessly and won't coredump on exit when the client disconnects and the systemd unit stops.

**Additional context**

I am trying to run rpicam-vid via a systemd service since rpicam-vid will exit whenever the streaming client disconnects, and I do need rpicam-vid to auto-restart then in order to keep serving the stream for the next client.

naushir, 2024-01-23:
What happens if you simply output a file in your script instead of making a tcp stream, i.e. using ``-o test.264`` for example?

LeBirlante, 2024-01-23:
I tried as you suggested, now the video script is:

<code>
rpicam-vid -t 0 -v 1 -n --framerate 56 --width 1920 --height 1080 --inline --listen -o test.264 --autofocus-mode=manual --autofocus-range=normal
</code>

and now it won't coredump:

<code>
Jan 23 09:27:08 chiquitinx systemd[1]: Stopping RPI-cam video streaming...
Jan 23 09:27:08 chiquitinx systemd[1]: rpicam-video.service: Failed to kill control group /system.slice/rpicam-video.service, ignoring: Invalid argument
Jan 23 09:27:08 chiquitinx systemd[1]: rpicam-video.service: Deactivated successfully.
Jan 23 09:27:08 chiquitinx systemd[1]: Stopped RPI-cam video streaming.
</code>

other than the warning about not being able to kill the control group everything works ok: the output h264 file is created, and the service stops cleanly.

So yes, it seems to coredump only when -o tcp://.... is being used within a systemd service.

naushir, 2024-01-23:
Is there some kind of permissions issue a play where systemd services are not allowed to use the network stack?  I'm sorry, not too familiar with this...

LeBirlante, 2024-01-23:
> Is there some kind of permissions issue a play where systemd services are not allowed to use the network stack? I'm sorry, not too familiar with this...

Systemd services run as root (unless stated otherwise, i.e., explicitly using a non privileged user) so I think it's unlikely we're facing a permissions issue here. In fact, when you wrap the rpicam-vid command within "camera-bug-report -c ....", everything works ok, so systemd is indeed able to run rpicam-vid.

I think the relevant bits come here:

<code>
Jan 22 21:49:06 chiquitinx video[2412]: terminate called after throwing an instance of 'std::runtime_error'
Jan 22 21:49:06 chiquitinx video[2412]:   what():  failed to send data on socket
</code>

Now I'm going out on a limb, but this looks like as if the systemd service is trying to stop the execution *while* rpicam-vid is still running/terminating.

naushir, 2024-01-24:
> Now I'm going out on a limb, but this looks like as if the systemd service is trying to stop the execution while rpicam-vid is still running/terminating.

This may be possible.  Do you have any checks to ensure the service cannot stop when the application is running?  Maybe an atomic file lock?

LeBirlante, 2024-01-26:
> > Now I'm going out on a limb, but this looks like as if the systemd service is trying to stop the execution while rpicam-vid is still running/terminating.
> 
> This may be possible. Do you have any checks to ensure the service cannot stop when the application is running? Maybe an atomic file lock?

I've done several tests, and I can tell for sure that the service won't stop until the very moment the client disconnects (I've even run a test during several hours over a full night, it was stable as a rock). 
As per rpicam-vid design, when streaming to a socket and the client disconnects, this will end its execution, which in turn will make systemd believe that the program is done (which, indeed, it is).

However, due to some unknown (at least, to me!) reason, it seems that systemd believes that rpicam-vid is done and finished *while* in reality rpicam-vid still has some running thread(s) , in what appears to be some kind of race condition: systemd threats the service as finished and tris to shut it off, while rpicam-vid still has not cleanly exited (it seems). Hence the lines:

Jan 22 21:49:06 chiquitinx video[2412]: terminate called after throwing an instance of 'std::runtime_error'
Jan 22 21:49:06 chiquitinx video[2412]:   what():  failed to send data on socket

it is worth nothing that wrapping rpicam-vid in a camera debug program like:

camera-bug-report -o report.txt -c 'rpicam-vid -t 0 -v 1 -n --framerate 56 --width 1920 --height 1080 --inline --listen -o tcp://0.0.0.0:8000 --autofocus-mode=manual --autofocus-range=normal'

will make a fine systemd service, with a clean exit and no coredump. I suppose camera-bug-report waits for rpicam-vid to cleanly exit before finishing, hence systemd gets a clean exit and everything works as expected. But a "pure" rpicam-vid command will not play along nicely with systemd.

camera-

naushir, 2024-01-26:
> As per rpicam-vid design, when streaming to a socket and the client disconnects, this will end its execution, which in turn will make systemd believe that the program is done (which, indeed, it is).

I think this may be a (the?) problem.  rpicam-vid opens the tcp socket, accepts the connection and streams.   The expectation is that rpicam-vid gets terminated first, then the client disconnects.  If the client disconnects first while the server is running, rpicam-vid throws an exception about a broken pipe or similar, and terminates abnormally.  There's no easy way to do a "nice cleanup" as things currently stand, this is just the way rpicam-vid works.

closed 2024-01-26 by naushir