raspberrypi/rpicam-apps #638: [BUG] libcamera-vid(rpicam-vid) ignore "--segment" flag

arsenicks, 2024-02-04:
**Describe the bug**
Trying to record a X minutes video(using -t X)  and want the recording to be split  in different segment so I use the "--segment X" flag. The video get recorded but in only one file. The program seems to get the parameter as you will see in the verbose mode but recording doesn't get split.

**Bug report**
[segmentbug.txt](https://github.com/raspberrypi/rpicam-apps/files/14154644/segmentbug.txt)

**Additional context**
For some reason the bug report tool doesn't report the correct command so here it is just in case this could cause any problem:
`rpicam-vid -t 100000 --segment 10000 -o "/nfs/riv/riv-pres-$(date +\%Y-\%m-\%d-\%H.\%M)-%04d.mp4" -n --width 1536 --height 864 --tuning-file /usr/share/libcamera/ipa/rpi/vc4/imx708_wide_noir.json --hdr --libav-audio --audio-source alsa --audio-device pseye_cap  --codec libav --libav-format mp4 --audio-codec alac -v 2`

Thanks

naushir, 2024-02-05:
This is a known issue. There was some bugs in the implementation that had to be reverted (see https://github.com/raspberrypi/rpicam-apps/issues/580). I've not had time to have a look again at this, but will do soon.

naushir, 2024-02-05:
Closing as a duplicate of https://github.com/raspberrypi/rpicam-apps/issues/580.

closed 2024-02-05 by naushir