raspberrypi/rpicam-apps #642: ERROR: *** no cameras available ***

aranis22, 2024-02-14:
So i recently bought the raspberry pi 4 along with the RPI-CAM-V2. The RPI-CAM-V2 did not work and had this for the following commands:
aranis@raspberrypi:~ $ libcamera-hello
[0:03:50.556163493] [2641] INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
Made X/EGL preview window
ERROR: *** no cameras available ***
aranis@raspberrypi:~ $ vcgencmd get_camera
supported=0 detected=0, libcamera interfaces=0

Many people said it gets damaged quite easily, so I assumed it was and returned it. I ordered an offbrand one instead, and got the same exact issue.
Im definitely missing something here. I highly doubt my pi doesn't support cameras. It has the csi ribbon connector with "CAMERA" written on it. What can I do to get this to work? I have to rpi 4 model B and am running bookworm.

Camera is missing in interface config : [https://cdn.discordapp.com/attachments/ ... e7b2d628dc](https://cdn.discordapp.com/attachments/908239289152847873/1207366711007776768/image.png?ex=65df62df&is=65cceddf&hm=2b4a61a28a255943d199266edc699a648f8081228388c19abe9396e7b2d628dc)&
enable legacy camera is also missing in the other config
camera i ordered instead : [https://www.amazon.com/dp/B07DNSKMZ1?ps ... ct_details](https://www.amazon.com/dp/B07DNSKMZ1?psc=1&ref=ppx_yo2ov_dt_b_product_details)[/url]

libcamera-vid --list-cameras shows:
No cameras available!

connection:
https://media.discordapp.net/attachments/908239289152847873/1207450073529716736/P_20240214_113506.jpg?ex=65dfb082&is=65cd3b82&hm=7a3c9c705f78ca8e6b42b30cec5241e74e54b7ec81f181a5ab544b24323328e4&=&format=webp&width=1766&height=994

connection:
(i do not know how to paste links in text on this, so sorry for that)

naushir, 2024-02-15:
Legacy camera is not available in our Raspberry Pi OS Bullseye images, and you don't need to set any config to enable cameras any more.  I would start again with a fresh OS image, apply all updates, and change no configuration items at all.  You might also find [this](https://forums.raspberrypi.com/viewtopic.php?t=362707) post helpful to read.

The Amazon camera you linked to is actually a Camera v1 (ov5647).  Do you still have the v2 camera (imx219)?
If you are working with the camera v1 camera from Amazon, I would start with unplugging the IR cut filter wire and reboot the device (with a fresh OS install).  Then if you run ``libcamera-hello --list-cameras`` and get none found, can you please post the output of the ``camera-bug-report`` tool here please?

aranis22, 2024-02-15:
I used raspberry pi imager and reinstalled the OS. I did not format it before doing so (does this matter?). I do not have the v2 camera.
It does work now, BUT i forgot to mention that it did detect the camera before and then stopped working after aswell. I'm not going to change anything now since it works, but will let you know if it stops working again.

I had to use libcamera-hello --qt -preview since im using xrdp. It only runs for 5 seconds and then shuts off. Is this normal? How can I run it for longer?
 Here's the output when i run libcamera-hello  --qt -preview:
[0:02:08.581324914] [2627]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:02:08.628972442] [2630]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:02:08.631159745] [2630]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media1
[0:02:08.631270245] [2630]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Made QT preview window
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[0:02:09.025015114] [2627]  INFO Camera camera.cpp:1183 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-SGBRG10_CSI2P
[0:02:09.025443893] [2630]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA

naushir, 2024-02-15:
Are you using the IR cut filter or the IR lights?  If so, the power draw will sometimes cause the camera to stop working.  We do not recommend you use these devices without an external power source!

naushir, 2024-02-15:
> I had to use libcamera-hello --qt -preview since im using xrdp. It only runs for 5 seconds and then shuts off. Is this normal? How can I run it for longer?

This is normal, it runs for 5 seconds then stops.  You can update the timeout with the ``--t`` command line argument.

naushir, 2024-02-15:
Resolving now as the original problem is working.