raspberrypi/rpicam-apps #643: [BUG] rpicam-vid libav bookworm

arsenicks, 2024-02-19:
**Describe the bug**
I have a cronjob running for weeks, same command. Worked like a champ untill I updated my raspianos today. 

**Bug report**

[encoder-prob.txt](https://github.com/raspberrypi/rpicam-apps/files/14326028/encoder-prob.txt)

**Additional context**
It was working for weeks and stopped working after the update. I simply did a "apt update" on bookwork. I saw no packages being removed, only update as far as I can tell.

I trying running a simple recording and it worked as expected, so it looks related to libav codec/packages:
`libcamera-vid -t 3590000 -o "/nfs/Priv/.hidden/priv-pres-$(date +\%Y-\%m-\%d-\%H.\%M).h264" -n --width 1536 --height 864 --tuning-file /usr/share/libcamera/ipa/rpi/vc4/imx708_wide_noir.json --hdr`

Here's the list of packages that have been updated. We can see the package libavdevice59 has been updated and might sound like a good suspect!

Here's the complete list:
```
Start-Date: 2024-02-18  13:27:06
Commandline: apt upgrade
Requested-By: XXXXX (1000)
Upgrade: libperl5.36:armhf (5.36.0-7, 5.36.0-7+deb12u1), libcups2:armhf (2.4.2-3+deb12u4, 2.4.2-3+deb12u5), libcurl4:armhf (7.88.1-10+rpi1+deb12u4, 7.88.1-10+rpi1+deb12u5), libpostproc-dev:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), exfatprogs:armhf (1.2.0-1, 1.2.0-1+deb12u1), python3-av:armhf (10.0.0-1+rpt2, 10.0.0-1+rpt3), bind9-host:armhf (1:9.18.19-1~deb12u1, 1:9.18.24-1), libnghttp2-14:armhf (1.52.0-1, 1.52.0-1+deb12u1), libcurl3-gnutls:armhf (7.88.1-10+rpi1+deb12u4, 7.88.1-10+rpi1+deb12u5), openssh-client:armhf (1:9.2p1-2+deb12u1, 1:9.2p1-2+deb12u2), lxplug-volumepulse:armhf (0.23, 0.25), perl:armhf (5.36.0-7, 5.36.0-7+deb12u1), tzdata:armhf (2023c-5, 2024a-0+deb12u1), chromium-codecs-ffmpeg-extra:armhf (120.0.6099.102-rpt1, 121.0.6167.155-rpt1), cups-common:armhf (2.4.2-3+deb12u4, 2.4.2-3+deb12u5), libzbar0:armhf (0.23.92-7+b1, 0.23.92-7+deb12u1), usb.ids:armhf (2023.05.17-0+deb12u1, 2024.01.20-0+deb12u1), ssh:armhf (1:9.2p1-2+deb12u1, 1:9.2p1-2+deb12u2), libgs10-common:armhf (10.0.0~dfsg-11+deb12u2, 10.0.0~dfsg-11+deb12u3), cups-client:armhf (2.4.2-3+deb12u4, 2.4.2-3+deb12u5), cups-ppdc:armhf (2.4.2-3+deb12u4, 2.4.2-3+deb12u5), gnome-session-bin:armhf (43.0-1, 43.0-1+deb12u1), cups-daemon:armhf (2.4.2-3+deb12u4, 2.4.2-3+deb12u5), libgs10:armhf (10.0.0~dfsg-11+deb12u2, 10.0.0~dfsg-11+deb12u3), libavdevice59:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), ffmpeg:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), libssh-gcrypt-4:armhf (0.10.5-2, 0.10.6-0+deb12u1), openssh-server:armhf (1:9.2p1-2+deb12u1, 1:9.2p1-2+deb12u2), libqpdf29:armhf (11.3.0-1, 11.3.0-1+deb12u1), libgs9-common:armhf (10.0.0~dfsg-11+deb12u2, 10.0.0~dfsg-11+deb12u3), libpostproc56:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), cups-ipp-utils:armhf (2.4.2-3+deb12u4, 2.4.2-3+deb12u5), usbutils:armhf (1:014-1, 1:014-1+deb12u1), libswscale-dev:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), ghostscript:armhf (10.0.0~dfsg-11+deb12u2, 10.0.0~dfsg-11+deb12u3), usrmerge:armhf (35, 37~deb12u1), libavdevice-dev:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), chromium-browser:armhf (120.0.6099.102-rpt1, 121.0.6167.155-rpt1), libavcodec59:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), libgstreamer-plugins-bad1.0-0:armhf (1.22.0-4+deb12u3, 1.22.0-4+deb12u5), libavutil57:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), libswscale6:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), libcryptsetup12:armhf (2:2.6.1-4~deb12u1, 2:2.6.1-4~deb12u2), libcupsimage2:armhf (2.4.2-3+deb12u4, 2.4.2-3+deb12u5), distro-info-data:armhf (0.58, 0.58+deb12u1), libavutil-dev:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), perl-base:armhf (5.36.0-7, 5.36.0-7+deb12u1), openssh-sftp-server:armhf (1:9.2p1-2+deb12u1, 1:9.2p1-2+deb12u2), bind9-libs:armhf (1:9.18.19-1~deb12u1, 1:9.18.24-1), cups-core-drivers:armhf (2.4.2-3+deb12u4, 2.4.2-3+deb12u5), libswresample4:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), libavfilter-dev:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), libisl23:armhf (0.25-1, 0.25-1.1), libgnutls30:armhf (3.7.9-2, 3.7.9-2+deb12u2), libavformat59:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), cups:armhf (2.4.2-3+deb12u4, 2.4.2-3+deb12u5), curl:armhf (7.88.1-10+rpi1+deb12u4, 7.88.1-10+rpi1+deb12u5), gstreamer1.0-plugins-bad:armhf (1.22.0-4+deb12u3, 1.22.0-4+deb12u5), libgs-common:armhf (10.0.0~dfsg-11+deb12u2, 10.0.0~dfsg-11+deb12u3), perl-modules-5.36:armhf (5.36.0-7, 5.36.0-7+deb12u1), cups-server-common:armhf (2.4.2-3+deb12u4, 2.4.2-3+deb12u5), libavformat-dev:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), libavcodec-dev:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), python3-v4l2:armhf (0.3.2-1, 0.3.3-1), chromium-browser-l10n:armhf (120.0.6099.102-rpt1, 121.0.6167.155-rpt1), gnome-session-common:armhf (43.0-1, 43.0-1+deb12u1), libswresample-dev:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1), libavfilter8:armhf (8:5.1.4-0+rpt1+deb12u1, 8:5.1.4-0+rpt2+deb12u1)
End-Date: 2024-02-18  13:41:46
```

axeljerabek, 2024-02-19:
I got the same error with my stream-recorder. It seems that you have to use the ' --libav-video-codec' option now. When I specify ' --libav-video-codec libx264', it works, but it then reencodes the stream from the sensor with cpu. I still did not find the option to simply copy the raw h264 from the sensor itself, without reencoding it using all my CPU power. The option ' --libav-video-codec copy' does not exist. Does anyone have a codec or a workaround for this?

My encoder line:
rpicam-vid -t 0 --width 1920 --height 1080 --nopreview --exposure long --sharpness 1.1 --contrast 1.1 --brightness 0.2 --saturation 1.0 --awb auto --denoise auto \
--level 4.2 --codec libav --libav-video-codec yuv420 --libav-audio --audio-source alsa --audio-device hw:2,0 --audio-channels 1 --audio-codec aac --audio-samplerate 48000 --audio-bitrate 128000 --av-sync 0\
 --libav-format flv -n --framerate 30 -b 5200000  --autofocus-mode manual --lens-position 0.25 --autofocus-window 0.25,0.25,0.5,0.5 --tuning-file /usr/share/libcamera/ipa/rpi/vc4/imx708.json\
 --vflip --hflip --inline --save-pts timestamp.pts -o "rtmp://192.168.178.25/pi/test/"

worked until last week. Since the upgrade of the rpicam packages, it gives me the above error message if I do not specify another video codec. Still, I want to use the raw h264 stream directly from the sensor without reencoding it with libav/libx264.

axeljerabek, 2024-02-22:
It seems that the setting "--libav-video-codec h264_v4l2m2m" should work, since it gives me the same error. It does at least not say that it cannot find the codec. So it seems that it should woor, but doesn't:

root@pi:/home/axel# ./streamfull
[122:35:53.275719398] [1761546]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[122:35:53.430467467] [1761556]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[122:35:53.436090733] [1761556]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media1
[122:35:53.436192509] [1761556]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Mode selection for 1920:1080:12:P(30)
    SRGGB10_CSI2P,1536x864/120.135 - Score: 2200
    SRGGB10_CSI2P,2304x1296/56.0255 - Score: 1150
    SRGGB10_CSI2P,4608x2592/14.3536 - Score: 33342.8
Stream configuration adjusted
[122:35:53.444682610] [1761546]  INFO Camera camera.cpp:1183 configuring streams: (0) 1920x1080-YUV420 (1) 2304x1296-SRGGB10_CSI2P
[122:35:53.445047196] [1761556]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SRGGB10_1X10 - Selected unicam format: 2304x1296-pRAA
[h264_v4l2m2m @ 0x55c807be00] Using device /dev/video11
[h264_v4l2m2m @ 0x55c807be00] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
[h264_v4l2m2m @ 0x55c807be00] requesting formats: output=YU12/drm_prime capture=H264/none
[h264_v4l2m2m @ 0x55c807be00] Encoder requires yuv420p pixel format.
ERROR: *** libav: unable to open video codec: -22 ***

naushir, 2024-02-26:
I've tried going back in history and this is still broken on Pi4 (where we use the ``h264_v4l2m2m`` libav codec).  I suspect something in our ffmpeg release has caused this to break.  @jc-kynesim any thoughts?

axeljerabek, 2024-02-26:
If it helps in finding the bug: I use my own compiled Ffmpeg 6.1.1?

/home/axel# ffmpeg
ffmpeg version 6.1.1-0+deb12u1 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --prefix=/usr --extra-version=0+deb12u1 --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --arch=arm64 --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libglslang --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librist --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --disable-sndio --enable-libjxl --enable-pocketsphinx --enable-librsvg --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-libplacebo --enable-librav1e --enable-shared
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Hyper fast Audio and Video encoder

But unfortunately this also does not work.

naushir, 2024-02-26:
> If it helps in finding the bug: I use my own compiled Ffmpeg 6.1.1?

Unfotunately that's not going to work as the ``h264_v4l2m2m`` codec is only available on the Raspberry Pi fork at https://github.com/jc-kynesim/rpi-ffmpeg.  However, I'm not sure which branch we currently use for our releases.

naushir, 2024-02-26:
> Would it help if I compile this ffmpeg fork then?

It ought to help, we can then see if there have been any changes introduced that caused the breakages.  However, I don't know what branch in that repo we use for our releases.  @jc-kynesim might be able to help us when he is able to.

axeljerabek, 2024-02-26:
Now removed my ffmpeg, installed the apt one (5.1):
ffmpeg version 5.1.4-0+rpt2+deb12u1 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)

And the rpicam line:
rpicam-vid -t 0 --width 1920 --height 1080 --nopreview --exposure long --sharpness 1.1 --contrast 1.1 --brightness 0.2 --saturation 1.0 --awb auto --denoise auto --level 4.2 --codec libav --libav-audio --audio-source alsa --audio-device hw:2,0 --audio-channels 1 --audio-codec aac --audio-samplerate 48000 --audio-bitrate 128000 --av-sync 0 --libav-format flv -n --framerate 30 -b 5200000  --autofocus-mode manual --lens-position 0.25 --autofocus-window 0.25,0.25,0.5,0.5 --vflip --hflip --inline --save-pts timestamp.pts -o "rtmp://192.168.178.25/pi/test"

gives the following error again:

[75:35:09.302512040] [1304047]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[75:35:09.440337481] [1304053]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[75:35:09.444054115] [1304053]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media1 and ISP device /dev/media3
[75:35:09.444140354] [1304053]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Mode selection for 1920:1080:12:P(30)
    SRGGB10_CSI2P,1536x864/120.135 - Score: 2200
    SRGGB10_CSI2P,2304x1296/56.0255 - Score: 1150
    SRGGB10_CSI2P,4608x2592/14.3536 - Score: 33342.8
Stream configuration adjusted
[75:35:09.450227813] [1304047]  INFO Camera camera.cpp:1183 configuring streams: (0) 1920x1080-YUV420 (1) 2304x1296-SRGGB10_CSI2P
[75:35:09.450605584] [1304053]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SRGGB10_1X10 - Selected unicam format: 2304x1296-pRAA
[h264_v4l2m2m @ 0x558186fe60] Using device /dev/video11
[h264_v4l2m2m @ 0x558186fe60] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
[h264_v4l2m2m @ 0x558186fe60] requesting formats: output=YU12/drm_prime capture=H264/none
[h264_v4l2m2m @ 0x558186fe60] Encoder requires yuv420p pixel format.
ERROR: *** libav: unable to open video codec: -22 ***

Seems that we have to wait for  @jc-kynesim

:)

naushir, 2024-02-27:
@jc-kynesim, the following basic command seems to fail on a Pi 4:

```
rpicam-vid -t 5s -o test.mkv --codec libav
```

arsenicks, 2024-02-27:
> @jc-kynesim, the following basic command seems to fail on a Pi 4:
> 
> ```
> rpicam-vid -t 5s -o test.mkv --codec libav
> ```
> 
> and the relevant libav buffer setup can be found at https://github.com/raspberrypi/rpicam-apps/blob/4ffc10ed0c65e0be30935037f43391c8395a685d/encoder/libav_encoder.cpp#L399

Yup from what I saw everything that use the call - -codec libav will faill since the update.

robtow, 2024-02-27:
"Fix will be in the next ffmpeg (which should be fairly soon)"
Hours, days...?

I am seeing the same problem, on a Pi4 with a newly dowbloaded & installed 64bit Bookworm, and a just today delivered Raspberry Pi Camera Module 3.

arsenicks, 2024-03-02:
> "Fix will be in the next ffmpeg (which should be fairly soon)" Hours, days...?
> 
> I am seeing the same problem, on a Pi4 with a newly downloaded, installed, _**and upgraded**_ 64bit Bookworm, and a just today delivered Raspberry Pi Camera Module 3. I reflashed the sd card, and _skipped_ the (suggested) upgrade, and subsequently did not see the problem. So for now, I'm sticking with that installation.

You can click follow, custom and select release on the repo, I think this is the right one @jc-kynesim is that it ?  https://github.com/jc-kynesim/rpi-ffmpeg

axeljerabek, 2024-03-04:
There was just an update for ffmpeg and libav, still I have the same error. Probably have to wait for the rpicam-vid update?

naushir, 2024-03-04:
There is no rpicam-vid update needed for this fix.  Can you post the output of ``ffmpeg --version``?

naushir, 2024-03-04:
That seems to match my ffmpeg version which does work, although I don't know if there was a version bump with this fix.  Do you get exactly the same error message with ``rpicam-vid``?

axeljerabek, 2024-03-04:
Additional:

rpicam-vid: error while loading shared libraries: libavcodec.so.60: cannot open shared object file: No such file or directory

Although I cannot find a libavcodec60 package.

naushir, 2024-03-04:
Are you manually building ``rpicam-apps`` or did you get it from the apt packages?

naushir, 2024-03-04:
This feels like it's trying using your custom build of libav.  What does ``which rpicam-vid`` and ``ldd $(which rpicam-vid)`` output?

axeljerabek, 2024-03-04:
Here my outputs:

rpicam-vid:
rpicam-vid: error while loading shared libraries: libavcodec.so.60: cannot open shared object file: No such file or directory

ldd $(which rpicam-vid):
        linux-vdso.so.1 (0x0000007fa47da000)
        rpicam_app.so.1.4.3 => /usr/local/lib/aarch64-linux-gnu/rpicam_app.so.1.4.3 (0x0000007fa4650000)
        libcamera.so.0.2 => /usr/local/lib/aarch64-linux-gnu/libcamera.so.0.2 (0x0000007fa44e0000)
        libboost_program_options.so.1.74.0 => /lib/aarch64-linux-gnu/libboost_program_options.so.1.74.0 (0x0000007fa4460000)
        libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007fa4240000)
        libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007fa4200000)
        libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007fa4050000)
        libcamera-base.so.0.2 => /usr/local/lib/aarch64-linux-gnu/libcamera-base.so.0.2 (0x0000007fa4000000)
        libavcodec.so.60 => not found
        libavdevice.so.60 => not found
        libavformat.so.60 => not found
        libavutil.so.58 => not found
        libswresample.so.4 => /lib/aarch64-linux-gnu/libswresample.so.4 (0x0000007fa3fc0000)
        libexif.so.12 => /lib/aarch64-linux-gnu/libexif.so.12 (0x0000007fa3f50000)
        libjpeg.so.62 => /lib/aarch64-linux-gnu/libjpeg.so.62 (0x0000007fa3ed0000)
        libtiff.so.6 => /lib/aarch64-linux-gnu/libtiff.so.6 (0x0000007fa3e30000)
        libpng16.so.16 => /lib/aarch64-linux-gnu/libpng16.so.16 (0x0000007fa3dd0000)
        libdrm.so.2 => /lib/aarch64-linux-gnu/libdrm.so.2 (0x0000007fa3d90000)
        libX11.so.6 => /lib/aarch64-linux-gnu/libX11.so.6 (0x0000007fa3c30000)
        libepoxy.so.0 => /lib/aarch64-linux-gnu/libepoxy.so.0 (0x0000007fa3ad0000)
        libQt5Core.so.5 => /lib/aarch64-linux-gnu/libQt5Core.so.5 (0x0000007fa3560000)
        libQt5Widgets.so.5 => /lib/aarch64-linux-gnu/libQt5Widgets.so.5 (0x0000007fa2e70000)
        libQt5Gui.so.5 => /lib/aarch64-linux-gnu/libQt5Gui.so.5 (0x0000007fa27c0000)
        libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007fa2720000)
        libpisp.so.1 => /usr/local/lib/aarch64-linux-gnu/libpisp.so.1 (0x0000007fa2690000)
        libgnutls.so.30 => /lib/aarch64-linux-gnu/libgnutls.so.30 (0x0000007fa2440000)
        libudev.so.1 => /lib/aarch64-linux-gnu/libudev.so.1 (0x0000007fa23f0000)
        libyaml-0.so.2 => /lib/aarch64-linux-gnu/libyaml-0.so.2 (0x0000007fa23b0000)
        /lib/ld-linux-aarch64.so.1 (0x0000007fa479d000)
        libdw.so.1 => /lib/aarch64-linux-gnu/libdw.so.1 (0x0000007fa22f0000)
        libunwind.so.8 => /lib/aarch64-linux-gnu/libunwind.so.8 (0x0000007fa22a0000)
        libavutil.so.57 => /lib/aarch64-linux-gnu/libavutil.so.57 (0x0000007fa20c0000)
        libsoxr.so.0 => /lib/aarch64-linux-gnu/libsoxr.so.0 (0x0000007fa2050000)
        libwebp.so.7 => /lib/aarch64-linux-gnu/libwebp.so.7 (0x0000007fa1fd0000)
        libzstd.so.1 => /lib/aarch64-linux-gnu/libzstd.so.1 (0x0000007fa1f10000)
        liblzma.so.5 => /lib/aarch64-linux-gnu/liblzma.so.5 (0x0000007fa1ec0000)
        libLerc.so.4 => /lib/aarch64-linux-gnu/libLerc.so.4 (0x0000007fa1e10000)
        libjbig.so.0 => /lib/aarch64-linux-gnu/libjbig.so.0 (0x0000007fa1de0000)
        libdeflate.so.0 => /lib/aarch64-linux-gnu/libdeflate.so.0 (0x0000007fa1da0000)
        libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007fa1d60000)
        libxcb.so.1 => /lib/aarch64-linux-gnu/libxcb.so.1 (0x0000007fa1d10000)
        libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007fa1ce0000)
        libdouble-conversion.so.3 => /lib/aarch64-linux-gnu/libdouble-conversion.so.3 (0x0000007fa1ca0000)
        libicui18n.so.72 => /lib/aarch64-linux-gnu/libicui18n.so.72 (0x0000007fa1960000)
        libicuuc.so.72 => /lib/aarch64-linux-gnu/libicuuc.so.72 (0x0000007fa1740000)
        libpcre2-16.so.0 => /lib/aarch64-linux-gnu/libpcre2-16.so.0 (0x0000007fa16a0000)
        libglib-2.0.so.0 => /lib/aarch64-linux-gnu/libglib-2.0.so.0 (0x0000007fa1540000)
        libGL.so.1 => /lib/aarch64-linux-gnu/libGL.so.1 (0x0000007fa1440000)
        libharfbuzz.so.0 => /lib/aarch64-linux-gnu/libharfbuzz.so.0 (0x0000007fa1320000)
        libmd4c.so.0 => /lib/aarch64-linux-gnu/libmd4c.so.0 (0x0000007fa1300000)
        libp11-kit.so.0 => /lib/aarch64-linux-gnu/libp11-kit.so.0 (0x0000007fa11b0000)
        libidn2.so.0 => /lib/aarch64-linux-gnu/libidn2.so.0 (0x0000007fa1160000)
        libunistring.so.2 => /lib/aarch64-linux-gnu/libunistring.so.2 (0x0000007fa0f90000)
        libtasn1.so.6 => /lib/aarch64-linux-gnu/libtasn1.so.6 (0x0000007fa0f50000)
        libnettle.so.8 => /lib/aarch64-linux-gnu/libnettle.so.8 (0x0000007fa0ee0000)
        libhogweed.so.6 => /lib/aarch64-linux-gnu/libhogweed.so.6 (0x0000007fa0e70000)
        libgmp.so.10 => /lib/aarch64-linux-gnu/libgmp.so.10 (0x0000007fa0dd0000)
        libelf.so.1 => /lib/aarch64-linux-gnu/libelf.so.1 (0x0000007fa0d90000)
        libbz2.so.1.0 => /lib/aarch64-linux-gnu/libbz2.so.1.0 (0x0000007fa0d60000)
        libva-drm.so.2 => /lib/aarch64-linux-gnu/libva-drm.so.2 (0x0000007fa0d30000)
        libva.so.2 => /lib/aarch64-linux-gnu/libva.so.2 (0x0000007fa0ce0000)
        libva-x11.so.2 => /lib/aarch64-linux-gnu/libva-x11.so.2 (0x0000007fa0cb0000)
        libvdpau.so.1 => /lib/aarch64-linux-gnu/libvdpau.so.1 (0x0000007fa0c80000)
        libOpenCL.so.1 => /lib/aarch64-linux-gnu/libOpenCL.so.1 (0x0000007fa0c50000)
        libgomp.so.1 => /lib/aarch64-linux-gnu/libgomp.so.1 (0x0000007fa0be0000)
[... cut ...]
        libX11-xcb.so.1 => /lib/aarch64-linux-gnu/libX11-xcb.so.1 (0x0000007f9e920000)
        libxcb-dri3.so.0 => /lib/aarch64-linux-gnu/libxcb-dri3.so.0 (0x0000007f9e8f0000)
        libbsd.so.0 => /lib/aarch64-linux-gnu/libbsd.so.0 (0x0000007f9e8b0000)
        libbrotlidec.so.1 => /lib/aarch64-linux-gnu/libbrotlidec.so.1 (0x0000007f9e880000)
        libmd.so.0 => /lib/aarch64-linux-gnu/libmd.so.0 (0x0000007f9e850000)
        libbrotlicommon.so.1 => /lib/aarch64-linux-gnu/libbrotlicommon.so.1 (0x0000007f9e810000)

naushir, 2024-03-04:
> rpicam_app.so.1.4.3 => /usr/local/lib/aarch64-linux-gnu/rpicam_app.so.1.4.3 (0x0000007fa4650000)

This suggests you might be linking with your local install of ``rpicam-apps``

> libavcodec.so.60 => not found
> libavdevice.so.60 => not found
> libavformat.so.60 => not found
> libavutil.so.58 => not found

And this also shows it's trying to link to your local install of libav, which is now missing.

Try running ``/usr/bin/rpicam-vid`` and see if that works.  Alternatively, you might want to reinstall a fresh OS image and do the updates to avoid the system getting confused with local installs.

axeljerabek, 2024-03-04:
/usr/bin/rpicam-vid also gives me the same error. I cannot simply install a fresh OS, there is many other services up and running here. Will remove everything I find and try again :) Thanks for your help so far! I will post when I found it and/or it works :)

naushir, 2024-03-04:
Another option would be to first purge your local copy of libav libraries, then re-configure and install rpicam-apps.   This should hopefully pick up the apt packages for libav.

jwainwright87, 2024-03-04:
> There is no rpicam-vid update needed for this fix. Can you post the output of `ffmpeg --version`?

I downloaded a new install of 64-bit Bookworm yesterday and was expecting a rpicam-vid update

naushir, 2024-03-04:
> I downloaded a new install of 64-bit Bookworm yesterday and was expecting a rpicam-vid update

The update only got pushed this morning, so you need to do an update via apt to get the latest packages.

axeljerabek, 2024-03-04:
Ok, purged everything, reinstalled and reconfigured and it works like a charm again at around 8-9 percent of CPU. THanks for fixing this one!

naushir, 2024-03-04:
Good to hear it's back to working!  Resolving this now...

closed 2024-03-04 by naushir

arsenicks, 2024-03-04:
I confirm it's fixed here too. 

Thanks guy's!

jwainwright87, 2024-03-04:
Fixed here too on apt-get upgrade.

Thanks to all involved!