raspberrypi/rpicam-apps #645: ERROR: *** no camera available ***

MightyMaestro, 2024-02-20:
Purchased the ArduCam 64MP camera, and had it running on my old Raspberry Pi 3. Had some buffer issues for the highest resolution (64MP) so I loaded a new installation onto my Raspberry Pi 4 (64 bit Bookworm). Camera just will not work after following all of the instructions found here:
https://forum.arducam.com/t/full-walkthrough-setup-rpi4-take-64mp-photos-and-control-focus/4653

I've looked at some other posts on this issue, and none of them relate to my problem. Attached is the report.

[report.txt](https://github.com/raspberrypi/rpicam-apps/files/14341875/report.txt)

MightyMaestro, 2024-02-20:
Thank you for your feedback.

I did realise the dtoverlay was missing last night, but it did not change the outcome.

naushir, 2024-02-26:
Please get in touch with Arducam about support for this sensor.

closed 2024-02-26 by naushir