raspberrypi/rpicam-apps #646: How to

KansaiUser, 2024-02-20:
I have a similar situation as [this issue](https://github.com/raspberrypi/rpicam-apps/issues/23). 
In my case I would like to use rpicam-vid to record a video with a resolution of 1600x1300  (or in another case 3280x2462)

However due to the limitations of h264 it seems this is not possible. 

```
rpicam-vid -t 5000 --width 3280 --height 2462
ERROR: *** failed to start output streaming ***
```

I read that using mjpeg this is possible and yes when I tried

`rpicam-vid -t 5000 --width 3280 --height 2462 --codec mjpeg
`
I could see my desired video, Yeah!!

However the problem is when I tried to save it

`rpicam-vid -t 5000 --width 3280 --height 2462 --codec mjpeg -o  hello.mjpeg
`

I get a file `hello.mjpeg` but it is just a static picture, not a movie. 

FYI
I am doing this in two rpi 4
in one the version of rpicam-apss is 1.4.3-1  and I have to check on the other but perhaps it is a bit older. 

Is there a way to be able to record high definition videos with rpicam-vid?

Gordon999, 2024-02-20:
If you want to prove you have saved a mjpeg video file try

ffmpeg -i output.mjpeg -pix_fmt yuv420p -b:v 4000k -c:v libx264 output.mp4

to convert to mp4 and you can then view it

Gordon999, 2024-02-20:
record with... rpicam-vid -t 10000 --width 3200 --height 1300 --codec mjpeg -o output.mjpeg

convert to mp4 for viewing... ffmpeg -i output.mjpeg -pix_fmt yuv420p -b:v 4000k -c:v libx264 output.mp4

rpicam-vid -t 5000 --width 3280 --height 2462 --codec mjpeg -o  hello.mjpeg doesn't play well when converted...

sandyol55, 2024-02-20:
If the output file is identified as  a transport stream, it can be played in mpv Media Player. 

```
rpicam-vid -t 10000 --width 3200 --height 1300 --codec mjpeg -o output.ts
```
Although on my 1920x1080 screen I only see the centre section :-)

KansaiUser, 2024-02-21:
Thank you very much! I managed to do it. 
One problem was that VLC could not play it correctly so I kept seeing one frame only.  (as 6bya said )
To reproduce it I had to do (and not in the RPI)

`vlc --demux=avformat  file.mjpeg
`

Now only two issues remain:

1. It seems that the end of the mjpeg is broken because when I read it with opencv, it reads fine but when it finishes it shows [mjpeg @ 00000284baee9480] overread 8  and also the cap.get(cv2.CAP_PROP_FRAME_COUNT) gives a negative number (where cap is cap = cv2.VideoCapture(video_file))
2. Also I have noticed that when I do the above adding `--framerate 30` I get only 25 fps. So a 15 sec video lasts 17.5 secs or so

They are minor nuisances but still wondering why are they happening and if there is a way to correct them

naushir, 2024-02-27:
> One problem was that VLC could not play it correctly so I kept seeing one frame only. (as 6bya said )

VLC has problems during playback if streams don't have timestamps.  Try ``ffplay`` and it ought to work correctly.

> . It seems that the end of the mjpeg is broken because when I read it with opencv, it reads fine but when it finishes it shows [mjpeg @ 00000284baee9480] overread 8 and also the cap.get(cv2.CAP_PROP_FRAME_COUNT) gives a negative number (where cap is cap = cv2.VideoCapture(video_file))

If ``ffplay`` or ``ffprobe`` does not report any errors in the file, I would ignore this opencv warning(?) if it does not cause any other problems further on.

naushir, 2024-02-29:
I think this can be resolved now.  Please reopen if there are further questions.

closed 2024-02-29 by naushir