raspberrypi/rpicam-apps #647: Camera not Detected

Mr-Essid, 2024-02-20:
camera v2

--------------------------------------------------------------------------------
rpicam-apps Bug Report
--------------------------------------------------------------------------------
Date: 20-02-2024 (13:11:57)
Command: /usr/bin/camera-bug-report

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
raspberrypi

** cat /proc/cpuinfo **
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Revision	: c03115
Serial		: 1000000014b35a52
Model		: Raspberry Pi 4 Model B Rev 1.5

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/cmdline.txt **
DO NOT EDIT THIS FILE

The file you are looking for has moved to /boot/firmware/cmdline.txt

** cat /boot/config.txt **
DO NOT EDIT THIS FILE

The file you are looking for has moved to /boot/firmware/config.txt

--------------------------------------------------------------------------------
Logs
[... 814 lines cut ...]
--------------------------------------------------------------------------------
Versions
--------------------------------------------------------------------------------
** uname -a **
Linux raspberrypi 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

** cat /etc/os-release **
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

** vcgencmd version **
Oct 17 2023 15:40:00 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start_x)

** rpicam-hello --version **
rpicam-apps build: 6202c09b2bfa 16-02-2024 (17:15:39)
libcamera build: v0.2.0+46-075b54d5

UserNameUTF-8, 2024-02-26:
> > [ 7.119203] imx219 10-0010: failed to read chip id 219
> > [ 7.128172] imx219: probe of 10-0010 failed with error -5
> 
> This suggests there is a problem communicating with the sensor over I2C.  Can you try with a fresh Raspberry Pi OS install and reset the camera and the camera cables pelase?

I do that many times (cause I have No other solution but unfortunately no result)

naushir, 2024-02-26:
If you have another sensor to connect, it would be worth trying it out as well.

UserNameUTF-8, 2024-02-26:
> orth trying it out

No I haven't Unfortunately

naushir, 2024-02-27:
I'm afraid I don't have any more suggestions other than contacting your supplier and exchanging your camera.

naushir, 2024-02-29:
Marking as resolved, as it feels like a HW issue.

closed 2024-02-29 by naushir