raspberrypi/rpicam-apps #648: RGGB12 format for mono IMX290/462

runcyclexcski, 2024-02-22:
I use rpicam-raw to write from a mono IMX290/462 stream at 12 bit into 1-frame files (as shown below). rpicam-hello produces a good mono image, so I presume rpicam-raw gets the same data. I get ~1.3Mb files, the size makes sense 1280 x 720 x (12/8) = 1382400. I am having difficulty reading the raw files (cv2 and numpy), since raw Bayer format is new to me. I do not get why I get data written as SRGGB12_CSI2P (bayer) for a monochrome image. I apologize for the silly question, the issue is not with rpicam-raw, but with me.

On a side note: does the IMX290 mono chip (or Rpicam chips, like the IMX219) output data as bayer raw or is the formatting done by the board? Mono cameras/chips I've worked with before (Andor, Photometrics, Roper, JAI) streamed raw as plain big-endian raw, no additional processing was needed.

Thank you in advance!

$ rpicam-raw -t 400 --segment 1 -o test_raw_1/test%01d.raw
[0:18:38.253535558] [3053]  INFO Camera camera_manager.cpp:284
libcamera v0.1.0+118-563cd78e
[0:18:38.297841027] [3056]  WARN RPiSdn sdn.cpp:39 Using legacy SDN
tuning - please consider moving SDN inside rpi.denoise
[0:18:38.300073137] [3056]  INFO RPI vc4.cpp:444 Registered camera
/base/soc/i2c0mux/i2c@1/imx290@1a to Unicam device /dev/media0 and ISP
device /dev/media2
[0:18:38.300268888] [3056]  INFO RPI pipeline_base.cpp:1142 Using
configuration file
'/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Stream configuration adjusted
[0:18:38.301784583] [3053]  INFO Camera camera.cpp:1183 configuring
streams: (0) 640x480-YUV420 (1) 1280x720-SRGGB12_CSI2P
[0:18:38.303660189] [3056]  INFO RPI vc4.cpp:608 Sensor:
/base/soc/i2c0mux/i2c@1/imx290@1a - Selected sensor format:
1280x720-SRGGB12_1X12 - Selected unicam format: 1280x720-pRCC
Raw stream: 1280x720 stride 1920 format SRGGB12_CSI2P

runcyclexcski, 2024-02-22:
>>>dtoverlay=imx290,mono in config.txt.

I did add dtoverlay=imx290, per instructions from the supplier (the instructions did not mention "mono" though). W/o dtoverlay I could not connect to the camera at all. -hello gives a proper grayscale image. 

Alright, I will try mono and  the unpacked mode!

closed 2024-02-26 by naushir

6by9, 2024-02-26:
Just as a caveat: when the system hasn't been told that this is a mono sensor it will still be running AWB etc as if it were a colour sensor, and typically falling back into an error case when it can't determine a colour temperature. The results will generally end up being near identical, but there is the possibility of going weird.
We really ought to look at getting a mono tuning file for imx290, but as it isn't one of our sensors, priority is low.