raspberrypi/rpicam-apps #650: [BUG] Camera frontend has timed out

oliviacarino, 2024-02-23:
**Describe the bug**
Running `libcamera-hello --list-cameras` shows the following:
`Available cameras
-----------------
0 : imx708_noir [4608x2592 10-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]
`

When running `libcamera-hello` or even `libcamera-jpeg`, I get an error stating: 
`Dequeue timer of 5000000.00us has expired! ERROR: Camera frontend has timed out!`

**Bug report**
Please use the ``camera-bug-report`` tool to create a bug report, and upload it here.
- Command used for `bug.txt:` `camera-bug-report -t 5 -o bug.txt -c "rpicam-still -t 1000 -o test.jpg"`
[bug2.txt](https://github.com/raspberrypi/rpicam-apps/files/14381035/bug2.txt)

- Command used for `bug2.txt`: `camera-bug-report -o bug.txt -c "rpicam-vid -t 0 -o test.264"`
[bug.txt](https://github.com/raspberrypi/rpicam-apps/files/14381036/bug.txt)

**Additional context**
What I have tried so far to fix the issue:
**I have tried the following Pi's:**
Pi 3A+
Pi 4B

**I have tried the following cameras:**
Raspberry Pi Camera Module V3
Raspberry Pi Camera Module V3 NoIR

**I have tried the following OS systems using their latest, stable versions:**
Bullseye
Bookworm

I tried a combination of all of the above with each other and continued to get the same error that everyone here has mentioned. I tried setting my camera timeout to be 5s and even 10s. Sadly, this did not work. I have installed fresh images of each OS multiple times when testing with each Pi, ran `apt update` and `apt upgrade`, this did not fix the issues either. Both cameras had the same error on both Pis and both OS (_please note, I tried each OS and cameras in different combinations on each Pi_).

I even tried turning on Glamor for the Pi 3, changing the gpu-memory on the Pi 3 to 256, and both things did not help :/ so I ended up buying a Pi 4B thinking it was a memory issue, but alas, it was not.

Reseating each camera multiple times didn't work either.

oliviacarino, 2024-02-27:
- No other peripherals are attached.
- I am using the 200mm connector cable that came with the camera.
- I am simply running it on my desk. I'm not using any housing for the pi itself, it just sits directly on my desk.

naushir, 2024-02-29:
I'm afraid I don't have any further suggestions.  We've been trying to reproduce this ourselves, but have had boards running for > 24 hours without showing any timeouts.

naushir, 2024-03-06:
Unfortunately I'll have to close as not-reproducible.  Similar to https://github.com/raspberrypi/picamera2/issues/952.

closed 2024-03-06 by naushir

phantum29, 2024-09-12:
@oliviacarino Did you ever get this resolved? I've been dealing with this for a while and finally decided to search for `Camera frontend has timed out!` dmesg also shows 
```
[   46.352011] edt_ft5x06 10-0038: Unable to fetch data, error: -5
[   52.979231] edt_ft5x06 10-0038: Unable to fetch data, error: -5
[   54.013809] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[   54.013840] edt_ft5x06 10-0038: Unable to fetch data, error: -110
[   54.014202] edt_ft5x06 10-0038: Unable to fetch data, error: -5
```
Maybe has something to do with our 'long' ribbon cables? Pi CM4 - Pi Cam v3, Bullseye. Tried all of the same things you have as well.

oliviacarino, 2024-09-12:
@phantum29 nope, I just gave up lol. Ended up going the route of using an OpenMV Cam and ESP-01s I had lying around to do the same project. Super disappointed, but is what it is I suppose.

I actually tried a super short ribbon cable (I assumed the same thing about the length potentially being an issue). Still had no luck. Tried with multiple Pis as well.