raspberrypi/rpicam-apps #651: [BUG] --circular option not working for h264 on rpi5 with rpicam-vid

ddbaron, 2024-02-26:
**Describe the bug**
The `--circular` argument is ignored for h264 (and libav) on rpi5/bookworm with rpicam-vid.

I have tested with a command line similar to what I used successfully on my rpi4/buster:
`libcamera-vid --level 4.2 --circular 1 --inline --width 1440 --height 320 --framerate 193 --shutter 100 --gain 16.0 --brightness 0.0 --contrast 1.0 --saturation 0.0 --sharpness 1.0 --denoise cdn_off --save-pts /dev/shm/tst.pts -t 0 -o /dev/shm/output.h264 -n
`
And also with the command line from the documentation (see my results below):
`rpicam-vid -t 0 --keypress --inline --circular -o test.h264`

**Bug report**
`cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
`
`uname -a
Linux rpi5 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux`

[bug.txt](https://github.com/raspberrypi/rpicam-apps/files/14408361/bug.txt)

**Additional context**
Using the following:
`rpicam-vid -t 0 --keypress --inline --circular -o /dev/shm/test.h264`

The output file is immediately written when rpicam-vid starts (it should write a 0 byte file and on exit or signal dump the buffer). It just continues to fill up, here is a snapshot after running 20+ seconds:

`ddbaron@rpi5:~ $ ls -al /dev/shm
total 12320
....
-rw-r--r--  1 ddbaron ddbaron 12582912 Feb 26 11:37 test.h264
`

6by9, 2024-02-26:
Forum thread https://forums.raspberrypi.com/viewtopic.php?t=365356

If a subject has already been discussed on the forums, please include a link if you raise a Github issue. It saves going over the same ground again.

ddbaron, 2024-02-26:
I created that thread, and there's been no replies to me on it, and I had already posted a link to this bug report. I'm not understanding your feedback, please clarify.

naushir, 2024-02-27:
Pi 5 uses the libav encoder framework for video encoding.  I have not found a simple way of enabling the ``--circular`` option with libav, it definitely won't be possible with containers like mkv/mov.

I will look into it at some point in the future, but this will be a low priority item on my list.

ddbaron, 2024-02-27:
Thank you for investigating. I will focus now on creating a workaround as this is a low priority for you. I need this primarily for use with mp4, and I view this as a key feature of `libcamera-vid / rpicam-vid`. 

If you don't have any near-term plans to address the defect (or remove the feature from use with rpi5), would it make sense to update [https://www.raspberrypi.com/documentation/computers/camera_software.html#known-issues](url)? The docs are wrong, the test case doesn't work, the feature isn't usable on rpi5, it's not clear which codec's, if any, work with --circular. I tried mjpeg and yuv and as I recall one worked and the other didn't.

Thanks

naushir, 2024-02-28:
I'll add a note to the documentation as suggested.

ddbaron, 2025-01-08:
@naushir @6by9 Can you fix this broken feature for (rpi5/cm5) in the rpicam-apps? I understand it's nontrivial but worth doing; unless you believe the original reason for adding these features no longer exists?

naushir, 2025-01-08:
I'll try and have a cursory look at what can be done this week, but no definite timeline to get it working unfortunately.

naushir, 2025-04-01:
No plan to work on this in the near future.  For such functionality, I suggest switching to picamera2, which has support for this feature.

closed 2025-04-01 by naushir