raspberrypi/rpicam-apps #652: 16-bit bayer mode selection not working

schoolpost, 2024-02-29:
I'm using an IMX585 sensor with 16-bit bayer mode, whenever I try to explicitly force into unpacked 16-bit operation, via `--mode 3856:2180:16:U` I can see in the output logs that rpicam-apps uses the compressed format.

This happens across the board for a variety of rpicam-apps. 

```
pi@raspberrypi:~/rpicam-apps $ rpicam-raw --mode 3856:2180:16:U
[22:46:47.975298338] [128943]  INFO Camera camera_manager.cpp:284 libcamera v0.0.0+4131-7ca67c3c
[22:46:47.995299232] [128944]  INFO RPI pisp.cpp:662 libpisp version v1.0.1 5652744a74d7 21-01-2024 (21:12:27)
[22:46:47.996736919] [128944]  WARN CameraSensorProperties camera_sensor_properties.cpp:261 No static properties available for 'imx585'
[22:46:47.996755752] [128944]  WARN CameraSensorProperties camera_sensor_properties.cpp:263 Please consider updating the camera sensor properties database
[22:46:48.000375738] [128944]  INFO RPI pisp.cpp:1119 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx585@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
Mode selection for 3856:2180:16:U
    SRGGB12_CSI2P,1928x1090/0 - Score: 8036
    SRGGB12_CSI2P,3856x2180/0 - Score: 2000
    SRGGB16,1928x1090/0 - Score: 6036
    SRGGB16,3856x2180/0 - Score: 0
Stream configuration adjusted
[22:46:48.001628129] [128943]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUV420 (1) 3856x2180-RGGB16_PISP_COMP1
[22:46:48.001795925] [128944]  INFO RPI pisp.cpp:1403 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx585@1a - Selected sensor format: 3856x2180-SRGGB16_1x16 - Selected CFE format: 3856x2180-PC1R
[22:46:48.002024018] [128944]  INFO IPARPI ipa_base.cpp:555 Adjusting mode minimum line length from 7.41us to 10.15us because of HW constraints.
[22:46:48.010374417] [128948]  WARN IPARPI ipa_base.cpp:1054 Could not set SHARPNESS - no sharpen algorithm
[22:46:48.010598491] [128948]  WARN IPARPI pisp.cpp:504 Could not set NOISE_REDUCTION_MODE - no Denoise algorithm
```

You can see it is wrong: `Raw stream: 3856x2180 stride 3904 format RGGB16_PISP_COMP1`

How to get 16-bit bayer Raw stream that the CFE does not do anything with.

schoolpost, 2024-02-29:
Fixed by adding 16-bit here: https://github.com/raspberrypi/rpicam-apps/blob/4ffc10ed0c65e0be30935037f43391c8395a685d/core/rpicam_app.cpp#L75C1-L95C2

naushir, 2024-02-29:
Would you be able to create a PR and we can merge the fix onto the tree please?

naushir, 2024-05-14:
Closing due to inactivity.  Please reopen if you need more help with this.

closed 2024-05-14 by naushir