raspberrypi/rpicam-apps #654: [BUG] yocto latest build *** stack smashing detected ***: terminated

smzahraee, 2024-03-02:
**Describe the bug**
I compile the raspicam with newes yocto version (master) for rpi zero 2 w

```
rpicam-vid -t 0 -v 
```
```
root@raspberrypi0-2w-64:~# rpicam-vid -t 0 -v 
[0:21:42.107511429] [1082]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+dirty (2024-03-02T11:10:06+00:00)
[0:21:42.197865804] [1083]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:21:42.204617262] [1083]  INFO RPI vc4.cpp:401 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
Options:
    verbose: 2
    info_text:#%frame (%fps fps) exp %exp ag %ag dg %dg
    timeout: 0ms
    width: 640
    height: 480
    output: 
    post_process_file: 
    preview: default
    qt-preview: 0
    transform: identity
    roi: all
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
    denoise: auto
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: (libcamera)
    lores-width: 0
    lores-height: 0
    autofocus-range: normal
    autofocus-speed: normal
    autofocus-window: all
    hdr: off
    mode: unspecified
    viewfinder-mode: unspecified
    metadata: 
    metadata-format: json
    bitrate: 0kbps
    profile: 
    level:  
    intra: 0
    inline: 0
    save-pts: 
    codec: h264
    quality (for MJPEG): 50
    keypress: 0
    signal: 0
    initial: record
    split: 0
    segment: 0
    circular: 0
Preview window unavailable
Running without preview window
Opening camera...
Acquired camera /base/soc/i2c0mux/i2c@1/imx477@1a
Configuring video...
Mode selection for 640:480:12:P
    SRGGB10_CSI2P,1332x990/0 - Score: 1305.05
    SRGGB12_CSI2P,2028x1080/0 - Score: 701.167
    SRGGB12_CSI2P,2028x1520/0 - Score: 607.329
    SRGGB12_CSI2P,4056x3040/0 - Score: 1494.33
[0:21:42.209984242] [1082]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUV420 (1) 2028x1520-SBGGR12_CSI2P
[0:21:42.210829919] [1083]  INFO RPI vc4.cpp:559 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
Camera streams configured
Available controls:
    Sharpness : [0.000000..16.000000]
    AwbEnable : [false..true]
    Contrast : [0.000000..32.000000]
    Saturation : [0.000000..32.000000]
    Brightness : [-1.000000..1.000000]
    AeFlickerPeriod : [100..1000000]
    HdrMode : [0..4]
    ExposureValue : [-8.000000..8.000000]
    ColourGains : [0.000000..32.000000]
    StatsOutputEnable : [false..true]
    ScalerCrop : [(0, 0)/128x128..(0, 0)/4056x3040]
    ExposureTime : [60..0]
    AeEnable : [false..true]
    NoiseReductionMode : [0..4]
    AeConstraintMode : [0..3]
    FrameDurationLimits : [24994..674193371]
    AnalogueGain : [1.000000..22.260870]
    AeFlickerMode : [0..1]
    AwbMode : [0..7]
    AeMeteringMode : [0..3]
    AeExposureMode : [0..3]
Buffers allocated and mapped
Video setup complete
Opened H264Encoder on /dev/video11 as fd 32
Got 6 output buffers
Got 12 capture buffers
Codec streaming started
Requests created
*** stack smashing detected ***: terminated
Aborted (core dumped)
```

naushir, 2024-03-03:
Can you provide the exact circumstances for this to happen.  Perhaps run in gdb to see where it occurred in the code exactly.

Normal operation does not produce any sort of exceptions so it would be useful to know where the buffer overflow may have occurred.

smzahraee, 2024-03-03:
I found it is related to yocto, so  I preapared a path for meta-raspberrypi 
here is the link
[libcamera bug fix](https://github.com/agherzan/meta-raspberrypi/pull/1293)

naushir, 2024-03-03:
I'm sorry I may have missed it in the pr above, but  could you point me at the exact change in rpicam-apps that fixes the stack overflow?

smzahraee, 2024-03-03:
yes, on building rpicam-apps you need libpisp library on YOCTO (main branch) and it is not enabled by default.
so in that I added libpsip library to YOCTO and used libcamera library from[ raspberry github repository](https://github.com/raspberrypi/libcamera)
this solved my problem

naushir, 2024-03-03:
Please run this under gdb and post the call stack when the exception occurrs.

naushir, 2024-03-06:
Resolving as not reproducible.  If you have any further information on this, please feel free to reopen this issue.

closed 2024-03-06 by naushir