raspberrypi/rpicam-apps #655: [BUG] Speed up rpicam-still capture

nvtkaszpir, 2024-03-03:
**Describe the bug**
I'm trying to capture still image with maximum resolution as fast as possible.
I still see that the rpicam-still tries to detect and select resolutions and this takes extra time to process.
As a result, the image is written with a delay, and all processing takes about 7 seconds to complete.

I'd like to skip detection phase so that image capture is done faster, and if user selects wrong mode then error should occur. How can I make stream configuration faster?

**Bug report**
```shell
camera-bug-report -t 40 -o ~/Pictures/bug-report.txt -c "rpicam-still -n --mode 2592:1944:12:P --lores-width 0 --lores-height 0 --thumb none -o /home/pi/Pictures/rpicam-jpeg.jpg"
```
output 
[bug-report.txt](https://github.com/raspberrypi/rpicam-apps/files/14472151/bug-report.txt)

**Additional context**
When capturing image I can see output such as (I've added line numbers)
```text
00 time rpicam-still -n --mode 2592:1944:12:P --lores-width 0 --lores-height 0 --thumb none -o /home/pi/Pictures/rpicam-jpeg.jpg
01 [1:05:01.337730465] [25538]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
02 [1:05:01.579054015] [25541]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
03 [1:05:01.605889854] [25541]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
04 [1:05:01.608262839] [25541]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
05 Mode selection for 1296:972:12:P
06    SGBRG10_CSI2P,640x480/0 - Score: 3296
07     SGBRG10_CSI2P,1296x972/0 - Score: 1000
08     SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
09     SGBRG10_CSI2P,2592x1944/0 - Score: 1567
10 Stream configuration adjusted
11 [1:05:01.646533609] [25538]  INFO Camera camera.cpp:1183 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-SGBRG10_CSI2P
12 [1:05:01.651062582] [25541]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
13 Mode selection for 2592:1944:12:P
14     SGBRG10_CSI2P,640x480/0 - Score: 7832
15     SGBRG10_CSI2P,1296x972/0 - Score: 5536
16     SGBRG10_CSI2P,1920x1080/0 - Score: 4238.67
17     SGBRG10_CSI2P,2592x1944/0 - Score: 1000
18 Stream configuration adjusted
19 [1:05:06.862648274] [25538]  INFO Camera camera.cpp:1183 configuring streams: (0) 2592x1944-YUV420 (1) 2592x1944-SGBRG10_CSI2P
20 [1:05:06.882996152] [25541]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGBRG10_1X10 - Selected unicam format: 2592x1944-pGAA
21 Still capture image received
22
23 real	0m7.428s
24 user	0m4.223s
25 sys	0m1.374s

```

 I was expecting that when providing ` --mode 2592:1944:12:P` it raspicam-still should skip processing which generates output in lines from 05 to 18, but apparently it is not doing it. How can I make stream configuration faster?

nvtkaszpir, 2024-03-03:
Adding `--immediate` to the options solved my problem, now image is captured in just around 3 seconds.

Apparently even selecting `-n` (for  no preview) triggers some additional mode processing.

naushir, 2024-03-03:
Mode selection is not the thing that takes up time during a capture cycle. By far the largest amount of time is the convergence of the 3A algorithms.

If you can manually set gaim, exposure and white balance values, this will significantly improve capture latency.

naushir, 2024-03-04:
Is there anything more information you need?  If not, I'll close this down shortly.

nvtkaszpir, 2024-03-04:
I was expecting that passing `-n` flag (no preview) would automatically skip code to fetch images for preview, because why bother if we are not going to show it?

If you believe this is working as intended then feel free to close the issue, because `--immediate` flag does what I wanted.

naushir, 2024-03-04:
``-n`` still runs preview frames, just does not display anything in the preview window.  Preview frames are always necessary to converge the 3A algorithms, unless you switch to manual operation.

closed 2024-03-04 by naushir