raspberrypi/rpicam-apps #658: [BUG]RP4, Cam module 3. 'rpicam-hello' detect cam but 'vcgencmd get_camera' not detect cam

morusyn, 2024-03-06:
**Describe the bug**
'rpicam-hello' command in Terminal show video, run Ok.  'vcgencmd get_camera' do not detect cam module3.

**Bug report**
 Report created by ``camera-bug-report`` tool attached:
[bug-report.txt](https://github.com/raspberrypi/rpicam-apps/files/14513092/bug-report.txt)

```
 'vcgencmd get_camera' return: 
supported=0 detected=0, libcamera interfaces=0

```
'rpicam-hello list-cameras' return:
[0:33:59.267675647] [2957]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[0:33:59.318764766] [2960]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:33:59.320971837] [2960]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media2
[0:33:59.321045318] [2960]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Made X/EGL preview window
Mode selection for 2304:1296:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 3400
    SRGGB10_CSI2P,2304x1296/0 - Score: 1000
    SRGGB10_CSI2P,4608x2592/0 - Score: 1900
Stream configuration adjusted
[0:33:59.807952941] [2957]  INFO Camera camera.cpp:1183 configuring streams: (0) 2304x1296-YUV420 (1) 2304x1296-SBGGR10_CSI2P
[0:33:59.808412459] [2960]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA

```
**Additional context**
/boot/firmware/config.txt:

# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Camera Module 3 (IMX708)
dtoverlay=imx708

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

enable_uart=1

naushir, 2024-03-06:
vcgencmd get_camera is deprecated and not used any more. There is no use for it with the libcamera stack.

closed 2024-03-06 by naushir

6by9, 2024-03-06:
Covered in https://forums.raspberrypi.com/viewtopic.php?t=362707