raspberrypi/rpicam-apps #664: [BUG] rpicam-vid hangs

the0, 2024-03-22:
**Describe the bug**
After some time of normal functioning rpicam-vid just hangs without producing any video data.

**Bug report**
[bug-report.txt](https://github.com/raspberrypi/rpicam-apps/files/14728808/bug-report.txt)
[debug.log](https://github.com/raspberrypi/rpicam-apps/files/14728838/debug.log)

**Additional context**
Same happens with two other RPi0 and two RPi4, all fully updated. Tried both 32 and 64 bit OS/package versions, different ov5647 modules. I noticed that problem occurs more frequent when low bitrates are used, tried 10k-256k range.

naushir, 2024-03-25:
I can reporduce this as well with very low bitrate (< 100kbps) encoding.  There does seem to be an error issued in the firmware logs:
```
049512.740: venc: venc_rc: frame 1 aborted bits 18168 > 11457 (HRD), QP=45
049552.856: venc: venc_rc: frame 2 aborted bits 18376 > 14582 (HRD), QP=45
052010.765: venc: venc_rc: frame 61 aborted bits 36320 > 22778 (HRD), QP=44
064478.152: venc: venc_rc: frame 360 aborted bits 28184 > 22042 (HRD), QP=51
064519.923: venc: venc_rc: frame 361 aborted bits 28328 > 25167 (HRD), QP=51
069559.587: venc: venc_rc: frame 422 aborted bits 46216 > 41525 (HRD), QP=46
```

Interestingly, once this error message is repeated 6 times, the ``rpicam-vid`` process stalls.  This corresponds to the 6 output buffers that ``rpicam-vid`` allocates for libcamera.  It seems like when this error message hits, the framebuffer does not get returned back to the kernel/userland and is effectively lost.  Once we lose all 6 buffers, there are no buffers that can be recycled back into libcamera, hence the stall.

@6by9 any thoughts on this?

naushir, 2024-03-25:
One possible workaround is to use the libav backend for HW encoding.  If I run something like

```bash
rpicam-vid -v -t 0 --framerate 24 --width 720 --height 576 --bitrate 50kbps -n --denoise off -o /dev/null --codec libav --libav-format h264
```

I still see the venc error log messages, but this does not lockup ``rpicam-vid``.

naushir, 2024-03-25:
It does seem to be related to how rpicam-apps drives the encoder, as the very same driver used via the libav backend seems to behave correctly.

In my very brief debugging, it looks like calling ``poll()`` on the driver after ``VIDIOC_QBUF`` constantly returns 0, so we never attempt a ``DQBUF`` and things just grind to a halt.

naushir, 2024-03-27:
@the0, did you try my suggestion of using ``--codec libav --libav-format h264`` and see if it fixes your problem?

the0, 2024-03-27:
> @the0, did you try my suggestion of using `--codec libav --libav-format h264` and see if it fixes your problem?

Yes, I'm using "--codec libav --libav-video-codec h264_v4l2m2m --libav-format h264" without any problems now, thank you. However, I didn't manage to make "--libav-video-codec-opts 'preset=ultrafast;tune=zerolatency'" work. Neither "v4l2-ctl --list-ctrls-menu -d 11" nor "ffmpeg -h encoder=h264_v4l2m2m" mention any of these opts are available, is that correct?

naushir, 2024-03-27:
> However, I didn't manage to make "--libav-video-codec-opts 'preset=ultrafast;tune=zerolatency'" work. Neither "v4l2-ctl --list-ctrls-menu -d 11" nor "ffmpeg -h encoder=h264_v4l2m2m" mention any of these opts are available, is that correct?

These options are only for the SW (libx264) codec.  The hardware codec will simply ignore these settings.

the0, 2024-03-28:
Is framerate option also being ignored by libav/h264_v4l2m2m?
```
$ rpicam-vid -t 0 --codec h264 --framerate 24 --flush -o - 2>/dev/null | ffprobe -hide_banner -i -
Input #0, h264, from 'pipe:':
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: h264 (High), yuv420p(progressive), 640x480, 24 fps, 24 tbr, 1200k tbn
$ rpicam-vid -t 0 --codec libav --libav-video-codec h264_v4l2m2m --libav-format h264 --framerate 24 --flush -o - 2>/dev/null | ffprobe -hide_banner -i -
Input #0, h264, from 'pipe:':
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: h264 (High), yuv420p(progressive), 640x480, 30 fps, 30 tbr, 1200k tbn
```

naushir, 2024-03-28:
Framerate should be correct as far as the sensor will generate frames at the requested rate.  However, elementary H.264 streams don't store timestamps, so have no real notion of farmerate in the file itself.

6by9, 2024-03-28:
Actually this looks like an FFmpeg issue, or possibly the interface to it/libav.

There is the `timing_info_present_flag` in the SPS headers which should represent the encoded framerate. Through V4L2 that is set via the VIDIOC_S_PARM ioctl.

In the native rpicam-vid case, that is being set to the requested 24fps. In the FFmpeg case it is being left at the default of 30fps.
libavcodec/v4l2_m2m_enc.c appears to be calling VIDIOC_S_PARM appropriately, so is libav being given the correct frame rate?

6by9, 2024-03-28:
I've not tested this, however:

v4l2_m2m_enc is taking the frame rate from AVCodecContext * framerate.[num|den] 
https://github.com/FFmpeg/FFmpeg/blob/master/libavcodec/v4l2_m2m_enc.c#L192-L193

libav_encoder.cpp sets a load of other parameters in AVCodecContext, but not that one 
https://github.com/raspberrypi/rpicam-apps/blob/main/encoder/libav_encoder.cpp#L119-L124

naushir, 2024-03-28:
That does get setup here:
https://github.com/raspberrypi/rpicam-apps/blob/70894c0c46855955442284aba9dedb66d8d4bfbf/encoder/libav_encoder.cpp#L26

6by9, 2024-03-28:
Missed that one.
And indeed we do get `timeperframe=1000/24000` passed in. Amusingly we get `timeperframe=3750/90000` from the native implementation.

It shouldn't matter, but FFmpeg calls S_PARM after REQBUFS(vid-out-mplane) I think we're missing a commit in the V4L2 driver that pushes that setting down to the firmware.

naushir, 2024-04-02:
If I use the ``libx264`` backend through ``libav_encoder``, the fps does report correctly.  So it does indeed suggest that the V4L2 driver might be missing something.

the0, 2024-04-12:
@naushir are there any workarounds to have hardware-accelerated 720x576@20 h264 capture at the moment?

the0, 2024-04-15:
I'm unable to capture at rates lower than 30 fps with libav/h264_v4l2m2m as discussed above.

naushir, 2024-04-15:
The framerate parameter will ensure the sensor frames at the correct rate.  You can check this with the ``--save-pts`` command line argument.  The ffmpeg info output will be wrong, and it may affect rate control, but the frames will be encoded at the right rate.

the0, 2024-04-16:
With libav/h264_v4l2m2m I'm getting empty pts file, but when I set framerate to 1 fps I saw it actually encodes at 1 fps. Thanks!

6by9, 2024-04-17:
https://github.com/raspberrypi/linux/pull/6108 should fix libav not resulting with the correct frame rate on the stream.

libav/h264_v4l2m2m doesn't support the --save-pts flag.
Save to an MP4 or MKV file, and use `ffprobe -show_packets -i foo.mp4 | grep pts_time` to get the PTS of all the packets.

naushir, 2024-04-19:
Closing this now, fixed by using the libav workaround.

closed 2024-04-19 by naushir