raspberrypi/rpicam-apps #668: [BUG] DRM preview unavailable on Raspberry Pi 1 Model B

jokubasver, 2024-04-13:
**Describe the bug**
DRM preview is broken on Raspberry Pi 1 Model B rev 1 (256MB). Tested with both OV5647 and IMX708 cameras. 

Cameras have been tested and verified to fully work with DRM preview on Raspberry Pi 4, so it is not a camera hardware issue.

Tried with no success:

- Tried all the versions of Raspberry Pi OS: Legacy/Bookworm, Full/Lite. Also tested wtih DietPi, same issue.
- Changing to  dtoverlay=vc4-kms-v4d, dtoverlay=vc4-fkms-v4d, or commenting this line out completely.
- Changing the display driver in raspi-config. 
- Enabling Glamor.
- Changing RAM configuration.
- Increasing CMA.
- Changing from fbturbo to fbdev in /usr/share/X11/xorg.conf.d/99-fbturbo.conf

**Bug report**
Please see attached bug.txt
[bug.txt](https://github.com/raspberrypi/rpicam-apps/files/14965643/bug.txt)

jokubasver, 2024-04-14:
I have now tested this on a Raspberry Pi 1 Model B (512MB) and DRM preview is working.

The 256MB Pi, when running kmsprint, says that no DRM card is detected. Makes me think if there is not enough RAM available, or if this Pi has a damaged GPU core...

6by9, 2024-04-14:
The 256MB Pis are severely constrained on memory.
From your bug.txt, 
```
gpu_mem_256=96
```
So you've assigned 96M of your 256 to the firmware. There is very little need to alter it from the default.

vc4-fkms-v3d and vc4-kms-v3d default to trying to assign 256MB to the CMA heap. You can change that via the `cma-128` or `cma-96` overrides (ie `dtoverlay=vc4-kms-v3d,cma-128`).
Memory says that the firmware will refuse to assign more memory to the CMA heap than is possible, as the kernel won't boot in that situation.

jokubasver, 2024-04-14:
Even with the default 64MB configuration, and cma-128, DRM preview still does not work

6by9, 2024-04-15:
Having dug out my model B, `sudo vclog -m` includes
```
002515.573: Overlay 'vc4-kms-v3d' needs at least 512MB
```

I've checked the firmware and it has a hardcoded limit that enforces that vc4-kms-v3d and vc4-fkms-v3d will only load on systems with 512MB or more.

You can rename the overlay to bypass this - I have my 256MB system running rpicam-hello -t 0 via DRM, but CPU usage is sitting around 84%.
Do ensure you restrict gpu_mem and cma otherwise you risk your system not booting at all.

jokubasver, 2024-04-15:
Thanks,

it would be good if documentation would reflect this, as currently one could assume that the 256MB Pi would work out of the box with rpicam-apps/libcamera preview with (f)kms.

Another option would be to introduce some workaround switch for config.txt to specifically bypass this restriction and set the RAM/CMA accordingly.

naushir, 2024-04-19:
Resolving now...

closed 2024-04-19 by naushir