raspberrypi/rpicam-apps #670: [BUG]

troublestarter, 2024-04-18:
**Describe the bug**
Hi

Sometime, in the dmesg i have some call trace linked to libcamera.
I use it in the allskyteam project from github to have an allsky camera.

I will try to add the report tool

**Additional context**

Here the dmesg error:

~~~
[Sat Apr  6 06:46:40 2024] libcamera-still: page allocation failure: order:4, mode:0x40cc0(GFP_KERNEL|__GFP_COMP), nodemask=(null),cpuset=/,mems_allowed=0
[Sat Apr  6 06:46:40 2024] CPU: 1 PID: 6937 Comm: libcamera-still Tainted: G         C         6.1.21-v8+ #1642
[Sat Apr  6 06:46:40 2024] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[Sat Apr  6 06:46:40 2024] Call trace:
[Sat Apr  6 06:46:40 2024]  dump_backtrace+0x120/0x130
[Sat Apr  6 06:46:40 2024]  show_stack+0x20/0x30
[Sat Apr  6 06:46:40 2024]  dump_stack_lvl+0x8c/0xb8
[Sat Apr  6 06:46:40 2024]  dump_stack+0x18/0x34
[Sat Apr  6 06:46:40 2024]  warn_alloc+0x114/0x1a8
[Sat Apr  6 06:46:40 2024]  __alloc_pages_slowpath.constprop.158+0xb70/0xbc0
[Sat Apr  6 06:46:40 2024]  __alloc_pages+0x2a8/0x318
[Sat Apr  6 06:46:40 2024]  __kmalloc_large_node+0xa0/0x180
[Sat Apr  6 06:46:40 2024]  __kmalloc+0x138/0x1e8
[Sat Apr  6 06:46:40 2024]  dma_common_contiguous_remap+0x50/0xd8
[Sat Apr  6 06:46:40 2024]  dma_direct_alloc+0x168/0x318
[Sat Apr  6 06:46:40 2024]  dma_alloc_attrs+0xac/0xc0
[Sat Apr  6 06:46:40 2024]  vb2_dc_alloc+0xa8/0x180 [videobuf2_dma_contig]
[Sat Apr  6 06:46:40 2024]  __vb2_queue_alloc+0x238/0x4a0 [videobuf2_common]
[Sat Apr  6 06:46:40 2024]  vb2_core_reqbufs+0x280/0x4e8 [videobuf2_common]
[Sat Apr  6 06:46:40 2024]  vb2_ioctl_reqbufs+0xc0/0x100 [videobuf2_v4l2]
[Sat Apr  6 06:46:40 2024]  v4l_reqbufs+0x58/0x70 [videodev]
[Sat Apr  6 06:46:40 2024]  __video_do_ioctl+0x184/0x3d8 [videodev]
[Sat Apr  6 06:46:40 2024]  video_usercopy+0x2d8/0x740 [videodev]
[Sat Apr  6 06:46:40 2024]  video_ioctl2+0x20/0x38 [videodev]
[Sat Apr  6 06:46:40 2024]  v4l2_ioctl+0x48/0x68 [videodev]
[Sat Apr  6 06:46:40 2024]  v4l2_compat_ioctl32+0x98/0xc0 [videodev]
[Sat Apr  6 06:46:40 2024]  __arm64_compat_sys_ioctl+0x168/0x180
[Sat Apr  6 06:46:40 2024]  invoke_syscall+0x4c/0x110
[Sat Apr  6 06:46:40 2024]  el0_svc_common.constprop.3+0xfc/0x120
[Sat Apr  6 06:46:40 2024]  do_el0_svc_compat+0x24/0x48
[Sat Apr  6 06:46:40 2024]  el0_svc_compat+0x30/0x88
[Sat Apr  6 06:46:40 2024]  el0t_32_sync_handler+0xe4/0x100
[Sat Apr  6 06:46:40 2024]  el0t_32_sync+0x190/0x194
[Sat Apr  6 06:46:40 2024] Mem-Info:
[Sat Apr  6 06:46:40 2024] active_anon:323050 inactive_anon:850 isolated_anon:0
                            active_file:174386 inactive_file:382640 isolated_file:0
                            unevictable:11 dirty:165 writeback:0
                            slab_reclaimable:27201 slab_unreclaimable:7793
                            mapped:92465 shmem:1499 pagetables:1959
                            sec_pagetables:0 bounce:0
                            kernel_misc_reclaimable:0
                            free:14661 free_pcp:0 free_cma:2449
[Sat Apr  6 06:46:40 2024] Node 0 active_anon:1292200kB inactive_anon:3400kB active_file:697544kB inactive_file:1530560kB unevictable:44kB isolated(anon):0kB isolated(file):0kB mapped:369860kB dirty:660kB writeback:0kB shmem:5996kB writeback_tmp:0kB kernel_stack:5392kB pagetables:7836kB sec_pagetables:0kB all_unreclaimable? no
[Sat Apr  6 06:46:40 2024] DMA free:23680kB boost:0kB min:3560kB low:4448kB high:5336kB reserved_highatomic:0KB active_anon:174300kB inactive_anon:8kB active_file:32844kB inactive_file:503592kB unevictable:0kB writepending:4kB present:917504kB managed:833168kB mlocked:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:9796kB
[Sat Apr  6 06:46:40 2024] lowmem_reserve[]: 0 2930 2930 2930
[Sat Apr  6 06:46:40 2024] DMA32 free:34964kB boost:28840kB min:41660kB low:44864kB high:48068kB reserved_highatomic:0KB active_anon:1117828kB inactive_anon:3392kB active_file:664700kB inactive_file:1027220kB unevictable:44kB writepending:656kB present:3080192kB managed:3001124kB mlocked:44kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
[Sat Apr  6 06:46:40 2024] lowmem_reserve[]: 0 0 0 0
[Sat Apr  6 06:46:40 2024] DMA: 155*4kB (UE) 69*8kB (U) 151*16kB (UEC) 108*32kB (UE) 41*64kB (UEC) 19*128kB (UEC) 2*256kB (U) 2*512kB (UC) 2*1024kB (UC) 2*2048kB (C) 1*4096kB (C) = 23876kB
[Sat Apr  6 06:46:40 2024] DMA32: 3027*4kB (UME) 1631*8kB (UME) 294*16kB (UME) 133*32kB (UME) 19*64kB (UME) 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 35332kB
[Sat Apr  6 06:46:40 2024] 558525 total pagecache pages
[Sat Apr  6 06:46:40 2024] 0 pages in swap cache
[Sat Apr  6 06:46:40 2024] Free swap  = 1048572kB
[Sat Apr  6 06:46:40 2024] Total swap = 1048572kB
[Sat Apr  6 06:46:40 2024] 999424 pages RAM
[Sat Apr  6 06:46:40 2024] 0 pages HighMem/MovableOnly
[Sat Apr  6 06:46:40 2024] 40851 pages reserved
[Sat Apr  6 06:46:40 2024] 131072 pages cma reserved
[... 35 lines cut ...]
[Tue Apr 16 06:23:42 2024] active_anon:297420 inactive_anon:758 isolated_anon:0
                            active_file:174739 inactive_file:408077 isolated_file:0
                            unevictable:11 dirty:41 writeback:0
                            slab_reclaimable:29520 slab_unreclaimable:7902
                            mapped:99429 shmem:1408 pagetables:1836
                            sec_pagetables:0 bounce:0
                            kernel_misc_reclaimable:0
                            free:12633 free_pcp:68 free_cma:619
[Tue Apr 16 06:23:42 2024] Node 0 active_anon:1189680kB inactive_anon:3032kB active_file:698956kB inactive_file:1632308kB unevictable:44kB isolated(anon):0kB isolated(file):0kB mapped:397716kB dirty:164kB writeback:0kB shmem:5632kB writeback_tmp:0kB kernel_stack:3792kB pagetables:7344kB sec_pagetables:0kB all_unreclaimable? no
[Tue Apr 16 06:23:42 2024] DMA free:17060kB boost:0kB min:3560kB low:4448kB high:5336kB reserved_highatomic:0KB active_anon:131084kB inactive_anon:224kB active_file:25312kB inactive_file:565916kB unevictable:0kB writepending:0kB present:917504kB managed:833168kB mlocked:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:2476kB
[Tue Apr 16 06:23:42 2024] lowmem_reserve[]: 0 2930 2930 2930
[Tue Apr 16 06:23:42 2024] DMA32 free:33472kB boost:20480kB min:33300kB low:36504kB high:39708kB reserved_highatomic:0KB active_anon:1058568kB inactive_anon:2808kB active_file:673644kB inactive_file:1066572kB unevictable:44kB writepending:164kB present:3080192kB managed:3001124kB mlocked:44kB bounce:0kB free_pcp:528kB local_pcp:0kB free_cma:0kB
[Tue Apr 16 06:23:42 2024] lowmem_reserve[]: 0 0 0 0
[Tue Apr 16 06:23:42 2024] DMA: 81*4kB (UE) 199*8kB (UEC) 225*16kB (UE) 93*32kB (UE) 38*64kB (UE) 8*128kB (UE) 2*256kB (U) 3*512kB (UC) 1*1024kB (U) 1*2048kB (C) 0*4096kB = 17068kB
[Tue Apr 16 06:23:42 2024] DMA32: 3444*4kB (UME) 648*8kB (UME) 337*16kB (UME) 256*32kB (UME) 15*64kB (UME) 2*128kB (M) 1*256kB (M) 0*512kB 0*1024kB 0*2048kB 0*4096kB = 34016kB
[Tue Apr 16 06:23:42 2024] 584150 total pagecache pages
[Tue Apr 16 06:23:42 2024] 0 pages in swap cache
[Tue Apr 16 06:23:42 2024] Free swap  = 1048572kB
[Tue Apr 16 06:23:42 2024] Total swap = 1048572kB
[Tue Apr 16 06:23:42 2024] 999424 pages RAM
[Tue Apr 16 06:23:42 2024] 0 pages HighMem/MovableOnly
[Tue Apr 16 06:23:42 2024] 40851 pages reserved
[Tue Apr 16 06:23:42 2024] 131072 pages cma reserved
[Tue Apr 16 06:23:42 2024] unicam fe801000.csi: dma alloc of size 18583552 failed
~~~

6by9, 2024-04-18:
You had insufficient contiguous memory to allocate the buffer.

troublestarter, 2024-04-18:
Hi @6by9 

Thanks you for your reply. Yes can be the origin.
How many minimal memory do i ensure to be free to run without this kind of message ?

6by9, 2024-04-18:
Based on `131072 pages cma reserved` you have 512MB of CMA heap reserved (a page is 4kB by default), and `999424 pages RAM` would be 4GB. CMA has to be in the bottom 1GB.

Admittedly other things come out of the CMA heap (display related allocations, and USB buffers for example), but 512MB is masses for most use cases. The default is 256MB.

The question is what is Allsky is doing to manage to keep buffers around for longer than normal.
You can try `sudo cat /sys/kernel/debug/dma_buf/bufinfo` to see what DMA bufs are allocated. That's not infallible as allocations don't  have to be exported as a DMA buf, but most libcamera related buffers are.

EricClaeys, 2024-04-18:
@6by9, Allsky uses a C++ program that calls `libcamera-still` to take a picture then sleeps for the user-requested delay and takes another picture.  A bash script optionally crops and resizes the image using `convert`, and optionally uploads it to a remote server for display on the Internet.
We don't do anything with buffers.  Most Pi's running Allsky have a 75 MB memory-based filesystem that images are written to and modified in (to decrease writes to the SD card).  The processed images are saved to the SD card.  Most Pi's also have from 1 to 4 GB of swap space depending on the RAM installed, so the total swap+RAM is 5 GB.

naushir, 2024-04-19:
@EricClaeys / @troublestarter what exact command line arguments do you use with ``libcamera-still``.  Have you tried playing with the ``buffer-count`` and ``viewfinder-buffer-count`` arguments?

naushir, 2024-05-14:
Closing due to inactivity.  Please reopen if you need more help with this.

closed 2024-05-14 by naushir

hemna, 2025-07-21:
I'm seeing this issue as well with allsky