raspberrypi/rpicam-apps #671: Cannot get frame from Camera Module 3 with RPI5 on Ubuntu 23 OS

bilkosem, 2024-04-19:
# Description
I cannot get image fom Camera Module 3 with RPI5 and Ubuntu 23. The cameras are detected by libcamera however when I tried to get a image I got the error: "Failed to start streaming: Invalid argument". As you can also see from bug report: I build libcamera(from rpi fork) and rpicam-apps from source. 

```
wfa@wfa-desktop:~$ sudo camera-bug-report -t 5 -o libcamera-jpeg.txt -c "libcamera-jpeg --nopreview -o output.jpeg"
[0:08:09.766704749] [1587]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+47-46b868fe
[0:08:09.798499531] [1588]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 19-04-2024 (13:29:33)
[0:08:09.947229568] [1588]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:08:09.947451772] [1588]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 19-04-2024 (13:29:33)
[0:08:10.070648825] [1588]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media3 and ISP device /dev/media1 using PiSP variant BCM2712_C0
Mode selection for 2304:1296:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 3400
    SRGGB10_CSI2P,2304x1296/0 - Score: 1000
    SRGGB10_CSI2P,4608x2592/0 - Score: 1900
Stream configuration adjusted
[0:08:10.072766807] [1587]  INFO Camera camera.cpp:1183 configuring streams: (0) 2304x1296-YUV420 (1) 2304x1296-BGGR16_PISP_COMP1
[0:08:10.073244436] [1588]  INFO RPI pisp.cpp:1405 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
[0:08:10.109305664] [1588] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video12[17:cap]: Failed to start streaming: Invalid argument
ERROR: *** failed to start camera ***
```

# Bug Report
[libcamera-jpeg.txt](https://github.com/raspberrypi/rpicam-apps/files/15040891/libcamera-jpeg.txt)

I see that my /boot/config.txt is not found since it is on /boot/firmware/config.txt, so I upload it here as well:
[config.txt](https://github.com/raspberrypi/rpicam-apps/files/15040907/config.txt)

naushir, 2024-04-19:
This suggest that your kernel is out-of-date.  You will likely need to compile our latest kernel and put it into your system.

bilkosem, 2024-04-22:
Sorry if this is a dummy question but, does it mean that I should use RaspberryPi OS (Bookworm) or can I do this update on Ubuntu23 as well (using raspberrypi/linux repo) to make the cameras work?

naushir, 2024-04-22:
I believe Canonical package their own kernel for Ubuntu releases.  To be honest, I don't know much about it.  I would recommend using a fresh Raspberry Pi OS image if you want an immediate no-hassle way to get the Camera V3 working.

bilkosem, 2024-04-30:
I will switch to Bookworm and close theis ticket then. For anyone in who reads it in future, I worked on it around a week but could not get it working, I would be happy if you have a solution.

closed 2024-04-30 by bilkosem