raspberrypi/rpicam-apps #672: [BUG] ERROR: Unable to find an appropriate H.264 codec

d13g4, 2024-04-19:
**Description**

Running rpicam-vid on the raspberry pi 5.
Shows what the name of the bug sais. Cannot find  H.264 codec.
Naturally it cannot, as there is none in hardware anymore.
**Terminal-Output**

$ rpicam-vid 
[1:30:37.009528996] [19787]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[1:30:37.021046221] [19788]  INFO RPI pisp.cpp:695 libpisp version v1.0.5 999da5acb4f4 19-04-2024 (11:25:29)
[1:30:37.035803755] [19788]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/ov5647@36 to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
Preview window unavailable
[1:30:37.039127287] [19787]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
Mode selection for 640:480:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 1000
    SGBRG10_CSI2P,1296x972/0 - Score: 1287
    SGBRG10_CSI2P,1920x1080/0 - Score: 1636.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1854
Stream configuration adjusted
[1:30:37.039458046] [19787]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUV420 (1) 640x480-GBRG_PISP_COMP1
[1:30:37.039628305] [19788]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected CFE format: 640x480-PC1g
ERROR: *** Unable to find an appropriate H.264 codec ***

**Additional context**

raspberry pi 5 with raspberry pi os.
everything up to date.
compiled rpicam-apps and ffmpeg from git because i was hoping the bleeding-edge would work.

If additional context or information are needed please ask, but i am unavailable this weekend.
I tried quite a lot (including making the hardware-encoder work for one day until i read the fine print and just thought "wtf why?!") 
i suspect its this tool, maybe the underlying v4l2 but havent looked into the code yet and i am not a very good programmer.

P.S.: Filed as bug because not finding a codec when there is none isnt an error; not noticing there cant be a h264 codec is a bug.

d13g4, 2024-04-22:
It works, thanks and sorry for the fuzz!

closed 2024-04-22 by d13g4

quadcom, 2024-05-14:
I'm still stuck with the same error.

I'm using a sample app that appears to be a basic RPI project (https://raspberrytips.com/how-to-live-stream-pi-camera/).

I tried installing the libavcodec-dev with RPIOSLite, but this still does not work. All I'm really trying to do is find a better-quality solution to monitor 3D prints from a Bambu P1S, and the RPI cameras are small enough to mount inside the chassis.

I originally tried to get .mediamtx up and running and ran into issues there. Which led me to this to test whether or not the camera was actually working.

```
root@picam:/home/picam# libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx708_wide [4608x2592 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]

```
```

root@picam:/home/picam# rpicam-vid -t 2 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8000/}' :demux=h264
bash: cvlc: command not found
[23:09:11.969478681] [28275]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[23:09:11.980196409] [28279]  INFO RPI pisp.cpp:695 libpisp version v1.0.5 999da5acb4f4 17-04-2024 (14:29:29)
[23:09:12.118771087] [28279]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
Preview window unavailable
[23:09:12.121663060] [28275]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
Mode selection for 640:480:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 1486.67
    SRGGB10_CSI2P,2304x1296/0 - Score: 1786.67
    SRGGB10_CSI2P,4608x2592/0 - Score: 2686.67
Stream configuration adjusted
[23:09:12.122496132] [28275]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUV420 (1) 1536x864-BGGR_PISP_COMP1
[23:09:12.122614224] [28279]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected CFE format: 1536x864-PC1B
ERROR: *** Unable to find an appropriate H.264 codec ***

```

6by9, 2024-05-15:
> I'm still stuck with the same error.

Commenting on a closed issue is generally not a good way of getting attention to your problem. As soon as the notification has been deleted, your comments will have been lost as the issue is closed and considered finished.
 
> I'm using a sample app that appears to be a basic RPI project (https://raspberrytips.com/how-to-live-stream-pi-camera/).

Those instructions appear to predate Pi5, so may not apply.

> root@picam:/home/picam# rpicam-vid -t 2 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8000/}' :demux=h264

`-t 2` is 2ms. Don't expect to get any frames in that time.

> bash: cvlc: command not found

`sudo apt install vlc` would be a good start point.

> [23:09:11.969478681] [28275]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
> [23:09:11.980196409] [28279]  INFO RPI pisp.cpp:695 libpisp version v1.0.5 999da5acb4f4 17-04-2024 [...]

Which rpicam-apps package have you installed or built?
The default on PiOS Lite has no libavcodec or X11 dependencies to keep it small. That does make it rather less useful on a Pi5 that has no hardware H264 encoder, but has reasons.
```
sudo apt purge rpicam-apps-lite
sudo apt install rpicam-apps
```