raspberrypi/rpicam-apps #673: [BUG] When omitting framerate option with rpicam-vid, ExposureTime control reports unexpected value range

christian-nils, 2024-04-25:
**Describe the bug**

When setting explicitely the `framerate` with `rpicam-vid --camera 1 --framerate 30 -v`, the following list of available commands is shown:

<details>
<summary>Available commands with framerate set</summary>

 ```shell
Available controls:
    Sharpness : [0.000000..16.000000]
    AwbEnable : [false..true]
    Contrast : [0.000000..32.000000]
    Saturation : [0.000000..32.000000]
    Brightness : [-1.000000..1.000000]
    AeFlickerPeriod : [100..1000000]
    HdrMode : [0..4]
    ExposureValue : [-8.000000..8.000000]
    ColourGains : [0.000000..32.000000]
    StatsOutputEnable : [false..true]
    ScalerCrop : [(0, 0)/82x64..(0, 0)/4056x3040]
    ExposureTime : [60..674191602]
    AeEnable : [false..true]
    NoiseReductionMode : [0..4]
    AeConstraintMode : [0..3]
    FrameDurationLimits : [24994..674193371]
    AnalogueGain : [1.000000..22.260870]
    AeFlickerMode : [0..1]
    AwbMode : [0..7]
    AeMeteringMode : [0..3]
    AeExposureMode : [0..3]
```

</details>

When omitting the `framerate` option: `rpicam-vid --camera 1 -v`, we get the following list:

<details>
<summary>Available commands with framerate set</summary>

 ```shell
Available controls:
    Sharpness : [0.000000..16.000000]
    AwbEnable : [false..true]
    Contrast : [0.000000..32.000000]
    Saturation : [0.000000..32.000000]
    Brightness : [-1.000000..1.000000]
    AeFlickerPeriod : [100..1000000]
    HdrMode : [0..4]
    ExposureValue : [-8.000000..8.000000]
    ColourGains : [0.000000..32.000000]
    StatsOutputEnable : [false..true]
    ScalerCrop : [(0, 0)/82x64..(0, 0)/4056x3040]
    ExposureTime : [60..0]
    AeEnable : [false..true]
    NoiseReductionMode : [0..4]
    AeConstraintMode : [0..3]
    FrameDurationLimits : [24994..674193371]
    AnalogueGain : [1.000000..22.260870]
    AeFlickerMode : [0..1]
    AwbMode : [0..7]
    AeMeteringMode : [0..3]
    AeExposureMode : [0..3]

```

</details>

The ExposureTime control possible value range changed from `[60..674191602]` to `[60..0]`.

**Bug report**

[report.txt](https://github.com/raspberrypi/rpicam-apps/files/15106082/report.txt)

**Additional context**
 The same bug appears in another app using libcamera (see https://github.com/christianrauch/camera_ros/issues/33), so I would tend to think the issue (if this is not intended) comes from `libcamera` not from `rpicam-apps`.

naushir, 2024-04-25:
I can also see this problem on my setup.  Quite strange, as the FrameDurationLimits comes out correct, and not setting ``--framerate 30`` defaults to a framerate of... 30fps!

Hopefully this is an easy one to fix, probably just a code path we are not going through.

christian-nils, 2024-04-25:
Thanks for checking so quickly. Yesterday, I went through `rpicam-apps` and `libcamera` to check the code path and see if I could find the problem. But I need to admit that my limited knowledge of the code base did not allow me to find the problem. Is this something you can look into? Or shall I report the bug on `libcamera` bug tracker?

Thanks again for the quick help

naushir, 2024-04-25:
It's almost certainly on the libcamera side, I'll have a look at this shortly.  No need to add a bug to the libcamera tracker.

naushir, 2024-04-26:
This patch ought to fix the issue.

```diff
diff --git a/src/ipa/rpi/common/ipa_base.cpp b/src/ipa/rpi/common/ipa_base.cpp
index 149a133ab662..1d12262bda01 100644
--- a/src/ipa/rpi/common/ipa_base.cpp
+++ b/src/ipa/rpi/common/ipa_base.cpp
@@ -592,6 +592,12 @@ void IpaBase::setMode(const IPACameraSensorInfo &sensorInfo)
        mode_.minAnalogueGain = helper_->gain(gainCtrl.min().get<int32_t>());
        mode_.maxAnalogueGain = helper_->gain(gainCtrl.max().get<int32_t>());

+       /*
+        * We need to give the helper the min/max frame durations so it can calculate
+        * the correct exposure limits below.
+        */
+       helper_->setCameraMode(mode_);
+
        /* Shutter speed is calculated based on the limits of the frame durations. */
        mode_.minShutter = helper_->exposure(shutterCtrl.min().get<int32_t>(), mode_.minLineLength);
        mode_.maxShutter = Duration::max();
```

I'll submit it to the libcamera list shortly.

christian-nils, 2024-04-26:
Awesome, thank you very much for the patch! Have a great day!

closed 2024-04-26 by christian-nils