raspberrypi/rpicam-apps #675: Question: Using multiple cameras at the same time?

ArosPrince, 2024-05-09:
Hi, 

this could be a feature request as well, but first I am mostly interested in some thoughts on the topic.

I have a question I am not sure how to tackle. I've been thinking about it for several days and I am still not sure how to proceed.

Initially I wanted to do a simple app that would display a camera feed on the display and be able to zoom in and out. I started with the hello example and eventually was able to succeed.

However, what if I want to use two cameras and be able to quickly switch between the two by, say pressing a keyboard button?

Initially I though I would rewrite `RPiCamApp` class not to have a one camera pointer, but have an array and just start them all. That however proved to be a very difficult task as the class contains a lot of variables that somehow relate to just the one active camera. Plus even if I somehow coped with that, both camera messages would end up in the same message queue and I would somehow had to distinguish which message is from which camera.

Then I realized I could probably be using two RPiCamApp instances each set up to use a different camera. But then I end up with having two message queues with each having it's own wait event. I could probably update the message queue to have something like TryPop, that would not block but it gets rather complicated... I sure could delve into these waters but first wanted to ask someone involved what would be the best way to tackle the problem.

Also I could probably also use libcamera directly, but the nice thing about rpicam-apps is it already has DRM rendering implemented. That part I wouldn't really want to reinvent...

I was also thinking about opencv, which I know quite well, but there the problem is this is Raspbian Lite (i.e. no X, DRM only) so I am not able to display the output (I guess it's possible I haven't found out any way to do so even in the past, I tried multiple times already...).

Thanks for any advice you can give me.

naushir, 2024-05-09:
Good questions, and certainly needs further investigations on what the best approach may be for multiple cameras. Here are some initial thoughts from me:

- The cameras will not be synchronised in any way, will that be a problem?
- Do you want to run both cameras simultaneously?  Or is it strictly one camera at a time?
- If it is one camera at a time, is it important to switch between them with absolutely the minimal amount of time.  Would both cameras run simultaneously, and you only present the output of one of these cameras to the display?
- As you probably found out, adding dual camera support directly into the RPiCamApp class is going to be very tedious.  I would go with the approach of adding this to the application level somehow with multiple RPiCamApp instances.
- Perhaps the best approach is to have 2 application threads handling one message queue each for the 2 instances of RPiCamApp.  Depending on what you want to do with regards to switching between cameras, this might be a very easy solution.

ArosPrince, 2024-05-09:
Hi,
thanks for the quick response.

1. No, not a big deal. This is even not a critical app in terms of latency.
2. Initially no. I just need to switch between the two so I guess the simplest scenario would be to always have one RPiCamApp active and the other one sleeping (StopCamera until activated again). I am just assuming a scenario like that (Picture in Picture, or a splitscreen) may come in future.
3. Partially answered in the previous point. It would be nice to get under say 200 ms for the switch, but I guess that shouldn't be a problem.
4. Yes, very tedious. I gave up midway...
5. Good idea, will give it a thought.

IMHO now that Pi5 has two camera slots, it would be nice to indeed have a dual-camera example in rpicam-apps, but I understand that would take a lot of effort to do it conceptually so that it fits the rest of the project.

ArosPrince, 2024-05-24:
I was finally able to try it out having two `RPiCamApp` instances each handled by a separate thread, but it won't work. I get the following error:

`[50:50:53.613603161] [13719] FATAL IPAManager ipa_manager.cpp:109 Multiple IPAManager objects are not allowed`

There's some underlying singleton that I will have share between the two `RPiCamApp` instances, right?

naushir, 2024-05-28:
Hmmm, it seems like we cannot load 2x rpicam-app instances in the same process as-is without major modifications :(

I wonder if you might be better off using python/picamera2?  We already have [dual camera support](https://github.com/raspberrypi/picamera2/blob/main/examples/multicamera_preview.py) available, and it's much easier to write an app in python when compared to C++.

ArosPrince, 2024-05-28:
Hm, I didn't know that. Looking at the example you linked though, it doesn't seem to be using DRM preview. The thing is, I need to use Raspi Lite image without X. Any chances I could get that to work with two cameras?

ArosPrince, 2024-05-29:
Hmmm, looks like I have wasted several days going this low-level. I had no idea the python library could do that. Somehow I assumed it's gonna be for X based apps only and thought I'd have to go real low-level to get it running with DRM. That's why I opted for the C++ implementation in the first place.

Anyway, thanks. I think I'll be fine from now on. The example you linked indeed does something similar I need. I have some questions, but I think it's better to ask them on the Picamera2 project obviously.

Feel free to close this issue unless you want to keep it around for some future cpp implementation for multiple camera support...

closed 2024-05-30 by naushir