raspberrypi/rpicam-apps #677: 2 sets of frame buffers?

Schaggo, 2024-05-13:
Hey, 
I am trying to run 2 cams on the pi5.
As long as I don't request framebuffers via the RPiCamApp::FLAG stuff kind of works but naturally very slow. 
Without frame buffers the framerate is limited to an awkward  4 fps. Trying to allocate Buffers  the framerate increases to the set rate, but when requesting the buffers for the second camera 4vl tells it 
`Failed to queue buffer 2: Invalid argument
[0:02:01.863328551] [1013] ERROR RPISTREAM rpi_stream.cpp:276 Failed to queue buffer for ISP Output1`

Is there a way to tell v4l to allocate 2+ separate buffers for the second camera?

Thank you for your time!

naushir, 2024-05-13:
Are you trying to run 2 separate instances of rpicam-apps for the 2 different cameras?  If so, this should just work.  Perhaps you can describe in detail what you are doing?

Schaggo, 2024-05-13:
Thank you for the quick answer!

Yes, for now I am trying to run two instances of a custom RPiCamApp based on rpicam_still.
I start the app with the options `--framerate 9 --mode 4624:3472:10:U --camera 0` and `--framerate 9 --mode 4624:3472:10:U --camera 1`, As mentioned, it works if I set the flag `FLAG_STILL_NONE` but `FLAG_STILL_DOUBLE_BUFFER` or `FLAG_STILL_TRIPLE_BUFFER`
 cause the error.

naushir, 2024-05-13:
What if you run 2x instances of ``rpicam-still`` with the exact same commands.  Does it work?

naushir, 2024-05-14:
I will give this a try when I get a chance and see if it is reproducible.  Can you provide the exact command line(s) you are running with?

Schaggo, 2024-05-14:
That's kind of you. 
I am using the following command(s): 
`rpicam-still -t 0 -o test.jpg --framerate 9 --mode 1920:1080:10:U --camera 0`   `// --camera 1` 

I just experimented with other sensor modes and 1080p does work with separate buffers but the framerate is unstable for the camera that gets instantiated later. Which makes me believe, that it is related to memory allocation.

Schaggo, 2024-06-06:
thanks! I will try to test it as soon as possible

closed 2024-06-06 by Schaggo