raspberrypi/rpicam-apps #679: [BUG] Can't build on Ubuntu server 22.04

oralian, 2024-05-28:
I'm having trouble building `rpicam-apps` on a Raspberry Pi 4 with Ubuntu server installed:

```
$ meson setup build -Denable_libav=disabled -Denable_drm=enabled -Denable_egl=disabled -Denable_qt=disabled -Denable_opencv=disabled -Denable_tflite=disabled
The Meson build system
Version: 1.4.0
Source dir: /home/aurelien/projects/github/rpicam-apps
Build dir: /home/aurelien/projects/github/rpicam-apps/build
Build type: native build
Project name: rpicam-apps
Project version: 1.4.4
C compiler for the host machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C linker for the host machine: cc ld.bfd 2.38
C++ compiler for the host machine: c++ (gcc 11.4.0 "c++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C++ linker for the host machine: c++ ld.bfd 2.38
Host machine cpu family: aarch64
Host machine cpu: aarch64
Run-time dependency dl found: YES
Found pkg-config: YES (/usr/bin/pkg-config) 0.29.2
Found CMake: /usr/bin/cmake (3.22.1)
Run-time dependency libcamera found: NO (tried pkgconfig and cmake)

meson.build:35:16: ERROR: Dependency "libcamera" not found, tried pkgconfig and cmake

A full log can be found at /home/aurelien/projects/github/rpicam-apps/build/meson-logs/meson-log.txt
```

I have `libcamera-dev` installed:

```
$ apt show libcamera-dev
Package: libcamera-dev
Version: 0~git20200629+e7aa92a-9
Priority: optional
Section: universe/libdevel
Source: libcamera
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian Multimedia Maintainers <debian-multimedia@lists.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 177 kB
Depends: libc6 (>= 2.34), libcamera0 (= 0~git20200629+e7aa92a-9), libgcc-s1 (>= 3.3.1), libstdc++6 (>= 11)
Homepage: https://libcamera.org/
Download-Size: 45.1 kB
APT-Manual-Installed: yes
APT-Sources: http://ports.ubuntu.com/ubuntu-ports jammy/universe arm64 Packages
Description: complex camera support library (development files)
 libcamera is a complex camera support library which handles low-level
 control of the camera devices, providing a unified higher-level
 programming interface to the applications.
 .
 This package provides the necessary files needed for development
```

I followed the official documentation [here](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-rpicam-apps-without-building-libcamera). Any ideas? Thanks!

naushir, 2024-05-29:
I'm afraid we cannot really provide much support on this as Canonical maintain their own packages including libcamera.  It's possible that they did not include the pkg-config file for libcamera, so the meson build system cannot locate the library.  On the Raspberry Pi OS, this is located in ``/usr/lib/aarch64-linux-gnu/pkgconfig/``

oralian, 2024-05-29:
I got it working by just building `libcamera` myself. Thanks anyways!