raspberrypi/rpicam-apps #682: [BUG] rpicam-vid deadlocks while streaming

LollosoSi, 2024-06-05:
**Describe the bug**
rpicam-vid deadlocks while streaming, output stops, program doesn't quit and no errors displayed

**Bug report**
[bugreport.txt](https://github.com/user-attachments/files/15596239/bugreport.txt)

**Additional context**
Waving my hand in front of the camera is enough to lock the program. higher bitrate delays the deadlock

naushir, 2024-06-12:
Can you try increasing the bitrate and see if that fixes the deadlock?  There was a previous issue reported with similar symptoms, and increasing the bitrate avoided the lockup.

LollosoSi, 2024-06-12:
Yes, increasing the bitrate (current command: `rpicam-vid -t 0 --inline --level 4.2 -b 120k --framerate 24 --width 480 --height 360 --codec h264 -o udp://192.168.10.1:5000`) didn't lock after 3000 frames.
It locked after pressing Ctrl+C.

My project uses very low bitrates (range 15k - 100k), are my expectations too high or is this actually a bug?

naushir, 2024-06-13:
This is a bug in the Broadcom firmware, but I'm afraid chances are slim that it would be fixed any time soon :(
However, in the other issue, I did manage to get low bitrate working with the ``libav`` encoder backend.  Can you try the following command:
```
rpicam-vid -t 0 --inline --level 4.2 -b 120k --framerate 24 --width 480 --height 360 --codec libav --libav-video-codec h264 -o udp://192.168.10.1:5000
```
and see if that works for your use case?

LollosoSi, 2024-06-13:
`--libav-video-codec h264` is not accepted, but this seems to do the trick:

rpicam-vid -t 0 --inline --level 4.2 -b 50k --framerate 24 --width 480 --height 360 --codec libav --libav-video-codec h264_v4l2m2m -o udp://192.168.10.1:5000

will do more tests and update, thanks!

I hope the bug is reported to broadcom

naushir, 2024-06-13:
Oops, sorry, yes you need to use ``h264_v4l2m2m``!

naushir, 2024-06-17:
Any update on this, if not i'll close this shortly.

LollosoSi, 2024-06-18:
Hi, everything seems to work, thank you!
Will reopen this issue in case related problems arise

closed 2024-06-18 by LollosoSi