raspberrypi/rpicam-apps #684: [BUG] rpicam-hello combined with hailo board doesn't work with HQ camera

folkertvanheusden, 2024-06-14:
**Describe the bug**
Situation:
- rpi 5 with the hailo board
- rpi HQ camera

I then run:

` rpicam-hello -t 0 --post-process-file ~/rpicam-apps/assets/hailo_yolov6_inference.json --lores-width 640 --lores-height 640`

This results in:

```
[0:03:54.970417008] [1445]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[0:03:54.978360826] [1448]  INFO RPI pisp.cpp:695 libpisp version v1.0.5 999da5acb4f4 17-04-2024 (14:29:29)
[0:03:54.994671099] [1448]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx296@1a to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
Made X/EGL preview window
Reading post processing stage "hailo_yolo_inference"
Reading post processing stage "object_detect_draw_cv"
[0:03:55.484445100] [1445]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
[0:03:55.484565028] [1445]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
ERROR: *** Low res image larger than viewfinder ***
```

**Bug report**

attached
[bug.txt](https://github.com/user-attachments/files/15836399/bug.txt)

**Additional context**
Add any other context about the problem here.

naushir, 2024-06-14:
Hmmm, this seems to work fine for me.  Can you add ``--viewfinder-width 1920 --viewfinder-height 1080`` to the command line and see if that works for you?

Could you also send me the output of ``rpicam-hello --version`` please?

folkertvanheusden, 2024-06-14:
Hi @naushir,

With the viewfinder settings it works perfectly. Thanks!

```
rpicam-apps build: 49344f2a8d18 31-05-2024 (10:50:13)
libcamera build: v0.2.0+120-eb00c13d
```

closed 2024-06-14 by folkertvanheusden

naushir, 2024-06-14:
Thanks for confirming.  I'm still unsure why you need to set that though, it should have been automatic!

naushir, 2024-06-14:
Are you running in a regular desktop environment or is it in the non-GUI console?

folkertvanheusden, 2024-06-14:
The situation is: regular desktop running on the rpi5. Then I login remote from my laptop into the rpi. After that I use export DISPLAY=:0 and run any commands. So it is not X over ssh, only the command invocation is remote.