raspberrypi/rpicam-apps #685: [LICENSE] Why do core/dma_heaps.h and core/dma_heaps.cpp have LGPL license, where everthing else is BSD?

panthuncia, 2024-06-15:
The files core/dma_heaps.h and core/dma_heaps.cpp have headers which indicate that they are distributed under the LGPL-2.1 license. This would mean that this repo cannot be used under the BSD 2-clause license unless the user stripped out these two files and rewrote them, or somehow dynamically linked just that small section (for LGPL compliance).

This looks like a mistake to me, but is there some reason that these two files must be under the LGPL? As it is, using this repo while complying with the license seems very difficult.

naushir, 2024-06-17:
The LGPL license is carried over from the (nearly) same file in the libcamera source tree.  However, since Raspberry Pi is the sole author of this file, I see no reason not to change the license in this repo to BSD2-clause and match the rest of the source licenses.  The libcamera repo has also moved on and we don't use this file as-is any more either.

If anyone has objections to this, let me know.  Otherwise, I'll update the license shortly.

naushir, 2024-06-26:
This is now fixed.

closed 2024-06-26 by naushir