raspberrypi/rpicam-apps #687: [BUG] rpicam-vid - cannot get bitrate higher than 3m using h264 codec

krzotr, 2024-06-22:
**Describe the bug**
When I use `rpicam-vid` with codec `h264` and bitrate e.g. 20m `-b 20000000`, the bitrate of the final file is less than `3m`. I tested with bitrates 5m, and 10m. Using `libav` + `libx264` everything is fine.

**Bug report**

```
camera-bug-report -o bug.txt -c "rpicam-vid -n --width 1920 --height 1080 --framerate 10 --frames 100  --codec h264  -b 20000000 -o 20M.h264"
```

[bug.txt](https://github.com/user-attachments/files/15937593/bug.txt)

For example,

I use `Raspberry Pi Zero 2 W` and `Raspberry Pi Camera Module 3`. Try to save 10 seconds of video the file with a bitrate of `20m` (quick calculation, 20m bitrate / 8 * 10 seconds, a file size will be ~25MB)

```
rpicam-vid -n --width 1920 --height 1080 --framerate 10 --frames 100  --codec h264  -b 20000000 -o 20M.h264
```

The final file size is below `3MB` not as expected `25MB`. I used `ffmpeg -i 20M.h264 -c copy 20M.h264.mp4` to make mp4 and `ffprobe -i 20M.h264.mp4` to see a bitrate. Is `2.4m` instead of `20m`!

When `libav` + `libx264` is used, everything is fine (quality is better)

```
rpicam-vid -n --width 1920 --height 1080 --framerate 10 --frames 100  --codec libav --libav-video-codec libx264 --libav-video-codec-opts "preset=ultrafast" -b 20000000 -o 20M.mp4
```

```
[10:44:32.623513499] [2752]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+65-6ddd79b5
[10:44:32.826678772] [2755]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[10:44:32.832154835] [2755]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
[10:44:32.832293846] [2755]  INFO RPI pipeline_base.cpp:1104 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Mode selection for 1920:1080:12:P(10)
    SRGGB10_CSI2P,1536x864/120.135 - Score: 2200
    SRGGB10_CSI2P,2304x1296/56.0255 - Score: 1150
    SRGGB10_CSI2P,4608x2592/14.3536 - Score: 2050
[10:44:32.847986928] [2752]  INFO Camera camera.cpp:1183 configuring streams: (0) 1920x1080-YUV420 (1) 2304x1296-SBGGR10_CSI2P
[10:44:32.849155422] [2755]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
[libx264 @ 0x55ae155890] using cpu capabilities: ARMv8 NEON
[libx264 @ 0x55ae155890] profile Main, level 4.1, 4:2:0, 8-bit
[libx264 @ 0x55ae155890] 264 - core 164 r3095 baee400 - H.264/MPEG-4 AVC codec - Copyleft 2003-2022 - http://www.videolan.org/x264.html - options: cabac=0 ref=1 deblock=0:0:0 analyse=0:0x1 me=dia subme=0 psy=1 psy_rd=1.00:0.00 mixed_ref=0 me_range=16 chroma_me=0 trellis=0 8x8dct=0 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=0 threads=6 lookahead_threads=1 sliced_threads=0 slices=1 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=1 b_pyramid=0 b_adapt=0 b_bias=0 direct=1 weightb=0 open_gop=0 weightp=0 keyint=10 keyint_min=1 scenecut=0 intra_refresh=0 rc=abr mbtree=0 bitrate=20000 ratetol=1.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 pb_ratio=1.30 aq=0
Output #0, mp4, to '20M.mp4':
  Stream #0:0: Video: h264, yuv420p(tv, bt709), 1920x1080, q=2-31, 20000 kb/s, 10 fps, 10 tbr, 1000k tbn
[libx264 @ 0x55ae155890] frame I:10    Avg QP: 6.40  size:541195
[libx264 @ 0x55ae155890] frame P:50    Avg QP: 9.22  size:290336
[libx264 @ 0x55ae155890] frame B:40    Avg QP:11.55  size:175845
[libx264 @ 0x55ae155890] consecutive B-frames: 20.0% 80.0%
[libx264 @ 0x55ae155890] mb I  I16..4: 100.0%  0.0%  0.0%
[libx264 @ 0x55ae155890] mb P  I16..4: 16.5%  0.0% 19.0%  P16..4: 61.2%  0.0%  0.0%  0.0%  0.0%    skip: 3.3%
[libx264 @ 0x55ae155890] mb B  I16..4:  7.5%  0.0%  2.1%  B16..8: 34.9%  0.0%  0.0%  direct:48.2%  skip: 7.3%  L0:40.1% L1:37.5% BI:22.3%
[libx264 @ 0x55ae155890] final ratefactor: 13.77
[libx264 @ 0x55ae155890] coded y,uvDC,uvAC intra: 93.9% 86.0% 79.0% inter: 76.9% 37.3% 10.5%
[libx264 @ 0x55ae155890] i16 v,h,dc,p: 15% 17% 35% 32%
[libx264 @ 0x55ae155890] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 18% 15% 34%  4%  8%  4%  5%  4%  7%
[libx264 @ 0x55ae155890] i8c dc,h,v,p: 40% 26% 18% 16%
[libx264 @ 0x55ae155890] kb/s:21571.53
```

Log report bitrate is `21571.53` File size is `25MB`.

Using `h264` and bitrate `1m` generate a result file of `~1MB`. It means I can change a bitrate, but above `3m` is lowered to `3m`.

```
rpicam-vid -n --width 1920 --height 1080 --framerate 10 --frames 100  --codec h264  -b 1000000 -o 1M.h264
```

I have not seen any bitrate limitation of `h264` codec using `rpicam-vid` in documentation.

naushir, 2024-06-26:
The final encoded bitrate might be scene dependent.  Can you describe what you are recording?

naushir, 2024-07-03:
Any update on this?  If not, I'll close it shortly.

closed 2024-07-06 by naushir