raspberrypi/rpicam-apps #688: [BUG] No information on when the bitrate changed using h264 codec

krzotr, 2024-06-22:
**Describe the bug**

After running `rpicam-vid` with params `--level 4.2 --codec h264 -b 26m` Bitrate has been changed from `26m` to `25m`

**Context**

During the investigation of #687 I found the bitrate limit for `h264 ` of my set `Raspberry Pi Zero 2 W` + `Raspberry Pi Camera Module 3` is `25m`. There is no clear information (WARN/INFO) about a bitrate change e.g. **WARN H264Encoder: Sorry, bitrate changed during limitations: 25m, (user bitrate 26m)**. I have not seen any information in the documentation.
Maybe `rpicam-vid` should exit with the above information?

**How to reproduce**

```diff
diff --git a/encoder/h264_encoder.cpp b/encoder/h264_encoder.cpp
index 11d003a..d2bf7c8 100644
--- a/encoder/h264_encoder.cpp
+++ b/encoder/h264_encoder.cpp
@@ -56,8 +56,12 @@ H264Encoder::H264Encoder(VideoOptions const *options, StreamInfo const &info)
        {
                ctrl.id = V4L2_CID_MPEG_VIDEO_BITRATE;
                ctrl.value = options->bitrate.bps();
+                LOG(2, "H264Encoder: Set bitrate (USER)  : " << ctrl.value);
+
                if (xioctl(fd_, VIDIOC_S_CTRL, &ctrl) < 0)
                        throw std::runtime_error("failed to set bitrate");
+
+               LOG(2, "H264Encoder: Set bitrate (DEVICE): " << ctrl.value);
        }
        if (!options->profile.empty())
        {
```

After recompiling the code and run `rpicam-vid` I see

```
./build/apps/rpicam-vid -v 2 --level 4.2 --codec h264 --framerate 120 --frames 1200 --width 1280 --height 720 -b 26m --save-pts timestamp.pts -o video.264 --denoise cdn_off -n 2>&1 | grep 'Set bitrate'
```

```
H264Encoder: Set bitrate (USER)  : 26000000
H264Encoder: Set bitrate (DEVICE): 25000000
```

As you can see `ctrl.value` has changed after call `xioctl` function. Bitrate has been changed from `26m` (user input) to `25m` without any information

**Additional context**

- Level 4.2 supports up to `50m`.
- Everything is ok with `libav` + `libx264`

6by9, 2024-06-23:
The Pi H264 encoder is only specified for level 4.0 (see https://www.raspberrypi.com/products/raspberry-pi-zero-2-w/ stating 1080p30).
Higher levels are permitted in the driver for transcode purposes, not real time use cases.

The driver clips the max frame rate at 25Mbit/s - https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c#L3421-L3424. It's not rpicam-apps that is reducing the bitrate.

naushir, 2024-06-26:
As @6by9 mentioned, this is expected behavior - I don't think there's anything to be done here.  I'll close this now but if you do have further questions, please do reopen.

closed 2024-06-26 by naushir