raspberrypi/rpicam-apps #689: [packaging conflict]

CarlFK, 2024-06-24:
`apt install rpicam-apps` = The following packages will be REMOVED:    libcamera0

I use: `gst-launch-1.0 libcamerasrc...`  which breaks because: 
```
gst-inspect-1.0 libcamerasrc 
  Filename                 /lib/arm-linux-gnueabihf/gstreamer-1.0/libgstlibcamera.so
  Version                  0.0.5+83-bde9b04f  
  Source module            libcamera
  Binary package           libcamera
```
I tried a bit and could not get both libcamerasrc and rpicam-hello installed at the same time.  I think. 

I don't really need rpicam-hello, but it did seem worth reporting this somewhere.  I hope this is the right place.

CarlFK, 2024-06-24:
This seemed relevant: 

The following additional packages will be installed:  raspberrypi-kernel

and  later

dkms: autoinstall for kernel 6.1.21-v8+ was skipped since the kernel headers for this kernel
do not seem to be installed.

Raspberry Pi 4 Model B Rev 1.5

Linux pi42 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

Raspbian GNU/Linux 12 (bookworm)
```
root@pi42:~# apt install libcamera0
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libaacs0 libavdevice59 libavfilter8 libavformat59 libbdplus0 libbluray2
  libboost-log1.74.0 libboost-program-options1.74.0 libboost-thread1.74.0 libcdio-cdda2
  libcdio-paranoia2 libcjson1 libevdev2 libexif12 libhwloc-plugins libhwloc15 libinput-bin
  libinput10 liblttng-ust-common1 liblttng-ust-ctl5 liblttng-ust1 libmbedcrypto7 libmd4c0
  libmtdev1 libmysofa1 libnorm1 libopencv-calib3d406 libopencv-core406 libopencv-dnn406
  libopencv-features2d406 libopencv-flann406 libopencv-imgproc406 libopencv-objdetect406
  libpciaccess0 libpgm-5.3-0 libpisp-common libpisp1 libplacebo208 libpocketsphinx3
  libpostproc56 libprotobuf32 libqt5dbus5 libqt5gui5 libqt5network5 libqt5qml5
  libqt5qmlmodels5 libqt5quick5 libqt5svg5 libqt5waylandclient5 libqt5waylandcompositor5
  libqt5widgets5 librabbitmq4 librist4 librubberband2 libsphinxbase3 libssh-gcrypt-4
  libswscale6 libtbb12 libtbbbind-2-5 libtbbmalloc2 libudfread0 libvidstab1.1
  libwacom-common libwacom9 libxcb-icccm4 libxcb-image0 libxcb-keysyms1 libxcb-render-util0
  libxcb-shape0 libxcb-util1 libxcb-xinerama0 libxcb-xinput0 libxnvctrl0 libzimg2 libzmq5
  pocketsphinx-en-us qt5-gtk-platformtheme qtwayland5
Use 'apt autoremove' to remove them.
The following additional packages will be installed:
  raspberrypi-kernel
The following packages will be REMOVED:
  gstreamer1.0-libcamera libcamera-ipa libcamera0.3 rpicam-apps
The following NEW packages will be installed:
  libcamera0 raspberrypi-kernel
0 upgraded, 2 newly installed, 4 to remove and 0 not upgraded.
Need to get 103 MB of archives.
After this operation, 113 MB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://archive.raspberrypi.com/debian bookworm/main armhf raspberrypi-kernel armhf 1:1.
20230405-1 [102 MB]
Get:2 http://archive.raspberrypi.com/debian bookworm/main armhf libcamera0 armhf 0~git2023072
0+bde9b04f-1 [611 kB]
Fetched 103 MB in 9s (12.1 MB/s)
(Reading database ... 173899 files and directories currently installed.)
Removing gstreamer1.0-libcamera:armhf (0.3.0+rpt20240617-1) ...
Removing rpicam-apps (1.5.0-2) ...
Removing libcamera0.3:armhf (0.3.0+rpt20240617-1) ...
Removing libcamera-ipa:armhf (0.3.0+rpt20240617-1) ...
dpkg: warning: while removing libcamera-ipa:armhf, directory '/usr/lib/arm-linux-gnueabihf/li
bcamera' not empty so not removed
Selecting previously unselected package raspberrypi-kernel.
(Reading database ... 173749 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel_1%3a1.20230405-1_armhf.deb ...
Adding 'diversion of /boot/kernel.img to /usr/share/rpikernelhack/kernel.img by rpikernelhack
'
Adding 'diversion of /boot/kernel7.img to /usr/share/rpikernelhack/kernel7.img by rpikernelha
ck'
Adding 'diversion of /boot/kernel7l.img to /usr/share/rpikernelhack/kernel7l.img by rpikernel
hack'
Adding 'diversion of /boot/kernel8.img to /usr/share/rpikernelhack/kernel8.img by rpikernelha
ck'
Adding 'diversion of /boot/bcm2708-rpi-b-plus.dtb to /usr/share/rpikernelhack/bcm2708-rpi-b-p
lus.dtb by rpikernelhack'
[... 1225 lines cut ...]
dkms: running auto installation service for kernel 6.1.21+.
dkms: autoinstall for kernel: 6.1.21+.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 6.1.21+ /boot/kernel.img
run-parts: executing /etc/kernel/postinst.d/z50-raspi-firmware 6.1.21+ /boot/kernel.img
run-parts: executing /etc/kernel/postinst.d/dkms 6.1.21-v7+ /boot/kernel7.img
/etc/kernel/postinst.d/dkms:
dkms: running auto installation service for kernel 6.1.21-v7+.
dkms: autoinstall for kernel: 6.1.21-v7+.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 6.1.21-v7+ /boot/kernel7.img
run-parts: executing /etc/kernel/postinst.d/z50-raspi-firmware 6.1.21-v7+ /boot/kernel7.img
run-parts: executing /etc/kernel/postinst.d/dkms 6.1.21-v7l+ /boot/kernel7l.img
/etc/kernel/postinst.d/dkms:
dkms: running auto installation service for kernel 6.1.21-v7l+.
dkms: autoinstall for kernel: 6.1.21-v7l+.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 6.1.21-v7l+ /boot/kernel7l.img
run-parts: executing /etc/kernel/postinst.d/z50-raspi-firmware 6.1.21-v7l+ /boot/kernel7l.img
run-parts: executing /etc/kernel/postinst.d/dkms 6.1.21-v8+ /boot/kernel8.img
/etc/kernel/postinst.d/dkms:
dkms: autoinstall for kernel 6.1.21-v8+ was skipped since the kernel headers for this kernel
do not seem to be installed.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 6.1.21-v8+ /boot/kernel8.img
run-parts: executing /etc/kernel/postinst.d/z50-raspi-firmware 6.1.21-v8+ /boot/kernel8.img
Setting up libcamera0:armhf (0~git20230720+bde9b04f-1) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u7) ...
```

CarlFK, 2024-06-24:
notable above: 

dkms: autoinstall for kernel 6.1.21-v8+ was skipped since the kernel headers for this kernel
do not seem to be installed.

A) it is trying to install  6.1.21 (or maybe modules for some other kernel?  I don't understand.) 

B) failed.  seems like there is something wrong.

CarlFK, 2024-06-24:
some notes from IRC

```
(02:17:28 PM) Tenkawa: gstreamer1.0-libcamera : Breaks: libcamera0 are the 2 culprits it looks like 
(02:17:46 PM) Tenkawa: gstreamer is causing a cascade failure
(02:19:07 PM) Tenkawa: libcamera0.3 seems to fix things
(02:22:13 PM) Tenkawa: this combination seems to work   rpicam-apps  libcamera-ipa libcamera0.3 gstreamer1.0-libcamera
```

naushir, 2024-06-26:
@XECDesign any thoughts on this?

XECDesign, 2024-07-09:
Sorry I took a while getting to this, but I'm having trouble reproducing any kind of issue here.

```
$ rpicam-hello --version
rpicam-apps build: 49344f2a8d18 17-06-2024 (12:09:07)
libcamera build: v0.3.0+65-6ddd79b5
$ gst-inspect-1.0 libcamerasrc
Factory Details:
  Rank                     primary (256)
  Long-name                libcamera Source
  Klass                    Source/Video
  Description              Linux Camera source using libcamera
  Author                   Nicolas Dufresne <nicolas.dufresne@collabora.com

Plugin Details:
  Name                     libcamera
  Description              libcamera capture plugin
  Filename                 /lib/aarch64-linux-gnu/gstreamer-1.0/libgstlibcamera.so
  Version                  0.3.0+65-6ddd79b5
  License                  LGPL
  Source module            libcamera
  Binary package           libcamera
  Origin URL               https://libcamera.org
...
$ apt policy rpicam-apps-lite gstreamer1.0-tools gstreamer1.0-libcamera
rpicam-apps-lite:
  Installed: 1.5.0-2
  Candidate: 1.5.0-2
  Version table:
 *** 1.5.0-2 500
        500 http://archive.raspberrypi.org/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
gstreamer1.0-tools:
  Installed: 1.22.0-2
  Candidate: 1.22.0-2
  Version table:
 *** 1.22.0-2 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
gstreamer1.0-libcamera:
  Installed: 0.3.0+rpt20240617-1
  Candidate: 0.3.0+rpt20240617-1
  Version table:
 *** 0.3.0+rpt20240617-1 500
        500 http://archive.raspberrypi.org/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
     0.0.3-6 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
```

What two packages are you claiming can't co-exist? libcamera0 getting removed is expected, since it has been replaced by libcamera0.3.

CarlFK, 2024-07-11:
>  libcamera0 getting removed is expected, since it has been replaced by libcamera0.3.

Should / could apt replace it?

CarlFK, 2024-07-11:
seems yes: 
"I would expect a Breaks and Replaces in libcamera0.3 on libcamera0 and so apt would install libcamera0.3"

XECDesign, 2024-07-16:
I'm not seeing anything unexpected in the output and the commands referenced work without issues as far as I can tell.

Unless there's an example of something that's actually not working, I think this can be closed.

CarlFK, 2024-07-16:
It breaks my working system, which was built using apt, so apt should not break it.
There is a way to keep it from breaking.

XECDesign, 2024-07-16:
Breaks in what way? What do I need to run to see a breakage? Maybe I'm blind or stupid, but I don't see evidence of any relevant breakage in any of the previous comments.

I'm happy to download an older bookworm image, upgrade it and try it, but I need something like "run this command, you should see X but what you actually get is Y".

naushir, 2024-07-16:
I'm also unable to reproduce the reported issue on a standard Raspberry Pi Bookworm OS environment.

XECDesign, 2024-07-18:
I'm just not seeing what you're seeing. It's all working as expected in my case.

Can you please post the output of:
```
apt policy
apt policy rpicam-apps gstreamer1.0-tools gstreamer1.0-libcamera
apt depends gstreamer1.0-libcamera rpicam-apps
```

CarlFK, 2024-07-18:
Here is what you asked for, but first something I just noticed that is confusing to me:

I am using `gst-launch-1.0 libcamerasrc ...`

`gst-inspect-1.0 libcamerasrc` shows ` Filename                 /lib/arm-linux-gnueabihf/gstreamer-1.0/libgstlibcamera.so`

```
dpkg -S libgstlibcamera.so
libcamera0:armhf: /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstlibcamera.so
```
```
 apt policy gstreamer1.0-libcamera
gstreamer1.0-libcamera:
  Installed: (none)
```
None, and yet this does not error: 
`gst-launch-1.0 -v libcamerasrc ! fakesink`

I will look into this more July 21 and post my findings.

Now answering your questions: 

This is after 
* image
* apt update/upgrade
* apt install gstreamer1.0-tools gstreamer1.0-libcamera 
* apt update/upgrade

so before trying to install rpicam-apps

```
20:32:02 pi@pi3:~ $ apt policy
Package files:
 100 /var/lib/dpkg/status
     release a=now
 500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
     release o=Raspberry Pi Foundation,a=stable,n=bookworm,l=Raspberry Pi Foundation,c=main,b=arm64
     origin archive.raspberrypi.com
 500 http://archive.raspberrypi.com/debian bookworm/main armhf Packages
     release o=Raspberry Pi Foundation,a=stable,n=bookworm,l=Raspberry Pi Foundation,c=main,b=armhf
     origin archive.raspberrypi.com
 500 http://raspbian.raspberrypi.com/raspbian bookworm/rpi armhf Packages
     release o=Raspbian,a=stable,n=bookworm,l=Raspbian,c=rpi,b=armhf
     origin raspbian.raspberrypi.com
 500 http://raspbian.raspberrypi.com/raspbian bookworm/non-free armhf Packages
     release o=Raspbian,a=stable,n=bookworm,l=Raspbian,c=non-free,b=armhf
     origin raspbian.raspberrypi.com
 500 http://raspbian.raspberrypi.com/raspbian bookworm/contrib armhf Packages
     release o=Raspbian,a=stable,n=bookworm,l=Raspbian,c=contrib,b=armhf
     origin raspbian.raspberrypi.com
 500 http://raspbian.raspberrypi.com/raspbian bookworm/main armhf Packages
     release o=Raspbian,a=stable,n=bookworm,l=Raspbian,c=main,b=armhf
     origin raspbian.raspberrypi.com
Pinned packages:
20:32:34 pi@pi3:~ $ 
```
```
20:32:34 pi@pi3:~ $ apt policy rpicam-apps gstreamer1.0-tools gstreamer1.0-libcamera
rpicam-apps:
  Installed: (none)
  Candidate: 1.5.0-2
  Version table:
     1.5.0-2 500
        500 http://archive.raspberrypi.com/debian bookworm/main armhf Packages
gstreamer1.0-tools:
  Installed: 1.22.0-2
  Candidate: 1.22.0-2
  Version table:
 *** 1.22.0-2 500
        500 http://raspbian.raspberrypi.com/raspbian bookworm/main armhf Packages
        100 /var/lib/dpkg/status
gstreamer1.0-libcamera:
  Installed: (none)
  Candidate: 0.3.0+rpt20240617-1
  Version table:
     0.3.0+rpt20240617-1 500
        500 http://archive.raspberrypi.com/debian bookworm/main armhf Packages
     0.0.3-6 500
        500 http://raspbian.raspberrypi.com/raspbian bookworm/main armhf Packages
20:33:14 pi@pi3:~ $ 
```
```
20:33:14 pi@pi3:~ $ apt depends gstreamer1.0-libcamera rpicam-apps
gstreamer1.0-libcamera
  Depends: libc6 (>= 2.32)
  Depends: libcamera0.3 (>= 0.3.0+rpt20240617)
  Depends: libgcc-s1 (>= 3.5)
  Depends: libglib2.0-0 (>= 2.37.3)
  Depends: libgstreamer-plugins-base1.0-0 (>= 1.12.0)
  Depends: libgstreamer1.0-0 (>= 1.18.0)
  Depends: libstdc++6 (>= 11)
  Breaks: libcamera0
  Replaces: libcamera0
rpicam-apps
  Depends: libavcodec59 (>= 7:5.0)
    libavcodec-extra59
  Depends: libavdevice59 (>= 7:5.0)
  Depends: libavformat59 (>= 7:5.0)
    libavformat-extra59
  Depends: libavutil57 (>= 7:5.1)
  Depends: libboost-program-options1.74.0 (>= 1.74.0+ds1)
  Depends: libc6 (>= 2.34)
  Depends: libcamera0.3 (>= 0.3.0+rpt20240617)
  Depends: libdrm2 (>= 2.4.99)
  Depends: libepoxy0 (>= 1.0)
  Depends: libexif12 (>= 0.6.21-1~)
  Depends: libgcc-s1 (>= 3.5)
  Depends: libjpeg62-turbo (>= 1.3.1)
  Depends: libopencv-core406 (>= 4.6.0+dfsg)
  Depends: libopencv-imgproc406 (>= 4.6.0+dfsg)
  Depends: libopencv-objdetect406 (>= 4.6.0+dfsg)
  Depends: libpng16-16 (>= 1.6.2-1)
  Depends: libqt5core5a (>= 5.0.2)
 |Depends: libqt5gui5 (>= 5.0.2)
  Depends: libqt5gui5-gles (>= 5.0.2)
  Depends: libqt5widgets5 (>= 5.0.2)
  Depends: libstdc++6 (>= 12)
  Depends: libswresample4 (>= 7:5.1)
  Depends: libtiff6 (>= 4.5.0~rc1)
  Depends: libx11-6
  Depends: <python3:any>
    python3
  Conflicts: rpicam-apps-lite
  Breaks: libcamera-apps (<< 1.4.1-1~)
  Replaces: libcamera-apps (<< 1.4.1-1~)
  Replaces: rpicam-apps-lite
20:33:46 pi@pi3:~ $ 
```

naushir, 2024-09-03:
Closing now.  Please re-open if you have the same problem.

closed 2024-09-03 by naushir