raspberrypi/rpicam-apps #690: v4l2src1 Failed to allocate required memory ov5647 ubuntu 22.04 64-bit

Jai-GAY, 2024-06-28:
```
$cat /boot/firmware/config.txt 
#camera_auto_detect=0
start_x=1
dtoverlay=ov5647

$ groups pi
pi : pi adm cdrom sudo dip video plugdev lpadmin lxd sambashare

$ sudo cheese 

(cheese:3509): GLib-GObject-CRITICAL **: 09:17:45.154: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:3509): GLib-GObject-CRITICAL **: 09:17:45.154: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:3509): GLib-GObject-CRITICAL **: 09:17:45.155: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:3509): GLib-GObject-CRITICAL **: 09:17:45.155: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:3509): GLib-GObject-CRITICAL **: 09:17:45.155: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:3509): GLib-GObject-CRITICAL **: 09:17:45.390: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:3509): GLib-GObject-CRITICAL **: 09:17:45.400: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:3509): GLib-GObject-CRITICAL **: 09:17:45.408: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:3509): GLib-GObject-CRITICAL **: 09:17:45.472: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:3509): GLib-GObject-CRITICAL **: 09:17:45.491: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:3509): cheese-WARNING **: 09:17:45.524: Failed to allocate required memory.: ../sys/v4l2/gstv4l2src.c(759): gst_v4l2src_decide_allocation (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin18/GstV4l2Src:v4l2src1:
Buffer pool activation failed
```
[image]

6by9, 2024-06-28:
Cheese and v4l2src don't support Bayer sensors - they want fully processed YUV or RGB images, which is not what ov5647 delivers. libcamera takes the raw Bayer frames, passes them through the ISP, and therefore produces processed YUV or RGB frames.

You can try and use `libcamerify cheese` to use libcamera's V4L2 compatibility layer.
For GStreamer you're better off using libcamerasrc to use libcamera directly.
What versions of each are installed on Ubuntu is down to Canonical rather than Raspberry Pi.

`start_x=1` is redundant for libcamera and will only result in using some extra memory for the firmware.
rpicam-apps shouldn't need to be run with `sudo`. Having to do so implies a permissions issue, probably on /dev/dma-heaps

Jai-GAY, 2024-07-01:
> `start_x=1` is redundant for libcamera and will only result in using some extra memory for the firmware.

oh I see, thanks for the guidance.

> rpicam-apps shouldn't need to be run with `sudo`.

I need to investigate if is it due to this command line [**sudo** ninja -C build install](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera).

> You can try and use `libcamerify cheese` to use libcamera's V4L2 compatibility layer.

thanks, it ran once for a few seconds then an error was logged, subsequent run failed.

[0:12:47.623382997] [2868] ERROR Request request.cpp:472 FrameBuffer already set for stream
[0:12:47.623461891] [2868] ERROR V4L2Compat v4l2_camera.cpp:259 Can't set buffer for request

Failed to allocate a buffer: ../sys/v4l2/gstv4l2src.c(1146): gst_v4l2src_create (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin18/GstV4l2Src:v4l2src1

```
$ sudo libcamerify cheese 
[sudo] password for pi: 
[0:12:37.083754826] [2784]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:12:37.174030864] [2785]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:12:37.180399833] [2785]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media1
[0:12:39.967233621] [2734]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:12:40.035550847] [2856]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:12:40.040479835] [2856]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media1
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:32.704: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:32.719: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:32.734: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:32.753: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:32.753: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:32.754: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
[0:12:40.633513263] [2870]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUYV
[0:12:40.634882343] [2856]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
[0:12:40.639818899] [2870]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUYV
[0:12:40.640634106] [2856]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:33.436: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:33.474: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
[0:12:40.938770931] [2869]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUYV
[0:12:40.940419326] [2856]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
[0:12:40.944527472] [2869]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUYV
[0:12:40.945345938] [2856]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:33.533: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:33.628: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:33.671: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:33.705: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
[0:12:47.623382997] [2868] ERROR Request request.cpp:472 FrameBuffer already set for stream
[0:12:47.623461891] [2868] ERROR V4L2Compat v4l2_camera.cpp:259 Can't set buffer for request
(cheese:2734): cheese-WARNING **: 12:11:40.319: Failed to allocate a buffer: ../sys/v4l2/gstv4l2src.c(1146): gst_v4l2src_create (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin18/GstV4l2Src:v4l2src1
```

naushir, 2024-07-01:
Closing as this is not related to ``rpicam-apps`` at all.

Jai-GAY, 2024-07-01:
> Closing as this is not related to `rpicam-apps` at all.

Can you help enlighten me as to[ why is it not related](https://www.raspberrypi.com/documentation/computers/camera_software.html#v4l2-drivers)?
Are you able to assist where I should report it?

naushir, 2024-07-01:
``rpicam-apps`` is an application suite for driving cameras through the native libcamera API.

``cheese`` and ``libcamerify`` are unrelated to this.  Probably best to add a new issue to https://github.com/raspberrypi/libcamera, and we can try and help.