raspberrypi/rpicam-apps #691: [BUG] --latest doesn't work when the target is not in the same directory as the output file

sroller, 2024-06-30:
**Describe the bug**
I'm using rpicam-still to create a timelapse.

    rpicam-still -v 2 --nopreview --mode 3280:2464:10 --width 1920 --height 1080 --timelapse 5000 --timeout 15000 --latest /root/temp/latest/last.jpg' -o /root/temp/tl%06d.jpg

I'd like to have the last picture linked to a directory within my web server.
When I use the parameter "--latest" and specify are target outside of the output directory then the created link fails.
The link for the first file of the series is not correct and therefore all further attempts to remove it fail as well.

I believe the the change has to happen in line 69 of rpicam_still.cpp. The call to symlink() needs to include the path of the target relative to the file.

That's how it worked with raspistill.

[bug.txt](https://github.com/user-attachments/files/16042970/bug.txt)

naushir, 2024-07-19:
https://github.com/raspberrypi/rpicam-apps/pull/699 ought to fix this.

naushir, 2024-08-12:
Fixed now.

closed 2024-08-12 by naushir