raspberrypi/rpicam-apps #692: Building rpicam-apps without building libcamera on Pi 4B + ubuntu 22.04

Jai-GAY, 2024-07-02:
I am just curious, does anyone [build the rpicam-apps without building the libcamera](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-rpicam-apps-without-building-libcamera) successfully on Ubuntu 22.04 64bit?

```
The Meson build system
Version: 1.4.1
Source dir: /home/pi/rpicam-apps
Build dir: /home/pi/rpicam-apps/build
Build type: native build
Project name: rpicam-apps
Project version: 1.5.0
C compiler for the host machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C linker for the host machine: cc ld.bfd 2.38
C++ compiler for the host machine: c++ (gcc 11.4.0 "c++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C++ linker for the host machine: c++ ld.bfd 2.38
Host machine cpu family: aarch64
Host machine cpu: aarch64
Run-time dependency dl found: YES
Found pkg-config: YES (/usr/bin/pkg-config) 0.29.2
Found CMake: /usr/bin/cmake (3.22.1)
Run-time dependency libcamera found: NO (tried pkgconfig and cmake)

meson.build:35:16: ERROR: Dependency "libcamera" not found, tried pkgconfig and cmake

A full log can be found at /home/pi/rpicam-apps/build/meson-logs/meson-log.txt
```

naushir, 2024-07-02:
Did you install ``libcamera-dev`` via apt?

Jai-GAY, 2024-07-02:
> Did you install `libcamera-dev` via apt?

Yes

naushir, 2024-07-02:
This sounds like Canonical may have something wrong with their ``libcamera-dev`` package.  Probably something to do with the ``pkg-config`` file which is used to locate the libcamera library.  I'm afraid you will have to get in touch with their support to resolve this.

Jai-GAY, 2024-07-03:
> Probably something to do with the `pkg-config` file which is used to locate the libcamera library.

from meson-log.txt
```
-----------
Command line: `c++ /home/pi/rpicam-apps/build/meson-private/tmpcgu460e4/testfile.cpp -o /home/pi/rpicam-apps/build/meson-private/tmpcgu460e4/output.exe -D_FILE_OFFSET_BITS=64 -O0 -fpermissive` -> 0
Run-time dependency dl found: YES
Pkg-config binary missing from cross or native file, or env var undefined.
Trying a default Pkg-config fallback at pkg-config
Found pkg-config: YES (/usr/bin/pkg-config) 0.29.2
Determining dependency 'libcamera' with pkg-config executable '/usr/bin/pkg-config'
env[PKG_CONFIG_PATH]: 
env[PKG_CONFIG]: /usr/bin/pkg-config
-----------
Called: `/usr/bin/pkg-config --modversion libcamera` -> 1
stderr:
Package libcamera was not found in the pkg-config search path.
Perhaps you should add the directory containing `libcamera.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libcamera' found
-----------
```
from meson.build
```
summary({
            'location' : libcamera_dep.get_variable('libdir'),
            'version' : libcamera_dep.version()
        },
        section : 'libcamera')

rpicam_app_src = []
rpicam_app_dep = [libcamera_dep, dl_dep]
```
```
$ dpkg -l | grep libcamera
ii  libcamera-dev:arm64                               0~git20200629+e7aa92a-9                 arm64        complex camera support library (development files)
ii  libcamera-tools                                   0~git20200629+e7aa92a-9                 arm64        complex camera support library (tools)
ii  libcamera0:arm64                                  0~git20200629+e7aa92a-9                 arm64        complex camera support library
ii  ros-humble-camera-ros                             0.1.0-2jammy.20240524.040616            arm64        node for libcamera supported cameras (V4L2, Raspberry Pi Camera Modules)
ii  ros-humble-libcamera                              0.1.0-1jammy.20240516.151057            arm64        An open source camera stack and framework for Linux, Android, and ChromeOS

```

naushir, 2024-07-03:
Can you find ``libcamera.pc`` anywhere  in ``/usr/lib`` or /usr/local/lib``?

Jai-GAY, 2024-07-04:
> Can you find `libcamera.pc` anywhere in `/usr/lib` or `/usr/local/lib`?

```
$ ll /usr/lib/l*
lrwxrwxrwx 1 root root       39 May  7 04:34 /usr/lib/ld-linux-aarch64.so.1 -> aarch64-linux-gnu/ld-linux-aarch64.so.1*
lrwxrwxrwx 1 root root       22 Feb  7  2022 /usr/lib/libarmadillo.so.10 -> libarmadillo.so.10.8.2
-rw-r--r-- 1 root root    65384 Feb  7  2022 /usr/lib/libarmadillo.so.10.8.2
-rw-r--r-- 1 root root  1755280 Jan 19  2022 /usr/lib/libCatch2WithMain.a
lrwxrwxrwx 1 root root       17 Sep 14  2021 /usr/lib/libdfalt.so.0 -> libdfalt.so.0.0.0
-rw-r--r-- 1 root root   511504 Sep 14  2021 /usr/lib/libdfalt.so.0.0.0
lrwxrwxrwx 1 root root       17 Mar 17  2022 /usr/lib/libgdal.so.30 -> libgdal.so.30.0.1
-rw-r--r-- 1 root root 20427840 Mar 17  2022 /usr/lib/libgdal.so.30.0.1
lrwxrwxrwx 1 root root       20 Sep 14  2021 /usr/lib/libmfhdfalt.so.0 -> libmfhdfalt.so.0.0.0
-rw-r--r-- 1 root root   157720 Sep 14  2021 /usr/lib/libmfhdfalt.so.0.0.0
lrwxrwxrwx 1 root root       17 Feb  8  2021 /usr/lib/libnetpbm.so.10 -> libnetpbm.so.10.0
-rw-r--r-- 1 root root   145928 Feb  8  2021 /usr/lib/libnetpbm.so.10.0
lrwxrwxrwx 1 root root       14 Jun 14  2021 /usr/lib/libogdi.so.4 -> libogdi.so.4.1
-rw-r--r-- 1 root root   104632 Jun 14  2021 /usr/lib/libogdi.so.4.1
lrwxrwxrwx 1 root root       13 Jun 14  2021 /usr/lib/libvpf.so.4 -> libvpf.so.4.1
-rw-r--r-- 1 root root   157856 Jun 14  2021 /usr/lib/libvpf.so.4.1

/usr/lib/libreoffice:
total 2056
drwxr-xr-x   5 root root    4096 Jun 24 15:11 ./
drwxr-xr-x 114 root root    4096 Jun 24 15:10 ../
-rw-r--r--   1 root root 1800346 Oct 27  2022 CREDITS.fodt
-rw-r--r--   1 root root  269216 May 15 15:06 LICENSE.html
-rw-r--r--   1 root root    3820 Oct 27  2022 NOTICE
drwxr-xr-x   7 root root    4096 Aug  7  2023 presets/
drwxr-xr-x  11 root root   12288 Jun 24 15:24 program/
drwxr-xr-x  33 root root    4096 Aug  7  2023 share/

/usr/lib/linux:
total 12
drwxr-xr-x   3 root root 4096 Aug  7  2023 ./
drwxr-xr-x 114 root root 4096 Jun 24 15:10 ../
drwxr-xr-x   2 root root 4096 Mar 19 14:19 triggers/

/usr/lib/linux-firmware-raspi:
total 21516
drwxr-xr-x   2 root root    4096 Aug  7  2023 ./
drwxr-xr-x 114 root root    4096 Jun 24 15:10 ../
-rw-r--r--   1 root root   52476 Jul 28  2022 bootcode.bin
-rw-r--r--   1 root root    3170 Jul 28  2022 fixup4cd.dat
-rw-r--r--   1 root root    5400 Jul 28  2022 fixup4.dat
-rw-r--r--   1 root root    8382 Jul 28  2022 fixup4db.dat
-rw-r--r--   1 root root    8386 Jul 28  2022 fixup4x.dat
-rw-r--r--   1 root root    3170 Jul 28  2022 fixup_cd.dat
-rw-r--r--   1 root root    7265 Jul 28  2022 fixup.dat
-rw-r--r--   1 root root   10229 Jul 28  2022 fixup_db.dat
-rw-r--r--   1 root root   10227 Jul 28  2022 fixup_x.dat
-rw-r--r--   1 root root  805084 Jul 28  2022 start4cd.elf
-rw-r--r--   1 root root 3746856 Jul 28  2022 start4db.elf
-rw-r--r--   1 root root 2250656 Jul 28  2022 start4.elf
-rw-r--r--   1 root root 2998120 Jul 28  2022 start4x.elf
-rw-r--r--   1 root root  805084 Jul 28  2022 start_cd.elf
-rw-r--r--   1 root root 4818728 Jul 28  2022 start_db.elf
-rw-r--r--   1 root root 2974880 Jul 28  2022 start.elf
-rw-r--r--   1 root root 3721800 Jul 28  2022 start_x.elf

/usr/lib/linux-sound-base:
total 16
drwxr-xr-x   2 root root 4096 Aug  7  2023 ./
drwxr-xr-x 114 root root 4096 Jun 24 15:10 ../
-rw-r--r--   1 root root 3916 Feb 22  2021 noALSA.modprobe.conf
-rw-r--r--   1 root root 1059 Feb 22  2021 noOSS.modprobe.conf

/usr/lib/llvm-14:
total 44
drwxr-xr-x   8 root root  4096 Dec 21  2023 ./
drwxr-xr-x 114 root root  4096 Jun 24 15:10 ../
drwxr-xr-x   2 root root  4096 Dec 21  2023 bin/
drwxr-xr-x   4 root root  4096 Dec 21  2023 build/
lrwxrwxrwx   1 root root    14 Jul 14  2023 cmake -> lib/cmake/llvm/
drwxr-xr-x   3 root root  4096 Dec 21  2023 include/
drwxr-xr-x   6 root root 16384 Dec 21  2023 lib/
drwxr-xr-x   2 root root  4096 Dec 21  2023 libexec/
drwxr-xr-x   8 root root  4096 Dec 21  2023 share/

/usr/lib/locale:
total 5880
drwxr-xr-x   3 root root    4096 Jun 24 15:18 ./
drwxr-xr-x 114 root root    4096 Jun 24 15:10 ../
drwxr-xr-x   3 root root    4096 Jun 24 15:10 C.utf8/
-rw-r--r--   1 root root 6043888 Jun 24 15:21 locale-archive

/usr/lib/lp_solve:
total 592
drwxr-xr-x   2 root root   4096 Aug  7  2023 ./
drwxr-xr-x 114 root root   4096 Jun 24 15:10 ../
-rw-r--r--   1 root root 594152 Mar 25  2022 liblpsolve55.so

/usr/lib/lsb:
total 24
drwxr-xr-x   3 root root  4096 Aug  7  2023 ./
drwxr-xr-x 114 root root  4096 Jun 24 15:10 ../
-rw-r--r--   1 root root 11526 Aug 26  2019 init-functions
drwxr-xr-x   2 root root  4096 Jan 26 16:29 init-functions.d/
```
```
$ ll /usr/local/lib/*
/usr/local/lib/aarch64-linux-gnu:
total 1076
drwxr-xr-x 5 root root    4096 Jul  1 16:37 ./
drwxr-xr-x 4 root root    4096 Jan  9 15:46 ../
drwxr-xr-x 2 root root    4096 Jul  1 16:37 gstreamer-1.0/
drwxr-xr-x 2 root root    4096 Jul  1 16:37 libcamera/
lrwxrwxrwx 1 root root      13 Jan  9 17:13 libcamera_app.so -> rpicam_app.so*
drwxr-xr-x 2 root root    4096 Jul  1 16:37 pkgconfig/
lrwxrwxrwx 1 root root      19 Jan  9 17:13 rpicam_app.so -> rpicam_app.so.1.4.1*
-rwxr-xr-x 1 root root 1079440 Jan  9 17:13 rpicam_app.so.1.4.1*

/usr/local/lib/python3.10:
total 12
drwxr-xr-x  3 root root 4096 Aug  7  2023 ./
drwxr-xr-x  4 root root 4096 Jan  9 15:46 ../
drwxr-xr-x 28 root root 4096 Feb 14 14:59 dist-packages/
```

naushir, 2024-07-04:
This sounds like it needs to be fixed by the Canonical team.  ``libcamera.pc`` is needed to locate the libcamera library needed by rpicam-apps.  I'm afraid there's nothing more I can do to help here, but I'm sure the Canonical support team will be able to help.

closed 2024-07-04 by naushir

Jai-GAY, 2024-07-05:
> `libcamera.pc` is needed to locate the libcamera library needed by rpicam-apps.

[I see](https://git.launchpad.net/ubuntu/+source/libcamera/tree/debian/libcamera-dev.install?h=ubuntu/jammy).
```
$ ll /usr/lib/aarch64-linux-gnu/pkgconfig/libc**.pc
-rw-r--r-- 1 root root  291 Jan 11  2023 /usr/lib/aarch64-linux-gnu/pkgconfig/libcanberra-gtk3.pc
-rw-r--r-- 1 root root  288 Jan 11  2023 /usr/lib/aarch64-linux-gnu/pkgconfig/libcanberra-gtk.pc
-rw-r--r-- 1 root root  241 Jan 11  2023 /usr/lib/aarch64-linux-gnu/pkgconfig/libcanberra.pc
-rw-r--r-- 1 root root  223 Jun  7  2023 /usr/lib/aarch64-linux-gnu/pkgconfig/libcap.pc
-rw-r--r-- 1 root root  300 Feb 16 16:51 /usr/lib/aarch64-linux-gnu/pkgconfig/libcrypto.pc
lrwxrwxrwx 1 root root   12 Dec 18  2021 /usr/lib/aarch64-linux-gnu/pkgconfig/libcrypt.pc -> libxcrypt.pc
-rw-r--r-- 1 root root 1886 Mar 19 20:16 /usr/lib/aarch64-linux-gnu/pkgconfig/libcurl.pc
```
So, which platform has the developer team tested?

naushir, 2024-07-05:
> So, which platform has the developer team tested?

We only officially support [Raspberry Pi OS](https://www.raspberrypi.com/software/).

automationboy, 2024-08-14:
shit