raspberrypi/rpicam-apps #701: rpicam-apps-1.5.0 fails to build on an RPi3

juanitotc, 2024-07-27:
Compiling rpicam-apps-1.5.0 natively 32-bit on an RPi3 using gcc-13.2.0 and against libcamera-0.3.0, the build fails with the error below.

Is there a way to get more debug detail?

Compiling natively 64-bit on an RPi4 using gcc-13.2.0 succeeds.

```
[5/43] g++ -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -Os -pipe -Ipost_processing_stages/core-postproc.so.p -Ipost_processing_stages -I../post_processing_stages -I.. -I../.. -I/usr/local/include/libcamera -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -pedantic -Wno-unused-parameter -faligned-new -Wno-psabi -fPIC -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -DLIBDRM_PRESENT=1 -DLIBEGL_PRESENT=1 -MD -MQ post_processing_stages/core-postproc.so.p/hdr_stage.cpp.o -MF post_processing_stages/core-postproc.so.p/hdr_stage.cpp.o.d -o post_processing_stages/core-postproc.so.p/hdr_stage.cpp.o -c ../post_processing_stages/hdr_stage.cpp
FAILED: post_processing_stages/core-postproc.so.p/hdr_stage.cpp.o 
```

naushir, 2024-07-29:
Can you configure the following meson option and see if it helps:

```
-Dneon_flasgs=armv8-neon
```

juanitotc, 2024-07-29:
I'll try that, but if I'm compiling for all versions of RPi using "-march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp", won't that (-mfpu=neon-fp-armv8) conflict?

Edit: I just tried "-Dneon_flags=armv8-neon" and got the same error as the first post.

juanitotc, 2024-07-29:
Ah - even though there were no out of memory errors, adding a large swap file has enabled the build to succeed.

Sorry for the noise...

closed 2024-07-29 by juanitotc