raspberrypi/rpicam-apps #705: [BUG] Error when using two identical cameras

KATLA666, 2024-08-18:
I am trying to run these to commands simultaneous
```
rpicam-hello --camera 0 --preview 0,0,400,480 --width 1080 --height 1200 --framerate 90 --denoise cdn_off  -t 0 
rpicam-hello --camera 1 --preview 400,0,400,480 --width 1080 --height 1200 --framerate 90 --denoise cdn_off  -t 0 
```
but I get only one preview and the second command runs with an Error.
```
ERROR V4L2 v4l2_device.cpp:353 'imx708_noir': Unable to set controls: Device or resource busy
```

I am using two RPI cam V3 modules together with the RPI5.
According to this post it is possible with two different Cameras.
[https://thepihut.com/blogs/raspberry-pi ... berry-pi-5](https://thepihut.com/blogs/raspberry-pi-tutorials/how-to-use-two-camera-modules-with-raspberry-pi-5)

Bugreport via the tool:
[bug.txt](https://github.com/user-attachments/files/16651919/bug.txt)

davidplowman, 2024-08-19:
Hi, thanks for the report. Just a few extra questions, if I may!

1. I wasn't quite clear how you're running this. You have the graphical desktop installed, and you're running both commands in separate terminal windows? Your `--camera 1` command line seemed to produce `Preview window unavailable` which confused me slightly. Or are you running them in console mode (no graphical desktop), or over ssh?

2. In bug.txt, the `rpicam-hello --list-cameras` output looks very strange. What does that report if you run it directly in the terminal window (with no cameras running)?

3. Do both your cameras work OK if you run them on their own, that is, with the other camera not being started?

4. Are you able to run both cameras successfully if you supply both with the `-n` option? This will disable the preview window, so you won't see any camera images, but it should be obvious enough if the camera is actually running. (What if you supply just one of them with `-n`?)

5. What happens if you try a fresh download of Raspberry Pi OS? Don't touch or change any configuration settings or files at all, just run the cameras and they should both work.

I've just tried this myself running on the standard Raspberry Pi GUI desktop, and it seemed to work OK. Though I could only lay my hands on a pair of non-"noir" modules, but fairly unlikely, I think, that it would make a difference. I say "worked", but I did note that the preview window offset doesn't seem to be respected ...

Thanks!

KATLA666, 2024-08-19:
Hi, thanks for the answer

1. I am running Raspberry Pi OS Lite, so without a graphical desktop and I was sending the commands over two separate ssh connections (sorry for the late clarification).
2. 
```
 Available cameras
-----------------
 0 : imx708_noir [4608x2592 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]

 1 : imx708_noir [4608x2592 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@80000/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]

```
3. Yes they do
4. No, the second camera still runs with the error When using ```-n``` on both.
 When using the first command with ```-n``` the second runs still with the error, but the preview for the second cam appears. 
5. When using a fresh download of Raspberry Pi OS (with desktop) they both work (when started from two terminals) and I can see two previews. However the second camera still runs with the same error.

davidplowman, 2024-08-19:
Hi again, so if you're running both cameras without the graphical desktop, then you aren't going to be able to display the previews at the same time. That's because each application has to get exclusive access to the DRM (Direct Rendering Manager) graphics stack in order to show the images, which means that the second application is going to get refused. Linux doesn't really provide APIs that make it easy for multiple application to share a bare display like this, I guess they feel that's the job of a proper windowing UI. The easiest workaround might just be to run the graphical desktop version of the OS (which will run very well on a Pi 5).

I think the V4L2 error is a red herring in this case. The second app may be complaining that it can't do certain things to the first camera, which is actually just as well seeing as another instance of the camera application is using it. But I don't think it's the cause of the 2nd camera not starting.

KATLA666, 2024-08-24:
Hi, thanks for the explanation. 
Anything I can to against this not respected preview window offset?

davidplowman, 2024-08-27:
I don't know of any workarounds, but nor have I tried anything. Is there some Qt API call to move a window that you could try calling a little after they've been created?

Another approach might just be to code the DRM version yourself. It's not terribly difficult - I gave a short example of displaying two images side by side [here](https://forums.raspberrypi.com/viewtopic.php?p=2246351&hilit=drm#p2246351).

naushir, 2024-09-03:
Closing now as DRM does not support 2 sources running simulataneously.

closed 2024-09-03 by naushir