raspberrypi/rpicam-apps #706: [BUG] Cam v3 wide and Raspberry 4B errror: *** no cameras available ***

migueltorresvalls, 2024-08-25:
I just bought the camera module 3 wide for my raspberry pi 4B and have been all morning trying to install it with no luck. I have been going over multiple threads with similar problems but none have been useful. I currently have a fresh new 64-bit BookWorm OS. 

Whenever I run `rpicam-hello` I get the following error. 
```
[0:08:48.591126406] [2057]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+65-6ddd79b5
Preview window unavailable
ERROR: *** no cameras available ***
```

I am pretty sure camera connection is good and camera is undamaged since I just took it out of the case. Any help is much appreciated, thank you!!

config.txt is as follows: 
```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]__

```

6by9, 2024-08-25:
[What to do if your camera is not detected](https://forums.raspberrypi.com/viewtopic.php?t=362707)

The contacts of the camera cable should face away from USB/ethernet ports of the Pi4, and face forward on the camera module (see the photo in https://www.raspberrypi.com/documentation/accessories/camera.html)

migueltorresvalls, 2024-08-26:
Hi @6by9, thank you for your quick resonse.

I followed the instructions you sent but still its not detecting the camera. 

- My kernel version is: 
`Linux raspberrypi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`
- My OS: 
```
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

- My libcamera version: 
```
rpicam-apps build: 49344f2a8d18 17-06-2024 (12:19:15)
libcamera build: v0.3.0+65-6ddd79b5
```
My camera is connected the same way as showed in the video demo.

naushir, 2024-08-27:
Can you please attach the output created with the ``camera-bug-report`` command?

migueltorresvalls, 2024-08-27:
[bug-report.txt](https://github.com/user-attachments/files/16759770/bug-report.txt)

naushir, 2024-08-27:
The camera seems to fail at the firmware detection stage - this might suggest a faulty module or camera cable.  If you have an alternate cable, I suggest trying that out.  If it still fails, you will need to get in touch with your supplier and get a replacement module sent to you.

closed 2024-08-28 by migueltorresvalls