raspberrypi/rpicam-apps #707: dmaHeap allocation failure for rpicam-apps0

jsalsburg, 2024-08-29:
**Describe the bug**
Running AllSky App on RPi4; RPiArduCam_16_MP. AllSky reports "capture program stopped". This appears to happen when AllSky enters 'end of night/begin day' some mornings but not always. 

**Bug report**
AllSky reports in capture_RPi_debug.txt - 
dmaHeap allocation failure for rpicam-apps0
ERROR: *** failed to allocate capture buffers for stream ***

**Additional context**
The AllSky log reports that this command will work sometimes and not other times.
libcamera-still --output '/home/rpi/allsky/tmp/image-20240828060449.jpg' --timeout 1 --nopreview --width 4656 --height 3496 --shutter 4077762 --analoggain 1 --quality 100

I do not have easy access to the headless RPi, it is remote and inconveniently inaccessible. 

AllSky does not recover from this error without reboot.

[image]

naushir, 2024-08-29:
It looks like you are running out of memory to allocate framebuffers from the CMA heap.  Have you adjusted anything in the config.txt file on the system?

jsalsburg, 2024-08-30:
Thank you for responding.
I may be causing this issue. To forgo the limitations of the RPi4 SD card, I attached a USB/SSD to the RPi and altered AllSky to store images to the SSD instead. I may have underestimated the capacity of the SSD and included too many DAYS_TO_KEEP=30, so I changes it to 20. We will see if this solves the issue.

naushir, 2024-09-03:
Closing this issue now.  Please reopen if you still have problems.

closed 2024-09-03 by naushir

naushir, 2024-09-06:
It's unlikely - you will need that to load the correct sensor driver anyway.

What is the output of ``rpicam-hello --version``?  Maybe also try adding ``--buffer-count 1 --viewfinder-buffer-count 3`` to the command line and see if that helps?