raspberrypi/rpicam-apps #711: [BUG] - Floating Point Exception

mymatenige, 2024-09-06:
**Describe the bug**
Since upgrading to 1.5.1 from 1.5.0 on a Raspberry PI I'm getting a floating point exception. I see this on a number of different models of device. The example below is from listing cameras, but it is stopping me capturing images with the camera.

```
$ /usr/bin/libcamera-still --list-cameras
Available cameras
-----------------
0 : ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@1/ov5647@36)
[bug-report.txt](https://github.com/user-attachments/files/16904949/bug-report.txt)

Floating point exception
```

**Bug report**
`camera-bug-report -c 'libcamera-still --list-cameras'`

**Additional context**
Add any other context about the problem here.

mymatenige, 2024-09-06:
[bug-report.txt](https://github.com/user-attachments/files/16904962/bug-report.txt)

mymatenige, 2024-09-06:
[bug-report.txt](https://github.com/user-attachments/files/16904968/bug-report.txt)

naushir, 2024-09-06:
I don't seem to be able to reproduce this error.  Have you tried a fresh SD card image?
Would you be able run under ``gdb`` and generate a call stack?

mymatenige, 2024-09-06:
I've tried it on a clean install and I get the same. I'm seeing it on both a RPI Zero and 2B.  I'll see what I can do with gdb ...

mymatenige, 2024-09-06:
What is interesting here is that one the attempts worked ...

```
pi@pi2b-01:~ $ gdb --args libcamera-still --list-cameras
GNU gdb (Raspbian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from libcamera-still...
(No debugging symbols found in libcamera-still)
(gdb) run
Starting program: /usr/bin/libcamera-still --list-cameras
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x7595dc80 (LWP 18584)]
[New Thread 0x7515cc80 (LWP 18585)]
[New Thread 0x7495bc80 (LWP 18587)]
[New Thread 0x73f7ac80 (LWP 18616)]
[New Thread 0x73779c80 (LWP 18618)]
Available cameras
-----------------
0 : ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [30.00 fps - (65535, 65535)/65535x65535 crop]
                             1296x972 [30.00 fps - (65535, 65535)/65535x65535 crop]
                             1920x1080 [30.00 fps - (65535, 65535)/65535x65535 crop]
                             2592x1944 [30.00 fps - (65535, 65535)/65535x65535 crop]

[Thread 0x73779c80 (LWP 18618) exited]
[Thread 0x73f7ac80 (LWP 18616) exited]
[Thread 0x7495bc80 (LWP 18587) exited]
[Thread 0x7595dc80 (LWP 18584) exited]
[Thread 0x76b4c780 (LWP 18411) exited]
[Thread 0x7515cc80 (LWP 18585) exited]
[New process 18411]
[Inferior 1 (process 18411) exited normally]
(gdb) quit
pi@pi2b-01:~ $ libcamera-still --list-cameras
Available cameras
-----------------
0 : ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@1/ov5647@36)
Floating point exception
pi@pi2b-01:~ $ gdb --args libcamera-still --list-cameras
GNU gdb (Raspbian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from libcamera-still...
(No debugging symbols found in libcamera-still)
(gdb) run
Starting program: /usr/bin/libcamera-still --list-cameras
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x7595dc80 (LWP 20463)]
[New Thread 0x7515cc80 (LWP 20464)]
[New Thread 0x7495bc80 (LWP 20465)]
[New Thread 0x73f7ac80 (LWP 20491)]
[New Thread 0x73779c80 (LWP 20493)]
Available cameras
-----------------
0 : ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@1/ov5647@36)

Thread 4 "libcamera-still" received signal SIGFPE, Arithmetic exception.
[Switching to Thread 0x7495bc80 (LWP 20465)]
__pthread_kill_implementation (threadid=1955970176, signo=signo@entry=8, no_tid=no_tid@entry=0) at pthread_kill.c:44
44	pthread_kill.c: No such file or directory.
(gdb) 

```

naushir, 2024-09-06:
Admittedly I am testing this on a 64-bit OS image.  Since you are running on a Pi Zero, it would be a 32-bit image.  I'm setting one up now and going to give it a try.

mymatenige, 2024-09-06:
I've just downgraded to 1.5.0 and I'm seeing the same. Apologies. I think there is something else in play here. Another package that has been updated. That said; I'm interested to see if you replicate it on a 32-bit arch.

mymatenige, 2024-09-06:
If I downgrade libcamera-ipa the error stops, but the cameras aren't detected. I'm out of time now, but I'll take another look over the weekend.

naushir, 2024-09-06:
I think I may have reproduced this on a Pi Zero.  I'll look into it further.

naushir, 2024-09-06:
I think a candidate fix is at https://github.com/raspberrypi/libcamera/commit/69a894c4adad524d3063dd027f5c4774485cf9db.  I'm testing this now.  Once we have verified this fixes the issue we will make a new release.

oxivanisher, 2024-09-06:
I have this error also on 2 Rpi 4 with the 64bit OS. I also upgraded the packages last night and this happens now. Let me know if I can help in testing or so.

naushir, 2024-09-06:
If you could try rebuilding libcamera with the above commit and let me know if it fixes your issue that would be great.

I'm fairly confident it should work after that change.

oxivanisher, 2024-09-06:
> If you could try rebuilding libcamera with the above commit and let me know if it fixes your issue that would be great.
> 
> I'm fairly confident it should work after that change.

I just tested your fix successfully. :) Its working on my machine with https://github.com/raspberrypi/libcamera/commit/69a894c4adad524d3063dd027f5c4774485cf9db

mymatenige, 2024-09-07:
That's fixed it for me as well. Many thanks.

I'm not sure if you have any influence on the instructions here

https://www.raspberrypi.com/documentation/computers/camera_software.html#build-libcamera-and-rpicam-apps

But the dependancies list for building rpicam-apps is missing libpng-dev.

oxivanisher, 2024-09-07:
Perfect, it works again with the new package. Thanks! :)

naushir, 2024-09-09:
Fixes have now been published.

closed 2024-09-09 by naushir