raspberrypi/rpicam-apps #713: [BUG] rpicam-vid does not draw Hailo based detections, bounding boxes coordinates are set to 0

eG9yIGF4LGF4, 2024-09-12:
**Describe the bug**
The following command does not draw detections rectangles, all the labels of detection are at the top-left corner and rectangles do not appear at all:
```bash
rpicam-vid -t 0 --fullscreen --post-process-file /home/pi/src/rpicam-apps/assets/hailo_yolov6_inference.json --lores-width 640 --lores-height 640 --width 2304 --height 1296
```

However, the following one does:
```bash
rpicam-hello -t 0 --fullscreen --post-process-file /home/pi/src/rpicam-apps/assets/hailo_yolov6_inference.json --lores-width 640 --lores-height 640 --width 2304 --height 1296
```

My setup:

- Raspberry Pi 5 (Bookworm, Lite)
- Hailo based AI Kit 
- Camera Module v3
- rpicam-apps cloned at /home/pi/src directory :) 

The issue appears on rpicam-apps installed via apt, as well as reproduced with app version compiled from source.

After debugging rpicam-vid and rpicam-hello with specified above arguments I have figured out that [these 2 lines](https://github.com/raspberrypi/rpicam-apps/blob/main/post_processing_stages/hailo/hailo_yolo_inference.cpp#L195-L196) of code are behaving a bit differently for rpicam-vid and rpicam-hello apps: for some unknown reason when I launch rpicam-hello `rpi_scaler_crop` contains 2 same items, however in case of rpicam-vid it has 3rd one as well.. As a result method `ConvertInferenceCoordinates` called [here](https://github.com/raspberrypi/rpicam-apps/blob/main/post_processing_stages/hailo/hailo_yolo_inference.cpp#L275) behave differently - in case of rpicam-hello it properly converts coordinates, in case of rpicam-vid the result is zero rectangle (all coordinates of it are 0).

Personally I have found quiet dumb but acceptable workaround for my own, but as far as this problem appears on apt-based installation as well, you might want to fix it in the future. Also attaching camera-bug-report result, in case it helps.

To make it a bit easier here are some screens of debugging process:

Command `rpicam-hello -t 0 --fullscreen --post-process-file /home/pi/src/rpicam-apps/assets/hailo_yolov6_inference.json --lores-width 640 --lores-height 640 --width 2304 --height 1296 `:
[image]
[image]

Command `rpicam-vid -t 0 --fullscreen --post-process-file /home/pi/src/rpicam-apps/assets/hailo_yolov6_inference.json --lores-width 640 --lores-height 640 --width 2304 --height 1296 `:
[image]
[image]

Hope that helps..

Thanks.

**Bug report**

Output of a `camera-bug-report -t 1 -o bug -c "rpicam-vid -t 0 --fullscreen --post-process-file /home/pi/src/rpicam-apps/assets/hailo_yolov6_inference.json --lores-width 640 --lores-height 640 --width 2304 --height 1296 "` command attached in a [bug.txt](https://github.com/user-attachments/files/16971753/bug.txt) file

naushir, 2024-09-12:
Thank you for the report.  I think I do spot the problem - although I cannot really explain why rpicam-hello did not show the issue as well.

I have a fix that needs to be applied to libcamera.  Are you able to test the fix that I have pushed to https://github.com/raspberrypi/libcamera/tree/next?

eG9yIGF4LGF4, 2024-09-16:
Sure @naushir.. But as far as previously I was building rpicam-apps with builtin version libcamera, it will take a bit of time for me...
Be back as soon as finish. Thank you!

eG9yIGF4LGF4, 2024-09-18:
@naushir that helped. Thanks!

naushir, 2024-09-18:
Yes, this will be merged soon.

closed 2024-09-18 by naushir