raspberrypi/rpicam-apps #714: [BUG] Image gets stretched to 16:9 ratio after updating from 1.5.0 to 1.5.1

Te-eMster, 2024-09-12:
**Describe the bug**
I have a Raspberry Cam v3 on a Raspberry Zero 2W working as a camera node for a Frigate Server.
Since the area I am surveilling is rather wide, I used to use a resolution of 1536x320. It worked fine that rpicam-apps just "cropped" the image and cut off the top, but the image still didn't look "stretched".
However, after the update from 1.5.0 to 1.5.1 the image seems to be re-sized to 16:9 aspect ratio, resulting in a very stretched looking image.
Is there a way to set this back to just "cropping" the image?

naushir, 2024-09-13:
Do you know the exact command line that is used in Frigate?
We have found a small bug in the crop code, a fix has been posted at https://github.com/raspberrypi/libcamera/commit/91ccff1bc838b858be8f7626cd513af1e21a9d0f.  Are you able to build a custom version of libcamera and test this change?

Te-eMster, 2024-09-13:
The command used is:
rpicam-vid -t 0 --framerate 20 --width 1536 --height 320 --inline --autofocus-mode manual --lens-position 0.5 -n -o -

Unfortunately I don't know how to build a custom version of libcamera and deploy it to my device.

naushir, 2024-09-30:
This should now be fixed with the latest release.  Please do reopen if you are still having problems.

closed 2024-09-30 by naushir

Te-eMster, 2024-10-09:
Thanks, works like a charm :-)