raspberrypi/rpicam-apps #715: Problems with colours in AllSky software

babsenfred, 2024-09-13:
Hallo all,

As discribed in the [AllSky Github issues](https://github.com/AllskyTeam/allsky/discussions/3678#discussioncomment-10632182), some users experience a problem with the Rpi HQ camera.

Mostly at night, some pictures are “too blue”.
Maintainer @EricClaeys suggested to open an issue here because he thinks it is a libcamera software issue.

I’m running a fully updated Bookworm on the RPI5 with the Raspberry HQ Camera.
Attached you can see a timeline of the images taken with a 60 seconds interval. 

Could this be a libcamera software issue?

Regards
[image]

naushir, 2024-09-13:
Can you provide us the exact rpicam-apps command line used to capture the image and the actual JPG/DNG images as well please?  Without this, there's not much we can investigate I'm afraid.

babsenfred, 2024-09-13:
Hello @naushir , sorry I’m really not very experienced with Linux. What do you mean with ”rpicam-apps command line”?

naushir, 2024-09-13:
rpicam-apps is the application that captures the images for allskycam.  The application is configured by a number of parameters on the command line.  This would be useful to know exactly how the system is configured to capture images.  Perhaps speak with the package authors and they can give you this information?

babsenfred, 2024-09-13:
Hello @naushir ; I will ask them, thanks.
In att 2 images taken 60 seconds apart.
[image]
[image]

naushir, 2024-09-13:
Are you able to attach the original images taken from the camera - the once you posted in the comment above don't have any metadata in them.

naushir, 2024-09-13:
Unfortunately these do not have metadata either :(

Perhaps you can ask the authors if the original camera image is stored anywhere on the filesystem?

6by9, 2024-09-13:
I suspect Github may strip metadata off images. It'd be worth trying ZIPing them and attaching the archive.

EricClaeys, 2024-09-13:
@babsenfred, to get the command line used to take the pictures, run:
~~~
grep rpicam-still /var/log/allsky.log > /tmp/rpicam.txt
~~~
There will be one line for every image.  You can attach `/tmp/rpicam.txt`.

All the images are stored in `~/allsky/images/YYYYMMDD` where `YYYYMMDD` is something like `20240912`

jfharvey, 2024-09-14:
I also have the same issue, so maybe I can share my own images and command line...

Here are two images taken about 60 seconds apart.
From what I can see, there is no metadata in these images.  They have been downloaded directly from the RPI5 using WinSCP (/Home/pi/allsky/images/20240913/)

[image]

[image]

and here are the commands line used for each image....

2024-09-13T22:23:32.191317-04:00 Allsky allsky[16016]: STARTING EXPOSURE at: 2024-09-13 22:23:32   @ 60.0 sec
2024-09-13T22:23:32.207545-04:00 Allsky allsky[16016]:  > Running: libcamera-still --output '/home/pi/allskyold/allsky/tmp/image-20240913222332.jpg' --timeout 1 --nopreview --width 4056 --height 3040 --shutter 60000000 --analoggain 8 --quality 100
2024-09-13T22:24:34.137377-04:00 Allsky allsky[16016]:   > Saving NIGHT image 'image-20240913222332.jpg'
2024-09-13T22:24:34.145002-04:00 Allsky allsky[16016]:   > Sleeping: 10.0 sec
2024-09-13T22:24:44.139692-04:00 Allsky allsky[16016]: -----
2024-09-13T22:24:44.139831-04:00 Allsky allsky[16016]: STARTING EXPOSURE at: 2024-09-13 22:24:44   @ 60.0 sec
2024-09-13T22:24:44.155749-04:00 Allsky allsky[16016]:  > Running: libcamera-still --output '/home/pi/allskyold/allsky/tmp/image-20240913222444.jpg' --timeout 1 --nopreview --width 4056 --height 3040 --shutter 60000000 --analoggain 8 --quality 100
2024-09-13T22:25:46.165833-04:00 Allsky allsky[16016]:   > Saving NIGHT image 'image-20240913222444.jpg'
2024-09-13T22:25:46.166357-04:00 Allsky allsky[16016]:   > Sleeping: 10.0 sec
2024-09-13T22:25:56.168243-04:00 Allsky allsky[16016]: -----

EricClaeys, 2024-09-14:
@jfharvey, you are using Auto White Balance, correct?

jfharvey, 2024-09-14:
No, Auto white balance is disabled for both day and night captures, with default settings for the RPI HQ camera.

[image]

I tried to enable Auto white balance few weeks ago, but I didn't see any change.

Thanks

naushir, 2024-09-16:
@jfharvey unfortunately I still cannot see any exif data in the images.  As @6by9 said in an eariler comment, it might be GitHub stripping metadata when attaching the files.  Could you zip up the image and attach the file instead please?

Also, has this only recently started happening, or was the problem always visible?

jfharvey, 2024-09-16:
Hi naushir

Here are the two files zipped.  But as I said, I checked on my PC before uploading them, and I can't see any metedata in the pictures, but you can double check, maybe I missed something
@EricClaeys , should metadata be present in images from  /Home/pi/allsky/images/yyyymmdd/ ?

The issue started when I switched to a RPI5.  On my RPI4 (using the same camera), there is no "blue image" issue.

Here are the last night thumbnails.  We can clearly see the issue.

[image]

and here is the zip file
[test.zip](https://github.com/user-attachments/files/17012475/test.zip)

EricClaeys, 2024-09-17:
@jfharvey, Allsky doesn't add any metadata to the images.  I don't know if `libcamera-still` adds any - none of the command-line arguments tell it to.

@naushir, several other people also see the "blue images".  I asked them to comment in this thread.

naushir, 2024-09-17:
@davidplowman for visibility.

@jfharvey and @EricClaeys can you run ``rpicam-hello --version`` and let me know the version string that is printed.  Additionally, can you enable RAW output in the command line by adding the ``-r`` option?  This will produce a DNG which might be useful for debugging.

If we are sure that AWB is disabled, can I suggest that we disable the ALSC algorthim and see if that fixes the issue for you.  To do this, edit ``/usr/share/libcamera/ipa/rpi/pisp/imx477.json``.  Under the ``rpi.alsc`` section, can you set the value of ``n_iter`` to 0, and try running again?

Thank you for your help in debugging this with us!

jfharvey, 2024-09-17:
Hello naushir,

rpicam-apps build: 6202c09b2bfa 16-02-2024 (17:15:39)
libcamera build: v0.2.0+46-075b54d5

I will check with @EricClaeys if it is possible to enable the RAW output and I will try again.

I will also disable the ALSC algorithm later today and see if it is better...or not.

Regards

naushir, 2024-09-17:
Those builds are quite old, would it be possible to update your libcamera libraries as well please?

jfharvey, 2024-09-17:
Thanks naushir

I'm 100% sure I did a sudo apt update/upgrade last week, after the OS installation, but I just did another one to be sure, and now, the version is the following

rpicam-apps build: d7a1a13b041e 05-09-2024 (14:07:57)
libcamera build: v0.3.1+50-69a894c4

Regards

EricClaeys, 2024-09-17:
@jfharvey, you can add additional command-line arguments via the EXTRA PARAMETERS setting.

jfharvey, 2024-09-18:
Hello,
I disabled the ALSC algorithm as requested.  Unfortunately, The blue image issue is still present, as you can see in the attached files.

I also included the RAW files.  Maybe you will find something useful in the metadata.

Here is an excerpt from the log showing the command line used for each picture.

2024-09-17T20:20:49.579876-04:00 Allsky allsky[5081]: STARTING EXPOSURE at: 2024-09-17 20:20:49   @ 10.0 sec
2024-09-17T20:20:49.595971-04:00 Allsky allsky[5081]:  > Running: libcamera-still --output '/home/pi/allsky/tmp/image-20240917202049.jpg' --raw --timeout 1 --nopreview --width 4056 --height 3040 --shutter 10000000 --analoggain 8 --quality 100
2024-09-17T20:21:01.707044-04:00 Allsky allsky[5081]:   > Got:    shutter_us: 10.0 sec, gain: 8.000, mean: 0.203, target mean: 0.200, diff (target - mean): -0.003
2024-09-17T20:21:01.707163-04:00 Allsky allsky[5081]:   > index: 0, meanHistory[]=0.189 exposureLevelHistory[]=226, newNean=0.189
2024-09-17T20:21:01.707184-04:00 Allsky allsky[5081]:   > index: 1, meanHistory[]=0.208 exposureLevelHistory[]=226, newNean=0.606
2024-09-17T20:21:01.707201-04:00 Allsky allsky[5081]:   > index: 2, meanHistory[]=0.203 exposureLevelHistory[]=226, newNean=1.215
2024-09-17T20:21:01.707218-04:00 Allsky allsky[5081]:   > New mean target: 0.201, mean_forecast: 0.198, mean_diff (newMean - target mean): 0.001, idx=2, idxN1=1
2024-09-17T20:21:01.707234-04:00 Allsky allsky[5081]:   > ExposureChange clipped to 3 (diff from last change: 0)
2024-09-17T20:21:01.707248-04:00 Allsky allsky[5081]:   > ++++++++++ Prior image mean good - no changes needed, mean=0.203, target mean=0.200 threshold=0.100
2024-09-17T20:21:01.707266-04:00 Allsky allsky[5081]:   > Saving NIGHT image 'image-20240917202049.jpg'
2024-09-17T20:21:01.707515-04:00 Allsky allsky[5081]:   > Sleeping: 80.0 sec

2024-09-17T20:22:21.709236-04:00 Allsky allsky[5081]: STARTING EXPOSURE at: 2024-09-17 20:22:21   @ 10.0 sec
2024-09-17T20:22:21.725597-04:00 Allsky allsky[5081]:  > Running: libcamera-still --output '/home/pi/allsky/tmp/image-20240917202221.jpg' --raw --timeout 1 --nopreview --width 4056 --height 3040 --shutter 10000000 --analoggain 8 --quality 100
2024-09-17T20:22:33.834788-04:00 Allsky allsky[5081]:   > Got:    shutter_us: 10.0 sec, gain: 8.000, mean: 0.204, target mean: 0.200, diff (target - mean): -0.004
2024-09-17T20:22:33.834916-04:00 Allsky allsky[5081]:   > index: 1, meanHistory[]=0.208 exposureLevelHistory[]=226, newNean=0.208
2024-09-17T20:22:33.834944-04:00 Allsky allsky[5081]:   > index: 2, meanHistory[]=0.203 exposureLevelHistory[]=226, newNean=0.614
2024-09-17T20:22:33.834962-04:00 Allsky allsky[5081]:   > index: 0, meanHistory[]=0.204 exposureLevelHistory[]=226, newNean=1.227
2024-09-17T20:22:33.834979-04:00 Allsky allsky[5081]:   > New mean target: 0.205, mean_forecast: 0.205, mean_diff (newMean - target mean): 0.005, idx=0, idxN1=2
2024-09-17T20:22:33.834995-04:00 Allsky allsky[5081]:   > ExposureChange clipped to 3 (diff from last change: 0)
2024-09-17T20:22:33.835010-04:00 Allsky allsky[5081]:   > ++++++++++ Prior image mean good - no changes needed, mean=0.204, target mean=0.200 threshold=0.100
2024-09-17T20:22:33.835027-04:00 Allsky allsky[5081]:   > Saving NIGHT image 'image-20240917202221.jpg'
2024-09-17T20:22:33.835293-04:00 Allsky allsky[5081]:   > Sleeping: 80.0 sec

[image_1.zip](https://github.com/user-attachments/files/17036755/image_1.zip)
[image_2.zip](https://github.com/user-attachments/files/17036763/image_2.zip)

EricClaeys, 2024-09-18:
@naushir,
here's a summary of the log output above, showing only the commands used to take the two pictures.  Notice the commands are identical, yet one image was blue:

libcamera-still --output '/home/pi/allsky/tmp/image-20240917202049.jpg' --raw --timeout 1 --nopreview --width 4056 --height 3040 --shutter 10000000 --analoggain 8 --quality 100

libcamera-still --output '/home/pi/allsky/tmp/image-20240917202221.jpg' --raw --timeout 1 --nopreview --width 4056 --height 3040 --shutter 10000000 --analoggain 8 --quality 100

naushir, 2024-09-18:
Thanks, these show some interesting results.

**image-20240917202049.dng**
EXIF:
```
ExifTool Version Number         : 12.40
File Name                       : image-20240917202049.dng
Directory                       : .
File Size                       : 24 MiB
File Modification Date/Time     : 2024:09:18 01:21:01+01:00
File Access Date/Time           : 2024:09:18 08:04:49+01:00
File Inode Change Date/Time     : 2024:09:18 08:04:49+01:00
File Permissions                : -rwxrwxrwx
File Type                       : DNG
File Type Extension             : dng
MIME Type                       : image/x-adobe-dng
Exif Byte Order                 : Little-endian (Intel, II)
Make                            : Raspberry Pi
Camera Model Name               : imx477
Orientation                     : Horizontal (normal)
Software                        : rpicam-still
Subfile Type                    : Full-resolution image
Image Width                     : 4056
Image Height                    : 3040
Bits Per Sample                 : 16
Compression                     : Uncompressed
Photometric Interpretation      : Color Filter Array
Samples Per Pixel               : 1
Planar Configuration            : Chunky
CFA Repeat Pattern Dim          : 2 2
CFA Pattern 2                   : 2 1 1 0
Black Level Repeat Dim          : 2 2
Black Level                     : 4096 4096 4096 4096
White Level                     : 65535
Strip Offsets                   : 0
Strip Byte Counts               : 0
Exposure Time                   : 10
ISO                             : 800
Date/Time Original              : 2024:09:17 20:21:01
DNG Version                     : 1.1.0.0
DNG Backward Version            : 1.0.0.0
Unique Camera Model             : Raspberry Pi imx477
Color Matrix 1                  : 0.7559810877 -0.1972008348 -0.04402547702 -0.1264587343 0.896514833 0.205433175 0.003762362292 0.0730144456 0.1681934893
As Shot Neutral                 : 0.4733974636 1 0.2597243786
Calibration Illuminant 1        : D65
CFA Pattern                     : [Blue,Green][Green,Red]
Image Size                      : 4056x3040
Megapixels                      : 12.3
Shutter Speed                   : 10
```
ISP output:
[image]

dcraw output:
[image]

**image-20240917202221.dng**
EXIF:
```
ExifTool Version Number         : 12.40
File Name                       : image-20240917202221.dng
Directory                       : .
File Size                       : 24 MiB
File Modification Date/Time     : 2024:09:18 01:22:33+01:00
File Access Date/Time           : 2024:09:18 08:04:53+01:00
File Inode Change Date/Time     : 2024:09:18 08:04:53+01:00
File Permissions                : -rwxrwxrwx
File Type                       : DNG
File Type Extension             : dng
MIME Type                       : image/x-adobe-dng
Exif Byte Order                 : Little-endian (Intel, II)
Make                            : Raspberry Pi
Camera Model Name               : imx477
Orientation                     : Horizontal (normal)
Software                        : rpicam-still
Subfile Type                    : Full-resolution image
Image Width                     : 4056
Image Height                    : 3040
Bits Per Sample                 : 16
Compression                     : Uncompressed
Photometric Interpretation      : Color Filter Array
Samples Per Pixel               : 1
Planar Configuration            : Chunky
CFA Repeat Pattern Dim          : 2 2
CFA Pattern 2                   : 2 1 1 0
Black Level Repeat Dim          : 2 2
Black Level                     : 4096 4096 4096 4096
White Level                     : 65535
Strip Offsets                   : 0
Strip Byte Counts               : 0
Exposure Time                   : 10
ISO                             : 800
Date/Time Original              : 2024:09:17 20:22:33
DNG Version                     : 1.1.0.0
DNG Backward Version            : 1.0.0.0
Unique Camera Model             : Raspberry Pi imx477
Color Matrix 1                  : 0.5130448341 -0.1162250116 -0.02393050492 -0.1746720523 0.9088459015 0.236192748 0.0329060331 0.1008179113 0.4090979695
As Shot Neutral                 : 0.3453533351 1 0.5775325894
Calibration Illuminant 1        : D65
CFA Pattern                     : [Blue,Green][Green,Red]
Image Size                      : 4056x3040
Megapixels                      : 12.3
Shutter Speed                   : 10
```

ISP output:
[image]

dcraw output:
[image]

naushir, 2024-09-18:
From the above results, it seems like the blue tint is present in the RAW image as well?  I assume ``dcraw`` is doing it's own processing for colours without relying on any EXIF data from the capture.

naushir, 2024-09-18:
Ive run the DNGs through [RawThreapee](https://www.rawtherapee.com/) and get the same results using the built-in AWB and get exactly the same results. It seems the blue tint is genuinely present in the RAW data.

naushir, 2024-09-18:
Can you share what power supply you are using on your Pi5?  Is it an official Raspberry Pi supply?

jfharvey, 2024-09-18:
Thanks nausir for your help on that issue.

I'm using a POE hat for the raspberry PI 5. 
[image]

naushir, 2024-09-18:
This is pure speculation, but I wonder if this is a power supply issue causing a glitch in the sensor.  Would you be able to run with a regular Raspberry Pi 5 power supply and see if it eliminates the problem?  You can probably just run this indoors for the test.

babsenfred, 2024-09-18:
I use the original Raspberry power supply and also have this issue.

naushir, 2024-09-18:
@babsenfred is this a Raspberry Pi 5 driver (https://www.raspberrypi.com/products/27w-power-supply/).

babsenfred, 2024-09-18:
Yes, that's him. (https://www.raspberrystore.nl/PrestaShop/raspberry-pi-v5/511-raspberry-pi-27w-usb-c-power-supply-white-eu-voor-raspberrypi-5-5056561803401.html)

naushir, 2024-09-18:
That looks like the correct supply.  If the blue tint is in the RAW data, the ISP processing is not going to be able to remove it.  I'm really not sure what more to look at...

It's very curious that this does not happen with Pi 4 though, the sensor runs exactly the same way on both platforms.

babsenfred, 2024-09-18:
I don't know if this is relevant but I was looking at the issues in the [INDI-Allsky Github page](https://github.com/aaronwmorris/indi-allsky/issues) (that is based on this build) and I couldn't find any colour issues.

babsenfred, 2024-09-19:
@jfharvey Yesterday I set the Auto-Gain and Auto-White-Balance to 'YES' in the nighttime settings. Since then I have no more 'blue' images... I see what happens tonight and I keep you posted.

sjtaubman, 2024-09-19:
> That looks like the correct supply. If the blue tint is in the RAW data, the ISP processing is not going to be able to remove it. I'm really not sure what more to look at...
> 
> It's very curious that this does not happen with Pi 4 though, the sensor runs exactly the same way on both platforms.

Thought I should mention that I am running Raspberry Pi 4 Model B Rev 1.5 and having the same problem with very blue images at dusk and dawn running the TJ Allsky software.

sjtaubman, 2024-09-19:
> @jfharvey Yesterday I set the Auto-Gain and Auto-White-Balance to 'YES' in the nighttime settings. Since then I have no more 'blue' images... I see what happens tonight and I keep you posted.

I have had the Auto-Gain and Auto-White-Balance set to "Yes" in the nighttime settings all along and get blue images, so this seems odd. My settings are attached here in case it is helpful. Thank you!

[Allsky Settings.pdf](https://github.com/user-attachments/files/17063888/Allsky.Settings.pdf)

aaronwmorris, 2024-09-19:
I am the author of indi-allsky and I can confirm I see the same "blue bias" behavior for the IMX477 with the latest packages on Rpi5, but only in jpeg mode.  I have not observed it in DNG raw mode.

What I found interesting is the blue images appeared to have an embossed/edge detection effect applied.  The contours/edges are highlighted in the resulting image.

Ignore the green images, that was when SCNR was disabled.  Everytime I switched to JPEG mode, it resulted in blue images.
[image]

Zoomed in on the effect.
[image]

naushir, 2024-09-20:
Would you be able to capture a DNG image that shows the error by adding a ``-r`` command line argument please?  This would be extremely useful to help us debug what's going on.  So far from the DNGs that I've seen, the blue tint might actually be present in the Bayer data from the sensor.

naushir, 2024-09-20:
I've been running the following script to see if I can capture this problem.  So far no luck. Could somebody try it out and see if they can capture a failure case and share the DNG/JPG and log file generated?
```bash
#!/bin/bash

while true; do

  ts=$(date "+%Y%m%d%H%M%S")
  LIBCAMERA_LOG_LEVELS=RPi*:0 rpicam-still --output "image-$ts.jpg" --timeout 1 --nopreview --width 4056 --height 3040 --shutter 600s --analoggain 8 --quality 100 -r 2>&1 | tee log_$ts.txt

  sleep 10
done
```

naushir, 2024-09-23:
I've used the above script with over 150000 captures over the weekend with a Pi 5 + IMX477.  Sadly, not one of the captured images shows the problem.  Has anybody been able to reproduce this with my script above?  The AWB logs would be really interesting to inspect when it goes wrong.

naushir, 2024-09-23:
I think we've now managed to reporduce this problem and it is indeed an issue with AWB.  The AWB does not have enough statistics to perform its search, and because the scene looks dark, it assumes indoor conditions, and sets the CT to 2500K.  In fact, even the good images fail AWB, but fail in a different way, and defaults the CT to around 4000K which looks about correct.

I suggest as a workaround switch to using manual WB gains.  Does anyone have an estimate of the colour temperature of the night sky?  It seems like for the "good" images, 4000K seems around correct, so  adding ``--awbgains 2.89,1.73`` to the command line ought to work, but note that these numbers are IMX477 specific.

It would be very useful for us to get some logs of the AWB algorithm in the "bad" case through.  If somebody could run my above script overnight and get these logs, it would be incredibly useful to us.  Thanks!

EricClaeys, 2024-09-24:
@naushir,
Of the 680 images I took using your script, 61 were good, i.e., "normal" color, and the rest were blue.  There appears to be no rhyme or reason to it.

[image]

Attached are 4 sets of files (dng, jpg, log).
`good-1.zip` contains a "normal" looking file (ignore the out of focus).
`blue-1.zip` is the next image, which is blue
Ditto for `good-2.zip` and `blue-2.zip`.

I use 2.5 for red and 2.0 for blue, but as we tell users, there is no "right" answer.  It depends on their sky conditions and the dome they use to cover their camera.

[good-1.zip](https://github.com/user-attachments/files/17109584/good-1.zip)
[blue-1.zip](https://github.com/user-attachments/files/17109590/blue-1.zip)
[good-2.zip](https://github.com/user-attachments/files/17109593/good-2.zip)
[blue-2.zip](https://github.com/user-attachments/files/17109598/blue-2.zip)

naushir, 2024-09-24:
As a test, could you add ``"delta_limit": 2`` to the ``rpi.awb`` section in the tuning file (normally located at ``/usr/share/libcamera/ipa/rpi/pisp/imx477.json`` and see if that will help?

EricClaeys, 2024-09-27:
@nashir,
Not sure if it helps or not.  All the pictures are less saturated.
[image]

I can upload some images in a few days if needed.

naushir, 2024-09-27:
The ``delta_limit`` change is not a full solution - but it does help reduce the occurrence of the original problem.  I have a patch for the AWB to better handle failure cases as is what's happening here.  This should hopefully eliminate the problem.

However given your use case for Allsky, I feel that perhaps manual white balance be the most robust solution.

perkaer, 2024-11-05:
Just pitching in. I did a fresh install of Allsky and PI OS 2 weeks ago, although the rpicam build are not that new:

rpicam-apps build: 49344f2a8d18 17-06-2024 (12:09:07)
libcamera build: v0.3.0+65-6ddd79b5

I use manual WB and I get the blue images :/

Do we expect the issue to be fixed on newer build of rpicam?

naushir, 2024-11-06:
The next update of the libcamera packages will have the following commit to address this: https://github.com/raspberrypi/libcamera/commit/ea8fd63d936f19bfc773199b9fe81e8d5d1f709b

Note that you will need to update your tuning files to provide a fallback colour temperature when the search fails to avoid the blue tint.

naushir, 2024-11-06:
> I use manual WB and I get the blue images :/

What gains are you using here?  Could you try targeting something like 4500K?

perkaer, 2024-11-10:
> > I use manual WB and I get the blue images :/
> 
> What gains are you using here? Could you try targeting something like 4500K?

Just to make sure I undestand, if I use manual WB, there should be no WB "search" and the colour temperature should not jump around as I observe? At least this is what I am used to from my mirrorless :) However this is not consistent with what I observe I guess.

I have observed the blue images both for a fixed gain of 8 and autogain settings, examples are gain of 4.86, 4.33. 

Where can I read more about the camera tuning file, so I can aim for a color temp of 4500 K?

davidplowman, 2024-11-11:
You can read about the tuning file [here](https://datasheets.raspberrypi.com/camera/raspberry-pi-camera-guide.pdf). However, it doesn't tell you about specific values for specific sensors. You might do better do look at the tuning file and find the `ct_curve` entry, for example like [this](https://github.com/raspberrypi/libcamera/blob/main/src/ipa/rpi/vc4/data/imx477.json#L122). 

That line is telling your that the normalised red and blue values at 4660K are  0.3529 and 0.68. You should use the reciprocals of those values for the red and blue gains.

Finally, when you have a DNG file, check the "As shot neutral" values in the metadata. It should be those red and blue value given above. If not, then something must have changed them.

naushir, 2024-11-11:
Also note that 4500K is just an educated guess on my part.  If possibly you should measure the actual conditions in your environment and chose an appropriate fixed colour temperature.

naushir, 2024-11-28:
The latest software release add support for an awb bias search term.  To use it, add the following keys/values to the ``rpi.awb`` algorithm section in the appropriate camera json tuning file:

```
"bias_proportion": 0.1,
"bias_ct": 4500,
```

In the above I've used 4500K as the bias target, but you should set this to the appropriate colour temperature measured in your environment.

closed 2024-11-28 by naushir

perkaer, 2024-11-28:
Thanks for the update!

I have yet to fully understand how changing the auto-white balance settings, can help on this issue, when I am using fixed white balance. But there are for sure details I do not understand!

EricClaeys, 2024-11-28:
@naushir,
Thanks for the update.
Do people get the update via `apt-get libcamera-apps` or something else?
What cameras does this apply to, what PiOS release(s), and what Pi models?
Can the changes to the tuning file be made to the files in the release?  I think most of our allsky users will have no idea how to add those lines, so I'm looking for an easier, more foolproof way to do this.  If the lines can be added to the files regardless of whether or not the user has the problem, and you are not able to add them to the release files, we could have the allsky installation script add them, but that is not a viable solution long term since users often change cameras.

Thoughts?

naushir, 2024-11-29:
>  Do people get the update via `apt-get libcamera-apps` or something else?

Its the libcamera libraries that have been updated with this feature.  So ``sudo apt update && sudo apt full-upgrade`` to upgrade all libraries is probably advisable.

> What cameras does this apply to, what PiOS release(s), and what Pi models?

This is an algorithm change, so will apply to any supported camera, if enabled in the tuning file.

>Can the changes to the tuning file be made to the files in the release? I think most of our allsky users will have no idea how to add those lines, so I'm looking for an easier, more foolproof way to do this. If the lines can be added to the files regardless of whether or not the user has the problem, and you are not able to add them to the release files, we could have the allsky installation script add them, but that is not a viable solution long term since users often change cameras. And requiring the script to work around an issue in another package is, well, a work around.

This is a tricky one.  I get the issue with having scripts do this, but from our end we try as had as possible not to introduce behavioral changes to algorithms, which this is.  Perhaps we can in this case, but I need to discuss with @davidplowman.

I did have one question,  This issue here can be mitigated with using manual white balance settings.  In the Allsky software GUI, there seems to be a control to set a manual colour temperature.  However, libcamera does not have a direct way of setting colour temperature (yet).  How is the conversion from colour temperature to red/blue gains done on a per-sensor basis?

EricClaeys, 2024-12-01:
@naushir,

The Allsky software has these settings for color cameras:
[image]

In manual AWB it simply passes to libcamera-still what the user enters for Red and Blue: `--awbgains RED,BLUE`.
In auto AWB it simply passes `--awb auto`.
Users look at their images and manually adjust the red and blue (separate settings for daytime, where we suggest auto AWB, and nighttime, where we suggest manual AWB).  Do you have a better way to determine good red and blue numbers?  Something that non-technical people could use?

Users can add any other libcamera-still arguments they want:
[image]

If we find that a fair number of users are entering the same Extra Arguments, for example, for auto-focus cameras, we'll add settings specific to them.

naushir, 2024-12-02:
Ideally you should select the target colour temperature and look up the gains on the CT curve defined in the camera json file.

For example, in the IMX477 case, the curve can be found [here](https://github.com/raspberrypi/libcamera/blob/1230f78d2f6d38d812568fd27f7b985b9ff19e39/src/ipa/rpi/vc4/data/imx477.json#L115).  Assuming you are targeting 3600K, the R gain is the inverse of the second column, ``2.35``,  and the B gain is the inverse of the third column, ``1.79``. 

As mentioned earlier, we are working with the libcamera team (pinging @kbingham) to add a colour temperature control so that uses can directly provide the CT value rather than gains.  This would avoid the manual lookup and is much more user friendly.