raspberrypi/rpicam-apps #717: [BUG]1.5.1 fails to build on PI Zero 2W

LeBirlante, 2024-09-16:
Building rpicam-apps in the past always worked in this system, but trying to build 1.5.1 now fails to compile.

Meson setup:

`meson setup build -Denable_libav=enabled -Denable_drm=enabled -Denable_egl=disabled -Denable_qt=disabled -Denable_opencv=enabled -Denable_tflite=disabled`

Result:

`rpicam-apps 1.5.1`

  `libcamera`
`    location             : /usr/lib`
`    version              : 0.3.1`

`  Build configuration`
`    libav encoder        : YES`
`    drm preview          : YES`
`    egl preview          : NO`
`    qt preview           : NO`
`    OpenCV postprocessing: YES`
`    TFLite postprocessing: NO`
`    Hailo postprocessing : NO`

`  User defined options`
`    enable_drm           : enabled`
`    enable_egl           : disabled`
`    enable_libav         : enabled`
`    enable_opencv        : enabled`
`    enable_qt            : disabled`
`    enable_tflite        : disabled`
`    neon_flags           : armv8-neon`

`Found ninja-1.12.1 at /usr/bin/ninja`

Then we try,

`meson compile -C build -j 1`

And here it goes:

`INFO: autodetecting backend as ninja`
`INFO: calculating backend command to run: /usr/bin/ninja -C /home/varet/rpicam-apps/build -j 1`
`ninja: Entering directory `/home/varet/rpicam-apps/build'`
`[5/49] Compiling C++ object rpicam_app.so.1.5.1.p/core_rpicam_app.cpp.o`
`FAILED: rpicam_app.so.1.5.1.p/core_rpicam_app.cpp.o `
`c++ -Irpicam_app.so.1.5.1.p -I. -I.. -Icore -I/usr/include/libcamera -I/usr/include -I/usr/include/libpng16 -I/usr/include/libdrm -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -O3 -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -mfpu=neon-fp-armv8 -ftree-vectorize -DLIBAV_PRESENT=1 -DLIBDRM_PRESENT=1 -fPIC -pthread -DBOOST_ALL_NO_LIB -MD -MQ rpicam_app.so.1.5.1.p/core_rpicam_app.cpp.o -MF rpicam_app.so.1.5.1.p/core_rpicam_app.cpp.o.d -o rpicam_app.so.1.5.1.p/core_rpicam_app.cpp.o -c ../core/rpicam_app.cpp`
`../core/rpicam_app.cpp: In member function ‘void RPiCamApp::StartCamera()’:`
`../core/rpicam_app.cpp:640:83: error: ‘ScalerCrops’ is not a member of ‘libcamera::controls::rpi’`
`  640 |         if (!controls_.get(controls::ScalerCrop) && !controls_.get(controls::rpi::ScalerCrops))`
`      |                                                                                   ^~~~~~~~~~~`
`../core/rpicam_app.cpp:668:46: error: ‘ScalerCrops’ is not a member of ‘libcamera::controls::rpi’`
`  668 |                 controls_.set(controls::rpi::ScalerCrops, libcamera::Span<const Rectangle>(crops.data(), crops.size()));`

First time I've run across this compile error... any ideas about what could I be doing wrong here?

thanks for your time,

naushir, 2024-09-16:
You need to have the latest libcamera-dev package installed. Or if you are building manually, please ensure you build libcamera from the latest source at https://github.com/raspberrypi/libcamera.

LeBirlante, 2024-09-17:
Thanks for your response. 

I think I'm already using latest libcamera:

-------------------------------------------------
`pacman -Qi libcamera`

`Name            : libcamera`
`Version         : 0.3.1-2`
`Description     : A complex camera support library for Linux, Android, and ChromeOS`
`Architecture    : armv7h`
`URL             : https://libcamera.org/`
`Licenses        : Apache-2.0  CC0-1.0  GPL-2.0-only WITH Linux-syscall-note  GPL-2.0-or-later  LGPL-2.1-or-later  GPL-2.0-or-later WITH Linux-syscall-note OR BSD-3-Clause  GPL-2.0-or-later WITH Linux-syscall-note OR MIT`
`Groups          : None`
`Provides        : libcamera.so=0.3-32  libcamera-base.so=0.3-32`
`Depends On      : gcc-libs  glibc  gnutls  libcamera-ipa  libelf  libunwind  libyaml  sh  systemd-libs  libudev.so=1-32`
`Optional Deps   : gst-plugin-libcamera: GStreamer plugin`
`                  libcamera-docs: for documentation [installed]`
`                  libcamera-tools: for applications [installed]`
`Required By     : libcamera-ipa  libcamera-tools`

`Installed Size  : 2.49 MiB`
`Packager        : Arch Linux ARM Build System <builder+xu2@archlinuxarm.org>`
`Build Date      : Mon Aug 5 20:32:29 2024`
`Install Date    : Tue Aug 13 16:58:45 2024`
`Install Reason  : Installed as a dependency for another package`
`Install Script  : No`
`Validated By    : Signature`
-----------------------------------------------------

Next, I follow the docs at https://www.raspberrypi.com/documentation/computers/camera_software.html#building-rpicam-apps-without-building-libcamera , since as seen above, I am not manually building libcamera. Also, libcamera-dev is not available as a package for my distro (Arch Linux ARM), so I have never installed it.

The thing is, this worked perfectly in the past (installing libcamera package, and then manually compiling rpicam-apps), but now it throws the compile error shown in my first post. Only thing I can think of, is that somehow the arch linux package 'libcamera' has some kind of problem, or perhaps is it not the latest version?

naushir, 2024-09-17:
```
URL : https://libcamera.org/
```
This suggests you are not using libcamera from  https://github.com/raspberrypi/libcamera.  Upstream libcamera does not yet have Pi 5  + ScalerCrops support.  Until this has been merged (we are actively working on this with the libcamera team), you need to use our repo for rpicam-apps to work correctly.

naushir, 2024-09-17:
It could also be that you don't have the latest libcamera dev packages in your system.  Have you done an ``apt update``  recently?  We made a new camera release one or two weeks ago that added this ScalerCrops feature.

LeBirlante, 2024-09-17:
> ```
> URL : https://libcamera.org/
> ```
>
> This suggests you are not using libcamera from https://github.com/raspberrypi/libcamera. Upstream libcamera does not yet have Pi 5 + ScalerCrops support. Until [...]

That seems to be the thing.... as you say, I am using a package built directly from libcamera.org, not from your repository. I use that package because building libcamera on a low ram device like the pi zero 2W gives a lot of headaches (OOM even when passing -j 1 to the ninja build command).

But now, knowing that you are about to merge with libcamera upstream, then I will simply wait for that to happen, as there's no hurry in my current project.

Thanks a lot for pointing me out my problem!

closed 2024-09-17 by LeBirlante