raspberrypi/rpicam-apps #718: [BUG] Failed to allocate buffers

Oxize, 2024-09-18:
**Describe the bug**
`
Failed to allocate buffers`

**Bug report**

`[23:50:08.514187869] [4304] �[1;31mERROR �[1;37mV4L2 �[1;34mv4l2_videodevice.cpp:1241 �[0;32m/dev/video0[12:cap]: �[0mUnable to request 1 buffers: Cannot allocate memory ERROR: *** failed to allocate capture buffers **`

**Additional context**
Error generated within Allsky software. This error is rpi cam related which we cant solve. It has been adressed by alot more ppl in the past. Please pick it up again. 

#555

** Hardware

RPI 4B 8GB with HQ Camera. Cables good attached, no throtteling when it happens, no memory leakage, no overheating.

naushir, 2024-09-18:
Please can you give more details on this.  There could be several reasons for your error.  What ``rpicam-apps`` command line are you running?  Have you modified anything in ``config.txt`` or any other configuration file?  Can you reproduce this error outside of the allsky software?

Oxize, 2024-09-20:
It uses libcamerastill in CL. I dont have modified anything in config.txt. Looks to me standard file.

No i cannot reproduce this error outside allysky.

naushir, 2024-10-07:
Any update on this?  If not I'll close shortly.

zanco, 2024-11-03:
Hi, hope it's ok to reply to this. I have ran into the same problem , with the same software, and I did run the command you asked. Below the error and the output of you requested command:

`Mode selection for 2028:1520:12:P
    SRGGB10_CSI2P,1332x990/0 - Score: 3456.22
    SRGGB12_CSI2P,2028x1080/0 - Score: 1083.84
    SRGGB12_CSI2P,2028x1520/0 - Score: 0
    SRGGB12_CSI2P,4056x3040/0 - Score: 887
Mode selection for 4056:3040:12:P
    SRGGB10_CSI2P,1332x990/0 - Score: 10552.2
    SRGGB12_CSI2P,2028x1080/0 - Score: 8179.84
    SRGGB12_CSI2P,2028x1520/0 - Score: 7096
    SRGGB12_CSI2P,4056x3040/0 - Score: 0
dmaHeap allocation failure for rpicam-apps0
ERROR: *** failed to allocate capture buffers for stream ***
pe2bz@allsky:~ $ grep -i cma /proc/meminfo
CmaTotal:         524288 kB
CmaFree:          289372 kB
`

NeilAllison147, 2025-05-07:
I'm experiencing the same issue with AllSky so I'd be interested in the solution

naushir, 2025-08-14:
There was no fix for this issue - depending on what else is running in the system and using resources, heap memory may be limited, and running with large cameras we may not be able to allocate the framebuffers required.

There are ways to tweak the number of framebuffers used (``--buffers`` and ``--viewfinder-buffers``), but this is going to be on a case by case basis.

naushir, 2025-08-14:
> Should users who have this problem create new Issues so you (or someone else) can guide them through fixing it?

Yes, that's probably best - it will be very use case dependent and there won't be a generic solution.