raspberrypi/rpicam-apps #721: [BUG]

EwingKang, 2024-09-23:
**Describe the bug**
Start to experience high rate of frame dropping after latest update (`v0.3.1+50-69a894c4`)

**Bug report**
[bugreport0.txt](https://github.com/user-attachments/files/17093715/bugreport0.txt)
[bugreport1.txt](https://github.com/user-attachments/files/17093714/bugreport1.txt)

**Additional context**
I'm using `rpicam-vid` to capture 2 cameras at the same time on a RPI5. I needed these two cameras to capture video in sync. Doing it by calling two instance of `rpicam-vid` does the job pretty well before without framedrop.
```bash
rpicam-vid --camera 0 $VIDOPT -o ${HOSTNAME}_cam0_${SUFFIX1}_${SUFFIX2} &> log_${HOSTNAME}cam0_$SUFFIX1_$DATETIMEFULL.txt &
rpicam-vid --camera 1 $VIDOPT -o ${HOSTNAME}_cam1_${SUFFIX1}_${SUFFIX2} &> log_${HOSTNAME}cam1_$SUFFIX1_$DATETIMEFULL.txt
```
I'm doing it with a very save setting of 1640*1232@15 FPS on a IMX219. (Previously I have tested the setting up to 1640*1232@25 fps without any issue) At this setting, the total CPU usage is < 65%
After the update `v0.3.1+50-69a894c4` random framedrop starts to appear quite frequently. The same 45 second, 15FPS video no longer gives a stable 668 frames, but varying between 668, 652, 659, 660, 663 frames.
I dropped the framerate to as low as 5, and tested with different bitrate, still seeing varying framerate.

naushir, 2024-09-23:
Can you post a clip of the captured video showing frame drops at 15fps that I can inspect please?

EwingKang, 2024-09-24:
@naushir Thanks for your response.
These are two videos comming from two RPIs. The left is the one with 660 frames, the right one is the 668 frames. I triggered capturing both rpi to within 10 ms accuracy. 
The cellphone screen is a **count down** timer. For 15 FPS, there should be 66 ms between each frame.

Here are 3 consecutive frames. Frame dropping occured between frame 2 and 3.
* 39.726
    [image]
* 39.792
    [image]
* 39.859
    [image]
* The missing 8 frames (only stepping the right camera frames)
    [image]
    [image]
    [image]
    [image]
    [image]
    [image]
    [image]
    [image]
* "Catch-up", notice the timestamp difference
    [image]

naushir, 2024-09-24:
Would you be able to post the actual video files so I can have a look at all the frame timestamps please?

EwingKang, 2024-09-24:
@naushir I can only provide the camera that has missing frames at [this link](https://fico365-my.sharepoint.com/:v:/g/personal/ewing_kang_fic_com_tw/EdtJ-wAhaQRGl34ODGayaCoB8-3dX8PaHJJDRq9ogNTuDw?nav=eyJyZWZlcnJhbEluZm8iOnsicmVmZXJyYWxBcHAiOiJPbmVEcml2ZUZvckJ1c2luZXNzIiwicmVmZXJyYWxBcHBQbGF0Zm9ybSI6IldlYiIsInJlZmVycmFsTW9kZSI6InZpZXciLCJyZWZlcnJhbFZpZXciOiJNeUZpbGVzTGlua0NvcHkifX0&e=fduP69)
Can't promise how long I can keep this link alive, though

naushir, 2024-09-24:
From the video you linked, I only see a single frame glitch, this is at approx frame 600, where the system seems to stall for approx 600ms.  Every other frame time is between 66.6 - 67ms apart as you would expect.  This glitch produces a large set of frame drops, but I cannot see any frame drops in any of the preceding frames:

[image]

You can use [this](https://github.com/raspberrypi/rpicam-apps/blob/main/utils/timestamp.py) script to verify your results.  Perhaps some other effect is causing what you are seeing?

naushir, 2024-09-24:
Are you saving to an sdcard or a faster medium? Wear on the sdcard may cause slowdowns like this when the system decides the flush the data into the card.  Another thing to try would be to add ``force_turbo=1`` to ``/boot/firmware/config.txt`` at the bottom of the file.  This will ensure no clock throttling will happen when you are running.

EwingKang, 2024-09-25:
@naushir 
It seems like the culprit is, indeed, the SD card degradation. 
I can no longer see the frame-dropping behavior after setting the destination to an external SSD drive. Due to space limitations, I'll probably have to use ramdisk (tmpfs).
Thank you very much for the help!
I'm perplexed by the coincidence of the event! All 3 Pis start showing such behavior at the same time. Maybe the update happens to make my Kingston SD card reach its degradation limit. ;(

* Original behavior
    [image]

* By using ramdisk (tmpfs) as described [here](https://askubuntu.com/questions/597268/virtual-ram-folder-in-ubuntu)
    [image]

P.S. It Seems like the ram-disk solution causes my CPU usage to become very high from the original 60% to >95%. I'll probably have to get an SSD Hat or something. Thanks for the help again.

closed 2024-09-25 by EwingKang