raspberrypi/rpicam-apps #726: Bug report camera-bug-report on picrawler

rkapuaala, 2024-10-03:
after running the scripts in picrawler examples, I noticed that the ones that had issues had issues because of the camera. After running libcamera-hello, I realised the camera wasn't even being read. So I ran camera-bug-report and it generated the attached file.

[bug-report.txt](https://github.com/user-attachments/files/17238677/bug-report.txt)

naushir, 2024-10-03:
I'm not familiar with picarawler or what it does with cameras, but the system has not detected any cameras attached to your Pi:
```
--------------------------------------------------------------------------------
Cameras
--------------------------------------------------------------------------------
** rpicam-hello --list-cameras **
No cameras available!
```
What sensor do you have attached to your platform?

rkapuaala, 2024-10-03:
> What sensor do you have attached to your platform?
It should be in the bug report, but jus to be brief, the Ultra Sonic sensor and the pi camera are the only two I know about.  Did I put this report in the right place???

naushir, 2024-10-03:
The report seems to be correct.  Which pi camera exactly have you connected?  Is it an official Raspberry Pi camera?

rkapuaala, 2024-10-03:
Nashir, I have what came with the picrawler kit from SunFounder. I'm not actually sure if that is official. I'm pretty new to this stuff, I bought the kit less than a week ago. That being said; the kit arrived with a second hat and a second ribon cable. I'm guessing they knew this might happen. This morning when I got up, I swap the original hat and the original ribon for the spares, and now. It is working perfectly. And when I say perfectly I mean even the sound (which was not functioning) is working now. So, I think I can call this matter rsolved. Now, I'm just waiting for them to send me a new servo.

naushir, 2024-10-03:
Thanks for the update.  Closing now.

closed 2024-10-03 by naushir