raspberrypi/rpicam-apps #734: Compatibility issue

Sgnarf, 2024-10-19:
I'm trying to build rpicam-apps but it seems like there's an issue with libcamera. Below the error. I've got libcamera 0.3.2 installed, on Ubuntu 24.04, on a Raspberry 5

ninja
[1/50] Compiling C++ object post_processing_stages/imx500/imx500-postproc.so.p/imx500_posenet.cpp.o
FAILED: post_processing_stages/imx500/imx500-postproc.so.p/imx500_posenet.cpp.o 
c++ -Ipost_processing_stages/imx500/imx500-postproc.so.p -Ipost_processing_stages/imx500 -I../post_processing_stages/imx500 -I. -I.. -I/usr/local/include/libcamera -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -O3 -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -ftree-vectorize -DLIBAV_PRESENT=1 -DLIBDRM_PRESENT=1 -DLIBEGL_PRESENT=1 -DQT_PRESENT=1 -fPIC -MD -MQ post_processing_stages/imx500/imx500-postproc.so.p/imx500_posenet.cpp.o -MF post_processing_stages/imx500/imx500-postproc.so.p/imx500_posenet.cpp.o.d -o post_processing_stages/imx500/imx500-postproc.so.p/imx500_posenet.cpp.o -c ../post_processing_stages/imx500/imx500_posenet.cpp
../post_processing_stages/imx500/imx500_posenet.cpp: In member function ‘virtual bool PoseNet::Process(CompletedRequestPtr&)’:
../post_processing_stages/imx500/imx500_posenet.cpp:584:70: error: ‘CnnOutputTensor’ is not a member of ‘libcamera::controls::rpi’
  584 |         auto output = completed_request->metadata.get(controls::rpi::CnnOutputTensor);
      |                                                                      ^~~~~~~~~~~~~~~
../post_processing_stages/imx500/imx500_posenet.cpp: At global scope:
../post_processing_stages/imx500/imx500_posenet.cpp:169:20: error: ‘std::vector<float> {anonymous}::format_tensor(const float*, unsigned int, unsigned int)’ defined but not used [-Werror=unused-function]
  169 | std::vector<float> format_tensor(const float *data, unsigned int size, unsigned int div)
      |                    ^~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
[2/50] Compiling C++ object post_processing_stag...postproc.so.p/imx500_post_processing_stage.cpp.o
FAILED: post_processing_stages/imx500/imx500-postproc.so.p/imx500_post_processing_stage.cpp.o 
c++ -Ipost_processing_stages/imx500/imx500-postproc.so.p -Ipost_processing_stages/imx500 -I../post_processing_stages/imx500 -I. -I.. -I/usr/local/include/libcamera -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -O3 -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -ftree-vectorize -DLIBAV_PRESENT=1 -DLIBDRM_PRESENT=1 -DLIBEGL_PRESENT=1 -DQT_PRESENT=1 -fPIC -MD -MQ post_processing_stages/imx500/imx500-postproc.so.p/imx500_post_processing_stage.cpp.o -MF post_processing_stages/imx500/imx500-postproc.so.p/imx500_post_processing_stage.cpp.o.d -o post_processing_stages/imx500/imx500-postproc.so.p/imx500_post_processing_stage.cpp.o -c ../post_processing_stages/imx500/imx500_post_processing_stage.cpp
../post_processing_stages/imx500/imx500_post_processing_stage.cpp: In member function ‘virtual bool IMX500PostProcessingStage::Process(CompletedRequestPtr&)’:
../post_processing_stages/imx500/imx500_post_processing_stage.cpp:146:69: error: ‘CnnInputTensor’ is not a member of ‘libcamera::controls::rpi’
  146 |         auto input = completed_request->metadata.get(controls::rpi::CnnInputTensor);
      |                                                                     ^~~~~~~~~~~~~~
[4/50] Compiling C++ object post_processing_stag...x500-postproc.so.p/imx500_object_detection.cpp.o
FAILED: post_processing_stages/imx500/imx500-postproc.so.p/imx500_object_detection.cpp.o 
c++ -Ipost_processing_stages/imx500/imx500-postproc.so.p -Ipost_processing_stages/imx500 -I../post_processing_stages/imx500 -I. -I.. -I/usr/local/include/libcamera -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -O3 -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -ftree-vectorize -DLIBAV_PRESENT=1 -DLIBDRM_PRESENT=1 -DLIBEGL_PRESENT=1 -DQT_PRESENT=1 -fPIC -MD -MQ post_processing_stages/imx500/imx500-postproc.so.p/imx500_object_detection.cpp.o -MF post_processing_stages/imx500/imx500-postproc.so.p/imx500_object_detection.cpp.o.d -o post_processing_stages/imx500/imx500-postproc.so.p/imx500_object_detection.cpp.o -c ../post_processing_stages/imx500/imx500_object_detection.cpp
../post_processing_stages/imx500/imx500_object_detection.cpp: In member function ‘virtual bool ObjectDetection::Process(CompletedRequestPtr&)’:
../post_processing_stages/imx500/imx500_object_detection.cpp:131:70: error: ‘CnnOutputTensor’ is not a member of ‘libcamera::controls::rpi’
  131 |         auto output = completed_request->metadata.get(controls::rpi::CnnOutputTensor);
      |                                                                      ^~~~~~~~~~~~~~~
../post_processing_stages/imx500/imx500_object_detection.cpp:132:68: error: ‘CnnOutputTensorInfo’ is not a member of ‘libcamera::controls::rpi’
  132 |         auto info = completed_request->metadata.get(controls::rpi::CnnOutputTensorInfo);
      |                                                                    ^~~~~~~~~~~~~~~~~~~
[6/50] Compiling C++ object post_processing_stages/core-postproc.so.p/hdr_stage.cpp.o
ninja: build stopped: subcommand failed.

naushir, 2024-10-21:
This is likely because you are not using libcamera from [our repo](https://github.com/raspberrypi/libcamera), and some of our newer/more advanced features have not yet been upstreamed into libcamera.

If you build libcamera from the above repo, rpicam-apps should build correctly again.

naushir, 2024-10-31:
Closing now.  If you still have issue please feel free to reopen.

closed 2024-10-31 by naushir