raspberrypi/rpicam-apps #735: [BUG] ERROR: unrecognised option '--libav-format'

ben-xD, 2024-10-19:
**Describe the bug**
Reading the current docs and github issues, I see commands like `rpicam-vid --codec libav --libav-format avi --libav-audio --output example.avi
` ([link](https://www.raspberrypi.com/documentation/computers/camera_software.html#libav-integration-with-rpicam-vid)).

However, this doesn't work, I get `ERROR: *** unrecognised option '--libav-format' ***`

## Versions
```
$rpicam-vid --version
rpicam-apps build: v1.5.2 a8ccf9f3cd9d 03-10-2024 (22:33:20)
libcamera build: v0.3.2+27-7330f29b
```

ben-xD, 2024-10-19:
Maybe it's related to me using a headless version. Judging by the advanced docs in building rpicam-apps, perhaps `Denable_libav=disabled` was set for the library. So I'm building it locally with the flag enabled.

naushir, 2024-10-21:
> Maybe it's related to me using a headless version raspberry pi os. Judging by the advanced docs in building rpicam-apps, perhaps `Denable_libav=disabled` was set for the library. So I built it locally with the flag enabled.

If you installed the ``lite`` OS variant, you won't get the libav libraries installed.  You can change this by doing something like:

```
sudo apt purge rpicam-apps-lite
sudo apt install rpicam-apps
```

and that should install all the libav libraries and rpicam-apps support.

ben-xD, 2024-10-26:
Thanks I haven't been able to test it because my RPi5 was bricked after installing and building some libraries (it stopped turning on the next day). A replacement has been sent and ill test it soon :)

naushir, 2024-11-01:
Closing now. Feel free to reopen if you run into any issues.