raspberrypi/rpicam-apps #736: [BUG] rpicam-still `--timestamp` option is overwritting files

bernardolansing, 2024-10-21:
**Describe the bug**
Unless I'm being stupid, I think there's a "bug" with `--timestamp` option for rpicam-still. When used along with `--timelapse`, it'll save every frame to a file which name is the capture timestamp for that frame. It does that indeed, but the recorded timestamp is in seconds, not in milliseconds. This impedes me from shoting timelapses with less that one second of interval.

If I do this:
`rpicam-still --timelapse 200 --width 680 --height 480 --timestamp`
I'll get five files in my directory:
`1728588549.jpg  1728588550.jpg  1728588551.jpg  1728588552.jpg  1728588553.jpg`
And that's because I'm using the default timeout of five seconds.

However, calling it like this:
`rpicam-still --timelapse 200 --width 680 --height 480 --output frame%04d.jpg`
Will produce seven frames:
`frame0000.jpg  frame0001.jpg  frame0002.jpg  frame0003.jpg  frame0004.jpg  frame0005.jpg  frame0006.jpg`

I've managed to capture seven frames within the same time span. That means that `--timestamp` will cause frames shot in the same second to be overwritten. I presume we can't change `--framerate` behavior now because it'd break existing implementations; but we could get a second option like `--framerate-millis`.

**Bug report**
[bug-report.txt](https://github.com/user-attachments/files/17466943/bug-report.txt)

naushir, 2024-10-22:
Unfortunately I think this is unworkable as the standard C/C++ time libraries only return seconds level precision.  Perhaps an alternative is to append ``_1/_2/_3`` or similar in your case where the timelapse value is less than a second?

naushir, 2024-10-22:
No there are ways, but it involves much more invasive changes.  Perhaps you could describe your use case in more detail?  Would it make sense to instead take an MJPEG (or similar) stream running at 0.5 fps to achieve the same thing?

naushir, 2024-11-01:
Closing for now then.  I'll consider what we can do for future releases.

closed 2024-11-01 by naushir