raspberrypi/rpicam-apps #739: [BUG] Building from outside the repository folder

fabh2o, 2024-11-01:
**Describe the bug**
(Not really sure if it's a bug, since we are supposed to run the meson commands inside the repo folder)

I'm trying to create an archlinux package. The build directory is not inside the repository folder. The process is more or less like this:
```bash
mkdir src; cd src
git clone https://github.com/raspberrypi/rpicam-apps.git
meson setup --prefix=/usr --buildtype=plain build rpicam-apps -D enable_libav=disabled -D enable_dr
m=enabled -D enable_egl=disabled -D enable_qt=disabled -D enable_opencv=disabled -D enable_tflite=disabled -D enable_hailo=disabled
# $ ls
# build/  rpicam-apps/
meson compile -C build
```

I can not build the package due to the error:
```bash
[24/42] Compiling C++ object post_processing_stages/core-postproc.so.p/hdr_stage.cpp.o
FAILED: post_processing_stages/core-postproc.so.p/hdr_stage.cpp.o
c++ -Ipost_processing_stages/core-postproc.so.p -Ipost_processing_stages -I../rpicam-apps/post_processing_stages -I.. -I/usr/local/include/libcamera -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -Wall -Winvalid-pch -Wextra
 -Wpedantic -Werror -std=c++17 -pedantic -Wno-unused-parameter -faligned-new -Wno-psabi -ftree-vectorize -fPIC -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -ftree-vectorize -DLIBDRM_PRESENT=1
-MD -MQ post_processing_stages/core-postproc.so.p/hdr_stage.cpp.o -MF post_processing_stages/core-postproc.so.p/hdr_stage.cpp.o.d -o post_processing_stages/core-postproc.so.p/hdr_stage.cpp.o -c ../rpicam-apps/post_processing_st
ages/hdr_stage.cpp
../rpicam-apps/post_processing_stages/hdr_stage.cpp:27:10: fatal error: core/rpicam_app.hpp: No such file or directory
   27 | #include "core/rpicam_app.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
Same outcome for other files (`post_processing_stages/core-postproc.so.p/motion_detect_stage.cpp.o`, `post_processing_stages/core-postproc.so.p/negate_stage.cpp.o`)

Changing this include from `'../..'` to `'..'` fixes the issue and i can build the package succesfully:
https://github.com/raspberrypi/rpicam-apps/blob/6de1ab6a899df35f929b2a15c0831780bd8e750e/post_processing_stages/meson.build#L33C35-L33C65

Note that this change does not affect building from inside the repo folder (as indicated in the build instructions) so it should be possible to change the relative path and build both from inside of the repo folder and from outside it.

I'm building the headless version of the package, maybe enabling other options breaks building for other files but i think that the fix still remains the same.

naushir, 2024-11-04:
Have you tried using ``meson dist --no-tests`` to create a zip package of the source?  We do this for building our packages and everything seems to work as epxected.

naushir, 2024-11-05:
One thing to try is to make the following change to the meson scripts and see if it works:
```diff
diff --git a/meson.build b/meson.build
index b57d9b12ee3b..f4834ac575d0 100644
--- a/meson.build
+++ b/meson.build
@@ -79,7 +79,7 @@ rpicam_app = library(
     'rpicam_app',
     rpicam_app_src,
     soversion : meson.project_version(),
-    include_directories : include_directories('.'),
+    include_directories : include_directories(meson.project_source_root()),
     install : true,
     name_prefix : '',
     dependencies : rpicam_app_dep,
```

naushir, 2024-11-05:
Actually please ignore the above, I think the correct fix is to replace ``../../`` with ``../`` as you suggested.

closed 2024-11-05 by naushir