raspberrypi/rpicam-apps #740: [Question] 600ms latency when streaming RTP?

ben-xD, 2024-11-02:
Im trying to stream video from a Pi 5 to my mac laptop.

I tried to run the example: `rpicam-vid -t 0 --codec libav --libav-format mpegts --libav-audio -o "tcp://<ip-addr>:<port>?listen=1"
`, but that error'd with `libav: cannot open pulse input device default`, so I removed the `--libav-audio` which fixed it.

This command works: `rpicam-vid -t 0 --codec libav --libav-format mpegts -o "udp://192.168.1.82:11024"`

However, when I play it on my mac with `ffplay udp://192.168.1.82:11024 -fflags nobuffer -flags low_delay -framedrop`, the latency is more than 600ms.

How can I reduce the latency further? I've used gstreamer with other cameras that have given me latencies of 250ms on the same network.

naushir, 2024-11-04:
I wonder if this is because of b-frame encoding.  Could you try disabling bframes by editing the following out:
https://github.com/raspberrypi/rpicam-apps/blob/2b61eca445158c57b78ca08263ecdbebe8740bbf/encoder/libav_encoder.cpp#L83

and see if that helps?

naushir, 2024-11-21:
Any update on this?  If not I will close down shortly.

ben-xD, 2024-11-21:
Sorry, I have not yet built this app before but ill do it today. Thanks!

ben-xD, 2024-11-21:
## Latency issue remains

I measured 850ms, so it looks like the latency issue remains

[image]

## Initial build issues

Following the docs made the build crash https://www.raspberrypi.com/documentation/computers/camera_software.html#building-rpicam-apps-without-building-libcamera

I had to add a new flag to disable IMX500 postprocessing when running `meson setup`:
```bash
meson setup build -Denable_libav=enabled -Denable_drm=enabled -Denable_egl=disabled -Denable_qt=disabled -Denable_opencv=disabled -Denable_tflite=disabled -Denable_hailo=disabled -Denable_imx500=false
```

naushir, 2024-11-22:
Can you apply the following changes and see if that helps?

```diff
diff --git a/encoder/libav_encoder.cpp b/encoder/libav_encoder.cpp
index 05aa996993ae..d7baf2bf71d7 100644
--- a/encoder/libav_encoder.cpp
+++ b/encoder/libav_encoder.cpp
@@ -80,22 +80,12 @@ void encoderOptionsH264M2M(VideoOptions const *options, AVCodecContext *codec)

 void encoderOptionsLibx264(VideoOptions const *options, AVCodecContext *codec)
 {
-       codec->max_b_frames = 1;
-       codec->me_range = 16;
-       codec->me_cmp = 1; // No chroma ME
-       codec->me_subpel_quality = 0;
        codec->thread_count = 0;
        codec->thread_type = FF_THREAD_FRAME;
        codec->slices = 1;

-       av_opt_set(codec->priv_data, "preset", "superfast", 0);
-       av_opt_set(codec->priv_data, "partitions", "i8x8,i4x4", 0);
-       av_opt_set(codec->priv_data, "weightp", "none", 0);
-       av_opt_set(codec->priv_data, "weightb", "0", 0);
-       av_opt_set(codec->priv_data, "motion-est", "dia", 0);
-       av_opt_set(codec->priv_data, "sc_threshold", "0", 0);
-       av_opt_set(codec->priv_data, "rc-lookahead", "0", 0);
-       av_opt_set(codec->priv_data, "mixed_ref", "0", 0);
+       av_opt_set(codec->priv_data, "preset", "ultrafast", 0);
+       av_opt_set(codec->priv_data, "tune", "zerolatency", 0);
 }
```

Other than that, I'm not really sure what more we can tweak here for latency.

naushir, 2024-11-22:
With the above settings I measured approx 400-500ms of latency.  Unfortunately I doubt this can be improved much more.

naushir, 2024-11-22:
Without network streaming, my latency reduces to approx 100ms, so the network conditions do play a huge factor here.

ben-xD, 2024-11-22:
> Without network streaming, my latency reduces to approx 100ms, so the network conditions do play a huge factor here.

What do you mean without networking streaming? Do you mean you just play the video directly on the raspberry pi, avoiding encoding?

I'm using an ethernet cable and other platforms over network have <250ms latency.

naushir, 2024-11-22:
The additional latency seems to be down to us using libav for video encoding.  Maybe you can try using gstreamer directly and avoid rpicam-apps and libav?  I'm afraid that I'm not familiar enough with gstreamer to provide examples though.

ben-xD, 2024-11-22:
Thanks, with those changed I'm down to 510ms. I will have a look at gstreamer directly.

[image]

[image]

naushir, 2024-11-22:
That pretty much matches my results.  But I don't really know what more to do to tweak things :(

naushir, 2024-12-04:
I've added a new flag (``--low-latency``) used with libav that tweaks the codec for low latency encode: https://github.com/raspberrypi/rpicam-apps/commit/6287bffec8553eea41fa30277c2aaeb5616f8a06.

I've measured end-to-end latency of 150ms (including sensor acquisition, processing, encoding, decoding, display) which is probably the lower limit now.  You can test this out with the following command:

```bash
rpicam-vid -t 0s -o - -n --bitrate 5mbps --libav-video-codec libx26 --libav-format mpegts --low-latency --width 1280 --height 720 | ~/hello_wayland/build/hello_wayland --sw --no-wait --low-delay pipe:
```

``hello_wayland`` is a replacement for ``ffplay`` and does pretty much the same thing (decode + display) with as low latency as possible.  You can find the source for the application here: https://github.com/jc-kynesim/hello_wayland

Feel free to try it out, I don't think there are any more tweaks I can add to this.

closed 2024-12-04 by naushir